felis/USB_Host_Shield_2.0 #411: [Arduino DUE] Conflicting constant declarations for tokSETUP/IN/OUT/INHS/OUTHS with Arduino SAM libs

marcio-ao, 2018-09-21:
In "max3421e.h", the following constants conflict with constants declared in the Arduino headers for the DUE:

```
/* Host transfer token values for writing the HXFR register (R30)   */
/* OR this bit field with the endpoint number in bits 3:0               */
#define tokSETUP  0x10  // HS=0, ISO=0, OUTNIN=0, SETUP=1
#define tokIN     0x00  // HS=0, ISO=0, OUTNIN=0, SETUP=0
#define tokOUT    0x20  // HS=0, ISO=0, OUTNIN=1, SETUP=0
#define tokINHS   0x80  // HS=1, ISO=0, OUTNIN=0, SETUP=0
#define tokOUTHS  0xA0  // HS=1, ISO=0, OUTNIN=1, SETUP=0
#define tokISOIN  0x40  // HS=0, ISO=1, OUTNIN=0, SETUP=0
#define tokISOOUT 0x60  // HS=0, ISO=1, OUTNIN=1, SETUP=0
```

In "[system/libsam/include/USB_host.h](https://github.com/matthijskooijman/arduino-platform-sam/blob/1911a8746292d1a6355131da7cbd918141cbb4e2/system/libsam/include/USB_host.h#L35):

```
#define tokSETUP		UOTGHS_HSTPIPCFG_PTOKEN_SETUP
#define tokIN			UOTGHS_HSTPIPCFG_PTOKEN_IN
#define tokOUT			UOTGHS_HSTPIPCFG_PTOKEN_OUT
#define tokINHS			UOTGHS_HSTPIPCFG_PTOKEN_IN
#define tokOUTHS		UOTGHS_HSTPIPCFG_PTOKEN_OUT
```

In my own code, I try to replace "#defines" with "constexpr" inside a "namespace". Not sure if there would be an interest in doing that here.

-- Marcio

xxxajk, 2018-09-21:
UHS30 avoids the namespace collisions for most things. If you can use it in your project, I urge you to give it a try.

marcio-ao, 2018-09-24:
@xxxajk: That's a possibility. The other option might be for me to simply fork this library and make a version just for Marlin and clean stuff up a bit. Marlin only needs the mass storage stuff, there is a lot in UHS that could probably be eliminated, like support for keyboards, controllers, multiple LUNs, hub enumeration and much of the SPI abstraction stuff (which is already handled by Marlin). I wonder how much simpler it could be made if it were to be whittled down to the bare essentials.