felis/USB_Host_Shield_2.0 #412: Dongle from list is not working

beowulf573, 2018-09-27:
I have a 3.3V Pro Mini and host board ordered from banggood.com and a couple dongles from the compatibility list, [this](https://www.ebay.com/itm/200468994562?ssPageName=STRK:MEWNX:IT&_trksid=p3984.m1439.l2649#ht_2636wt_1189) one.  I'm powering the VBUS line with 5V tied to the raw input on the mini and cut the trace from VBUS to the rest of the board.

So far I've not been able to get a PS Navigation controller to connect to it.  I was able to get the same host board and code to work with an IOGear dongle, so I think everything is setup correctly.

Has anyone recently purchased one of these dongles and had it work?   Any other suggestions?  I can always purchase another IOGear dongle but want to ensure there's nothing left to try since I already have these.  Thanks.

beowulf573, 2018-09-28:
Here's some logging information.  Until the failure the only differences I notice (aside from the MAC) is the Poll Intrv settings is 0x00 with the failing dongle and 0x01 on the successful dongle.  I'll be digging more into this as time permits.

> PS3 Bluetooth Library Started
> BTD ConfigureDevice
> BTD Init
> Addr: 01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	81
> Attributes:	03
> MaxPktSize:	0040
> Poll Intrv:	01
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	82
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	00  // is 1 on successful connection
> Endpoint descriptor:
> Length:		07
> Type:		05
> Address:	02
> Attributes:	02
> MaxPktSize:	0040
> Poll Intrv:	00 // is 1 on successful connection
> Bluetooth Dongle Initialized
> HCI Reset complete
> Write class of device
> Local Bluetooth Address: 33:03:30:09:89:F3
> Wait For Incoming Connection Request
> Gamepad is connecting
> Class of device: 00 05 08
> Incoming Connection Request
> Connection Failed: 10
> Connection Failed: 10
>

Lauszus, 2019-01-06:
Thanks. I have removed it from the list.

closed 2019-01-06 by Lauszus