felis/USB_Host_Shield_2.0 #415: board_qc running error

ashish1779, 2018-10-11:
Issue with Aruino Mega ADK

Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision invalid. Value returned: 00
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test

Simple code without including USB_Host_Shield_2.0 is running but with this library it is showing the error
"OSC did not started".
I also got "OSCOKIRQ failed to assert" error some time.

Lauszus, 2018-10-14:
They have most likely just moved the SS and INT pins to some other pins.

ashish1779, 2018-10-14:
No SS PIN is as it is

Lauszus, 2018-10-14:
How have you verified that?

Lauszus, 2019-01-06:
Closing, as it is likely a hardware issue.

closed 2019-01-06 by Lauszus