felis/USB_Host_Shield_2.0 #422: Feature request: Implement (abstract) serial class

riban-bw, 2018-10-14:
It may be advantageous to implement a generic serial class. This may then be used to interface various systems, e.g. use MIDI library with UHS2 Serial interface to give feature rich MIDI interface (rather than re-implement in UHS2 MIDI code).

Serial extends Stream which extends Print which have following virtual functions:

```
virtual int available() = 0;
virtual int read() = 0;
virtual int peek() = 0;
virtual size_t write(uint8_t) = 0;
virtual size_t write(const uint8_t *buffer, size_t size);
virtual int availableForWrite() { return 0; }
virtual void flush() { /* Empty implementation for backward compatibility */ }
```

Serial implements following public functions:

```
SoftwareSerial(uint8_t receivePin, uint8_t transmitPin, bool inverse_logic = false);
~SoftwareSerial();
void begin(long speed);
bool listen();
void end();
bool isListening() { return this == active_object; }
bool stopListening();
bool overflow();
```

Lauszus, 2018-10-14:
FYI this has already been done for the SPP driver: https://github.com/felis/USB_Host_Shield_2.0/blob/9b829e28b49ebe2c772b72793ce78f3cc18aed7f/SPP.h#L61.

Lauszus, 2018-10-14:
Would be great if someone that uses the MIDI driver could send a PR. If not @riban-bw, then maybe @YuuichiAkagawa?

xxxajk, 2018-10-15:
Just a small note here, UHS30 will be replacing any native host modes at some point. 
Not as a device, but as host.
The benefit of that is you can add USB Host Shields (up to two of them!) along with the native, gives you 3 ports.

PaulStoffregen, 2018-10-15:
On Teensy, I've tried to implement the same API as Francios Best's Arduino MIDI Library, for both device and host USB MIDI.  Having talked with numerous musicians and DJs over the years, it's pretty clear people understand the Arduino MIDI Library functions with names that actually represent the MIDI messages.

While MIDIUSB has fewer functions and follows the read/write naming convention, I believe it's rather confusing to most users, partly due to the more complex syntax for handling the arrays, partly due to use of code numbers rather than recognizable names like sendNoteOn().

riban-bw, 2018-10-16:
@PaulStoffregen I agree that MIDIUSB is too low level. My point was that Francios Best;s Arduino MIDI Library already has USB transport functionality so it would be good if we could hook in UHS to that which would leverage all that high-level MIDI interface goodness. Francios interfaces with USB via the MIDIUSB library. If we could add MIDI Library interface to UHS and add that new UHS interface to MIDI Library then we could use the MIDI Library. This would require changes to both projects but may be worth investigating.

@xxxajk I tried UHS3 but it does not (yet) support ESP8266 which is the platform I am currently developing on (mostly because it was the only 3.3V MCU I had to hand). I would prefer to use UHS3 with its lovely interrupt driven code and remove all that (naff) polling but I doubt I will have time to invest in adding ESP8266 support to UHS3 any time soon.

riban-bw, 2018-10-16:
I will create a separate feature request for MIDI Library interface and add this detail.