felis/USB_Host_Shield_2.0 #423: Feature request: Interface with Arduino MIDI Library

riban-bw, 2018-10-16:
It may be advantageous to interface UHS2 with Francios Best's high-level Arduino MIDI Library which has USB transport functionality. AML interfaces with USB via the MIDIUSB library. A similar interface could be added to UHS2 then a request to add support for this interface AML would enable the interface.

MIDIUSB implements the following public functions:

- MIDI_(void);
- uint32_t available(void);
- midiEventPacket_t read(void);
- void flush(void);
- void sendMIDI(midiEventPacket_t event);
- size_t write(const uint8_t *buffer, size_t size);
- operator bool();

It may be simplest to implement these as a new interface to UHS2 and add a definition like USBCON.

riban-bw, 2018-10-20:
I have worked up some proof of concept code. I will submit a PR when it is tested and proven to work.

xxxajk, 2018-10-21:
I suggest that this is made as a separate library that can refer to this one instead. That could make it easier to add extra midi interfaces with a single api... 
reason? UHS2.0 is getting pretty bloated.

riban-bw, 2018-10-22:
@xxxajk On the contrary, my proposal is to replace the MIDI driver with one that only provides the USB interface to the device and uses an external library to provide all the high-level MIDI functions, removing _some_ of the bloat from UHS2. This is taking a little longer than I had hoped because I cannot see any documentation on how UHS2 works or the driver (plugin) architecture (which is neat) so I am effectively reverse engineering it - well figuring out from code. There may be some documentation delivered with (or along side) this PR when I eventually get it out.

riban-bw, 2018-10-31:
Just a quick progress update. I found that the USB transport in MIDI Library was incomplete and inefficient (when used with UHS2) so have implemented MIDI USB transport within the PR (which is yet to be submitted but imminent). My implementation reduces the quantity and size of buffers required and also reduces load compared with using the MIDI Library USB transport. I have implemented the library and completing some tests before submitting a PR. (Some issues that have been delaying me are actually within the core UHS2 code so expect issue reports for those soon!)

Lauszus, 2019-01-06:
@riban-bw any updates?

Lauszus, 2019-01-28:
@riban-bw sounds good, no hurries.

YuuichiAkagawa, 2020-12-20:
I have started developing the transport for Arduino MIDI Library Ver.5. 
https://github.com/YuuichiAkagawa/Arduino-UHS2MIDI

It is developed as an independent project like other transports.