felis/USB_Host_Shield_2.0 #424: Transmitting keyboard inputs via bluetoth to pc [Help]

mettdrescher, 2018-10-31:
I am some sort of a noob and need some help.
If i use the USBHIDMultimediaKbd example sketch i can receive the inputs from my keyboard with a usb host shield, a arduino mini pro and a FT232RL on my computer. Thats working fine.

But how can i send the received data via a HC-05/ HC-06 with flashed software to my bluetooth dongle.
I think i wired up everthing correctly, i can even pair the bluetooth chip with my dongle.

I just need the programm which i must put on my arduino. I am sure the information i need is in this forum but i can't really find it....

Thanks in advance

karakum, 2018-12-29:
I am working on the same project now.
It seems that you haven't seen this https://youtu.be/qyRJgtwX0cY
You must "upgrade" your HC-05/ HC-06 with RN-42 wirmware. Otherwise it just serial port(HC0X) and unnable to deal with BT HID

Lauszus, 2019-01-06:
Closing, as this is not really relevant to this library.

closed 2019-01-06 by Lauszus