felis/USB_Host_Shield_2.0 #425: Stops detecting USB device after disconnection

riban-bw, 2018-10-31:
Occasionally UHS2 stops detecting newly plugged device after previous device is removed. I have found that performing a USB::Reset seems to stop this behaviour so maybe something configured by USB::Reset is not being set during initialisation or being subsequently changed.

maustin-lpi, 2018-11-28:
I'm seeing a similar issue with my XBOX ONE Controller using the XBOXONE sketch.

I can see the controller is recognised after being reconnected, based on the returned value from XBox.XboxOneConnected(), but its not reliably reading button presses, etc from the connected controller.

If I disconnect and reconnect the USB cable, I can usually get it to connect properly.  I can also achieve this with a hardware reset of the Arduino or USB Shield.

I might try adding the USB::Reset approach you've mentioned above until this issue gets properly resolved

Lauszus, 2019-02-04:
@Ryzee119 thanks for the feedback. Any change you could test the following PR: https://github.com/felis/USB_Host_Shield_2.0/pull/438 and see if that fixes the issue.

Lauszus, 2019-02-04:
@felis please see the comment by @Ryzee119 seems like this could be called by always calling ```busprobe``` here:

https://github.com/felis/USB_Host_Shield_2.0/blob/ffaba6dad3f4bd41bf65f0833977adf65b628efd/usbhost.h#L570-L573

Sounds like the ```CONDETIRQ``` is not always set?