felis/USB_Host_Shield_2.0 #426: Add support for HoodLoader2

limited660, 2018-11-03:
Can support be added for [HoodLoader2](https://github.com/NicoHood/HoodLoader2)? I've tried to find the correct board name to add to avrpins.h but nothing is working for me.

xxxajk, 2018-11-04:
Hoodloader2 is just a bootloader, I don't understand the question.

limited660, 2018-11-04:
I believe it changes the board type because when I try to upload to 16u2 it tells me I need to define the board in avrpins.h

xxxajk, 2018-11-04:
Are you running the library on the 16u2 or the target MCU?

limited660, 2018-11-04:
Attempting to upload it to 16u2, that's the only way to use HID between Uno and computer

limited660, 2018-11-07:
I'll get you the output once I get a chance to install Hoodloader2 again.

Lauszus, 2019-01-06:
Closing as irrelevant.

closed 2019-01-06 by Lauszus