felis/USB_Host_Shield_2.0 #429: oscillator state unexpected (examples/board_qc/board_qc.ino)

manuel-91, 2018-11-15:
I've run the board_qc and get an error about the oscillator. Serial output:
```
Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K................ SPI long test passed
GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
Test failed. Value written: 00 Value read: FF 
Press any key to continue...
GPIO test passed.
PLL test. 100 chip resets will be performed
Resetting oscillator
Reset number 0 Time to stabilize - 361 cycles
Reset number 1 Time to stabilize - 360 cycles
Reset number 2 Time to stabilize - 361 cycles
Reset number 3 Time to stabilize - 10 cycles
Reset number 4
Current oscillator state unexpected.
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test
```
1. Do I need a new board? 
I have never detected a USB device with USB_desc (multible USB devices), in the mainloop it doesn't run into the if part: Usb.getUsbTaskState() == USB_STATE_RUNNING.

2. Is it worth to do the GPIO test when I don't want to use them? 
3.Will this give a hint about the oscillator?

Thanks for your help!

xxxajk, 2018-11-16:
Looks like reset is failing. Can you send a photo of your setup?

manuel-91, 2018-11-18:
It's just a PC -> Arduino Leonardo -> SainSmart USB Host Shield -> Keyboard (or keypad, mouse, usb flash drive)

[image]

manuel-91, 2018-11-18:
I've got a multimeter and oscilloscope available. Something I shall check about the reset or SPI?

xxxajk, 2018-11-18:
Put your 'scope probe into 10X and check for wave forms on the crystal pins. 
You should see 12MHz during the long test. Also isn't Leonardo 3.3v?

xxxajk, 2018-11-18:
Actually, I looked it up, it is 5V...

xxxajk, 2018-11-18:
One other thing to check, is if the 3.3v rail is smooth. 
I've had USB Host shields fail when the 3.3V regulator was bad.
If 3.3v is not stable, replace U3, it is an LP2985-33DBVR. 
I have actually stopped using this particular regulator in my projects as it tends to not hold up very well..

manuel-91, 2018-11-19:
Thank you for your advices.
I can see a 12MHz sinus at oscillator after powering up. Seems legit. On the one side of the oscillator the amplitude is higher, on the other pin it is higher. It did not die when measuring, even on the input side:-)
[image]

* When I started the board_qc script: Oscillator running nicely and 100cycles run fine. Oh, I forgot to change the probe to 10x! The text "waiting for usb device" was shown. But the USB keyboard I connected was not detected.

* When I started the board_qc script with 10x probe on the oscillator I got
```
[...]
PLL test. 100 chip resets will be performed
Resetting oscillator
Reset number 0 Time to stabilize - 361 cycles
Reset number 1 Time to stabilize - 12 cycles
Reset number 2
Current oscillator state unexpected.
```
I repeated this another 2 times, same result, only differet amount of working lines (0-5), then it crashes.

I checked the 3.3V line: Nothing special:
* With DC Coupling the voltage is between 3.25..3.35V.
* With AC Coupling the voltage is within ±2mV, scrolled from ns to second scale.

In the next session I will try to disconnect the 3.3V between arduino and the shield and will power the shield with an external power supply.

Any ideas?

manuel-91, 2018-11-19:
One question about 9V wall wart: Shall I connect both 9V and 5V? Will the 5V from voltage regulator onboard and 9V interfer? Or do you mean _only_ 9V without the usb/serial output? ...to surround this I could use the LED to indicate a positive/negative result...

On the keyboard it's written that it just needs 50mA − not a gaming LED energy hungry keyboard. I can hardly imagine that the 500mA of the USB standard are not sufficient.

The 3.3v is smooth in my understanding. I'd say <10mV noise is smooth in that case.

xxxajk, 2018-11-19:
The board will automatically switch to the wallwart, and both may be plugged in at the same time without worry. Many issues like this can be resolved with using the 9v power input or a powered USB hub on the host shield output and/or from the PC. It is something you need to fiddle with to find out sometimes.

10mV ripple should be acceptable.

manuel-91, 2018-11-20:
Thanks for the hint with the powersupply.
I found an arduino _uno_ and ran the board_qc with the shield on this arduino. Result: Passed the oscillator section with success. But: Still it doesn't find a device:
```
[...]
Reset number 98 Time to stabilize - 329 cycles
Reset number 99 Time to stabilize - 329 cycles

Checking USB device communication.

Waiting for device..
```
So the uno is doing something different compared to the leonardo.

Also the USB_desc script cannot find a USB decive.

Supprisingly to me your recomended combination 9V powersupply + usb cable -> arduino _leonardo_  did not work (board_qc), after 3 cycles I get `Current oscillator state unexpected.`

And yes, the powersupply is working. Only the powersupply makes the arduino light up. I connected _first_ the 9V and then the usb to make sure that it starts up with smooth supply.

I'm slightly confused!^^

xxxajk, 2018-11-21:
Since your board lacks the USB power switch, you may have to start with the target device unplugged and wait until that message? 
Checked continuity on the vBUS, ground D+ and D- lines? 
Are the resistors OK?

As far as the board that does not even get to that point, I would replace the 3.3V regulator.
If all those things do not help you find the issue, then I would say the board is broken somehow.

manuel-91, 2018-12-08:
Sorry, I'm giving up. Thank you Andrew for your help anyway!

closed 2018-12-08 by manuel-91

xxxajk, 2018-12-09:
Probably a leonardo problem or bug. 
I do not have this board, thus can not confirm if it works or not with it.