felis/USB_Host_Shield_2.0 #432: PS4 BT Connection - Issue controlling two robots

mj2554, 2018-11-28:
I have an interesting issue happening. I have two robots - each with their own Arduino + shield + BT dongle, both using PS4 controllers. I pair each robot individually to their respective dongles with no issues. Then after pairing, I replace the line in the sketch that simply creates the instance when the power button is  pressed on the PS4 controller. However, when I power on both robots and then press the power button on a PS4 controller, the controller which I had paired to robot #1 will connect to robot #2. Further, when I power on PS4 controller #2, it will flash white LED and then turn to solid white. I cannot figure out why after pairing each controller to it's  respective dongle, why they would then connect to the opposite dongles as I've described. I thought that once paired, the connection is unique and the PS4 would only connect to the dongle which it had been paired. Is there some explanation for this, and can multiple robots be run in the same space, each with their own PS4 BT connection?

mj2554, 2018-11-28:
OK, I've discovered the issue is that the BT dongles that I have all have the same MAC address. Is there a method to change the MAC address of the BT dongle?

dogtopus, 2018-12-05:
Some bluetooth dongles (espeacially those which use CSR chipsets) allow you to change their bdaddr permanently by using the bdaddr command under Linux system.

For dongles that don't support permanent change you may need to implement something that does the change on your robot. (If you decided to go this route you may find [this](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/tools/bdaddr.c) helpful)

If your dongle does not support bdaddr changing, buy a new one.

Thinh-Houy, 2021-09-25:
I have meet this same problem and looking for the solution.