felis/USB_Host_Shield_2.0 #434: How to access files from a USB stick? Any examples?

rtek1000, 2018-12-03:
Hello,

Congratulations! Great job!

I found a sample code that allows you to upload and download files to an SD card (webserver):
https://github.com/G6EJD/ESP32-8266-FIle-Upload
(Demo video: https://www.youtube.com/watch?v=1OkEXeRbq_M )

I wish I could do this with a USB stick.

I've noticed that common Arduino boards have poor memory, but would it be possible with ESP8266, or maybe with ESP32?

Is there any possibility of leaving the access to the files in a more transparent way, or similar to the access to the SD card?

The reason for this search is the need to update files from the SD card that is on an mp3 player with the DFplayer Mini.
[image]

The DFplayer mini has a USB port that can operate as a Host (to read files from a USB stick) or can also operate as a device, so that the PC can access the files from the SD card (and there is also a possibility to use the DFplayer Mini as sound card, but I believe the "clones" do not have this support)
[image]

I accessed the USB port of the DFplayer Mini by Windows 7 and also by Linux (Raspbian - Raspberry Pi). A mass storage device appears in Windows (XY5200 system). And in Linux it is possible to access the DFplayer Mini as a common USB stick, it appears as a partition "/ dev / sda1" (attached image).
[image]

The Raspberry Pi problem is the instability of the system in relation to data corruption (of your own Operating System), especially when there is a power outage.

Would it be possible to provide a compileable example (.ino) with NodeMCU (ESP8266) or ESP32?

dogtopus, 2018-12-05:
There is a sample project called `testusbhostFAT` under the example directory. It isn't pretty but certainly has a lot of information you need.

Unfortunately you need to do the web server part yourself, which doesn't look that hard from the example code you mentioned.

rtek1000, 2018-12-05:
Hello, thanks for the tip, I kept working here and found another way, I managed to switch access to the SD card between the DFPlayer Mini and ESP8266, so I can already use the code I gave you above "ESP32-8266-FIle-Upload", with few changes.

rtek1000, 2018-12-05:
But it would be great to have a sample file access code, similar to what the SDFat library already offers. Would standardization be possible?

rtek1000, 2018-12-05:
DFPlayer Mini SD card being accessed by ESP8266:
[image]

If one wants to do this: I simply used 2 micro AXICOM relays to toggle the SPI connections, it is interesting that the DFPlayer Mini operates with the Chip Select from the floating card, it is not used.

closed 2018-12-05 by rtek1000

rtek1000, 2019-10-10:
@imranaalam 

Hello,

The code is pretty much the same as the link above (https://github.com/G6EJD/ESP32-8266-FIle-Upload).

The only secret on the hardware side is the power of the SD card. (Can use a MOSFET, I had success with AO3401, also with BC328 but is less efficient).

Pull-up resistors* to the SD card are recommended, to avoid random errors. Relay connections only disconnect the SD card from one circuit (DFPlayer), and connect to another circuit (ESP8266) as if it were a manual operation. On the DFPlayer side does not need to add pull-up resistors. 

You may need to turn off DFPlayer before connecting it to the SD card (only 3 communication pins, MOSI, MISO, SCK) and power the SD card at the same time. (Please, see if in your DFPlayer the CS pin is also floating, as it was in mine). You may need to send some command to force DFPlayer to operate the SD card again (DFPlayer usually recognizes SD card automatically after inserting in circuit).

Each time you switch between ESP8266 and DFPlayer, the SD card needs to be initialized. In DFPlayer, you may need just cycle the power.

For ESP8266 need to connect the 4 wires (MOSI, MISO, SCK, SS). Initializing the library after powering up the SD card, be careful to properly power up the SD card so that it does not lock (SD card does not accept to be booted by library, or does not accept write).

Maybe the power up sequence is described in the SD specification, I didn't have time to look much, but I haven't found anything yet about it. Just found a datasheet that reports about 250ms of power up time, this power up time should already be included in library initialization (SDFat.h).

I've been testing this sequence (code) below to disable (and reactivate later without the SDFat library reporting error) to save battery power, I noticed that a delay (100us) had to be added before powering up the SD card so that the control pins are properly polarized.

Note: This code below was tested on STM32F103CBT6, BluePill as the SPI port needed to be remapped (SPI Clock: 35MHz). Please, try modifying the power-up sequence for different SD cards to test if it can be considered as a standard sequence.

```C++
#include <SPI.h>
#include "SdFat.h"

void setup() {
  // put your setup code here, to run once:

  SD_powerOff(); // If the SPI port has been automatically initialized, it forces disabling
}

void loop() {
  // put your main code here, to run repeatedly:
  SD_powerOn();
  // put your SD code here:
  
  if (!sd.begin(chipSelect, SD_SCK_MHZ(35))) {
    //sd.initErrorHalt();
  }

  SD_powerOff(); // when you no longer use the SD card

  Delay(1000); // Sleep function is better than delay function, to save energy

}

void SD_powerOn() {
  digitalWrite(chipSelect, HIGH);
  digitalWrite(PB5, HIGH); // MOSI
  digitalWrite(PB4, HIGH); // MISO
  digitalWrite(PB3, HIGH); // SCK
  pinMode(chipSelect, OUTPUT);
  pinMode(PB5, OUTPUT);
  pinMode(PB4, INPUT);
  pinMode(PB3, OUTPUT);

  delayMicroseconds(100);

  digitalWrite(SD_PWR, LOW);
  pinMode(SD_PWR, OUTPUT);
}

void SD_powerOff() {
  pinMode(PB3, INPUT);
  pinMode(PB4, INPUT);
  pinMode(PB5, INPUT);
  pinMode(chipSelect, INPUT);
  digitalWrite(SD_PWR, HIGH);
  pinMode(SD_PWR, INPUT);
}
```

*(Illustrative image of MOSFET and pull-up resistors)

[image]

rtek1000, 2019-10-11:
ESP8266 is very limited, try searching for ESP32

DIY 10 Player Game Console [ESP32, Arduino, Audio Video]:
https://www.youtube.com/watch?v=VvkpKVtYKmk

imranaalam, 2019-10-12:
I am using ESP32. i have a similar problem like yours. i have to download the new words file (english words) from internet (cloud) and store them in sdcard. 

i send a word sentence to the esp32 and if no new words are found in this sentence then SP32 will play sentence words from local SDcard else it will request for the new words from cloud and then it will be stored in the SDcard for next time. simply, It acts as a cache.

sentence is usually 10-12 words long. I can download this from internet on each request as wav file. but, can i play it directly through DAC on ESP32?

imranaalam, 2019-10-12:
when SDcard is attached to PCB:
https://www.acmesystems.it/pcb_microsd
http://elm-chan.org/docs/mmc/mmc_e.html

rtek1000, 2019-10-12:
I found it using Google, see if it helps:

Make your ESP32 talk, playing WAV files on your ESP32, digital sound DAC ADC:
https://www.youtube.com/watch?v=gxpHDUHcBMk

imranaalam, 2019-10-13:
yes i have tried ESP32 talk and it is working. 

i researched some more on your guidelines and found a library that can read MP3 from SDcard and from  HTTPStream and play it on ESP32 using a single $0.05 NPN 2N3904 transistor and optionally adding a 220uF cap from USB5V to GND 

Refernce:
Arduino library to play MOD, WAV, FLAC, MIDI, RTTTL, MP3, and AAC files on I2S DACs or with a software emulated delta-sigma DAC on the ESP8266 and ESP32
https://github.com/earlephilhower/ESP8266Audio#software-i2s-delta-sigma-dac-ie-playing-music-with-a-single-transistor-and-speaker

imranaalam, 2019-10-14:
ESP8266Audio is taking too much space 
Sketch uses 840666 bytes (64%) of program storage space. Maximum is 1310720 bytes.
Global variables use 39484 bytes (12%) of dynamic memory, leaving 288196 bytes for local variables. Maximum is 327680 bytes.

//https://vk.com/wall-80234063_2336
//http://forum.amperka.ru/threads/%D0%98%D0%BD%D1%82%D0%B5%D1%80%D0%BD%D0%B5%D1%82-%D1%80%D0%B0%D0%B4%D0%B8%D0%BE%D0%BF%D1%80%D0%B8%D0%B5%D0%BC%D0%BD%D0%B8%D0%BA-%D0%BD%D0%B0-esp32.15262/page-6
//http://192.155.89.84/chris/M5Stack-MultiApp-Advanced/blob/665540a83bc4b2318440080cc81f41fbf161c3ce/src/Apps/WebRadio.h
//https://github.com/earlephilhower/ESP8266Audio/issues/13
//https://www.hackster.io/julianfschroeter/stream-your-audio-on-the-esp32-2e4661

#include <Arduino.h>
#include <WiFi.h>
//#ifdef ESP32
//    #include <WiFi.h>
//#else
//    #include <ESP8266WiFi.h>
//#endif
#include "AudioFileSourceICYStream.h"
#include "AudioFileSourceBuffer.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"

// To run, set your ESP8266 build to 160MHz, update the SSID info, and upload.

// Enter your WiFi setup here:
const char *SSID = "dlink";
const char *PASSWORD = "-------";

// Randomly picked URL
const char *URL="http://philae.shoutca.st:8446/stream/1/";

AudioGeneratorMP3 *mp3;
AudioFileSourceICYStream *file;
AudioFileSourceBuffer *buff;
AudioOutputI2SNoDAC *out;

// Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  const char *ptr = reinterpret_cast<const char *>(cbData);
  (void) isUnicode; // Punt this ball for now
  // Note that the type and string may be in PROGMEM, so copy them to RAM for printf
  char s1[32], s2[64];
  strncpy_P(s1, type, sizeof(s1));
  s1[sizeof(s1)-1]=0;
  strncpy_P(s2, string, sizeof(s2));
  s2[sizeof(s2)-1]=0;
  Serial.printf("METADATA(%s) '%s' = '%s'\n", ptr, s1, s2);
  Serial.flush();
}

// Called when there's a warning or error (like a buffer underflow or decode hiccup)
void StatusCallback(void *cbData, int code, const char *string)
{
  const char *ptr = reinterpret_cast<const char *>(cbData);
  // Note that the string may be in PROGMEM, so copy it to RAM for printf
  char s1[64];
  strncpy_P(s1, string, sizeof(s1));
  s1[sizeof(s1)-1]=0;
  Serial.printf("STATUS(%s) '%d' = '%s'\n", ptr, code, s1);
  Serial.flush();
}

void setup()
{
 
  Serial.begin(115200);
  delay(1000);
  Serial.println("Connecting to WiFi");

  WiFi.disconnect();
  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
  
  WiFi.begin(SSID, PASSWORD);

  // Try forever
  while (WiFi.status() != WL_CONNECTED) {
    Serial.println("...Connecting to WiFi");
    delay(1000);
  }
  Serial.println("Connected");

  audioLogger = &Serial;
  file = new AudioFileSourceICYStream(URL);
  file->RegisterMetadataCB(MDCallback, (void*)"ICY");
  buff = new AudioFileSourceBuffer(file, 5*1024);
  // change it ir the librart on which it depends
  //i.e SPIRAM or change the streaming link that can present slowly n slow buffers
  // bibber buffer more stable 
  buff->RegisterStatusCB(StatusCallback, (void*)"buffer");
  out = new AudioOutputI2SNoDAC();
  mp3 = new AudioGeneratorMP3();
  mp3->RegisterStatusCB(StatusCallback, (void*)"mp3");
  mp3->begin(buff, out);
}

void loop()
{
  static int lastms = 0;

  if (mp3->isRunning()) {
    if (millis()-lastms > 1000) {
      lastms = millis();
      Serial.printf("Running for %d ms...\n", lastms);
      Serial.flush();
     }
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}

imranaalam, 2019-10-18:
@rtek1000 
why don't you write an article on hackster.io?

DFplayer comes too be the best choice; i.e. value/ price. 
its low cost and high quality mp3 out. 
Don't have to put strain on ESP32, as it works through RX TX.
its easy to code and easy to understand.
 

![dfplayer schema](https://user-images.githubusercontent.com/17583722/67093265-a676c480-f1ca-11e9-9769-fd8cc18b898a.jpg
DFplayer diagram
https://i.stack.imgur.com/QwcLV.jpg

imranaalam, 2019-10-25:
i have found some additional content that may be useful in making an article and videos  for hackster or hackaday. it includes internal schematic diagram and attaching to USB video.

Reference: 
video: 
https://www.youtube.com/watch?v=n8pCJTTYzZo
Schematic: page:18-19
https://www.elechouse.com/elechouse/images/product/MP3%20Sound%20Mini%20SD%20Card%20Module/MP3%20Mini%20SD%20Card%20Module.pdf
circuit images: 
https://www.mikrocontroller.net/attachment/241766/DSC08740.jpg
some more videos:
https://www.youtube.com/watch?v=hCRpr2SnBac
https://www.youtube.com/watch?v=m_qBmrf9Ulw

imranaalam, 2019-10-25:
select either SDcard or Flash circuit:
https://robu.in/wp-content/uploads/2017/04/Mini-SD-Card-MP3-Sound-Module-For-PIC-Arduino-WTV020-SD-ROBU.IN_.pdf

imranaalam, 2019-10-25:
similar module, and similar problem. you may like to answer the thread
https://arduino.stackexchange.com/questions/63550/can-i-use-wtv020-sound-module-as-sd-card-reader

imranaalam, 2019-10-25:
easyEDA SDcard and arduino and DFplayer diagram:
https://easyeda.com/Kurisu/Shield_WTV020_SD_copy-3effb6dec0e740d0a733be37ce3d7ab1

imranaalam, 2019-10-25:
DFPlayer schematic diagram
http://domoticx.com/wp-content/uploads/2018/12/DFPlayer-mini-module-schema.jpg

imranaalam, 2019-11-11:
some more interesting articles about SPI and MP3 modules

https://www.winbond.com/resource-files/w25q16jv%20spi%20revd%2008122016.pdf
https://www.eevblog.com/forum/microcontrollers/mp3-files-on-spi-flash-for-yx5200-24ss-mp3-chip/
https://www.aliexpress.com/item/33004786600.html

inommo, 2020-05-25:
> DFPlayer Mini SD card being accessed by ESP8266:
> [image]
> 
> If one wants to do this: I simply used 2 micro AXICOM relays to toggle the SPI connections, it is interesting that the DFPlayer Mini operates with the Chip Select from the floating card, it is not used.

Hi, please can I have the schematics of your connexion? I am working hard to connect ESP32 to DFPLAYER.

rtek1000, 2020-05-26:
> 
> 
> DFPlayer Mini SD card being accessed by ESP8266:
> [image]
> 
> If one wants to do this: I simply used 2 micro AXICOM relays to toggle the SPI connections, it is interesting that the DFPlayer Mini operates with the Chip Select from the floating card, it is not used.

Hi, the connections between the SD card and the DFplayer are made on the PCB board of the DFplayer module, and are not available to users. To access these connections I removed the connector from the SD card and used an external connector. Without the SD card connector, it is easier to solder wires on the DFplayer PCB board. 

[image]

jskrnbindra, 2022-01-17:
@rtek1000  and @imranaalam  I've read and re-read this issue 10 times but I'm still unable to get how can I connect my microcontroller (Arduino or Raspberry Pi PICO) to read from DFplayer's SD card. 
Really appreciate all the links you've added @imranaalam .

I'm a complete noobie into electronics and have been into software all my life.   
Can someone please provide a noob friendly schematic diagram on how to get this thing to work?

I have the following doubts:
1. If we remove the card holder from DF player how do we connect the wires to the card? Do you solder?
2. The diagram image that @rtek1000 posted I'm unable to understand (even after googling) the jargon used in the diagram like which component is connected to which. What are things like /ON, /PRESENT, NC, etc. This looks like an excerpt of a bigger diagram, can I please see it all?
3. Are EMI filters necessary? I'm unable to find one in India.
4. Is there another solution (other module or sth) to the problem of being able to play songs and read/write to the card reliably?

rtek1000, 2022-01-17:
Hello,

- If you want to use DFplayer Mini with Arduino see this post:

https://create.arduino.cc/projecthub/munir03125344286/play-audio-in-arduino-dde2e2

- If you want to access the SD card as if it were a USB card reader, connect a USB cable to the DFplayer Mini, then connect to computer, or Raspberry Pi, see the USB pinout:
- - VCC (5V)
- - D-
- - D+
- - GND

https://www.electronics-notes.com/articles/connectivity/usb-universal-serial-bus/connectors-pinouts-cables.php

- If you want to read the DFplayer Mini's SD card using a microcontroller, it is necessary to disconnect the SD card from the DFplayer Mini. So I used relays to switch the SD card between the microcontroller and the DFplayer Mini.

That's why this post is confusing, there are several situations here:
- 1- Access SD card with micro controller, need to unplug the card from DFplayer Mini
- 2- Play music using DFplayer Mini, need to connect SD card (or USB flash drive) to DFplayer Mini
- 3- Access SD card via USB port, DFplayer Mini's USB port operates in Device mode.

And it's also possible to use the DFplayer Mini's USB port as a USB Host, and then connect a USB stick.

The DFplayer Mini's USB port can operate as a host and as a device.

-
-
-

> I have the following doubts:

> If we remove the card holder from DF player how do we connect the wires to the card? Do you solder?

- - To use the relays I soldered the cables.

> The diagram image that @rtek1000 posted I'm unable to understand (even after googling) the jargon used in the diagram like which component is connected to which. What are things like /ON, /PRESENT, NC, etc. This looks like an excerpt of a bigger diagram, can I please see it all?

- - /ON: It is a power control that this scheme has but it is not necessary
- - /PRESENT: It is a switch key that may exist in the SD card socket, usually this switch closes contact when the SD card is inserted.
- - NC: this is very common, it means Not Connected (N.C.). (You might be better off taking a course to learn the basics before you head around.)
- - This image was added only for reference of the SD card pinout, not necessarily the electronic circuit.

> Are EMI filters necessary? I'm unable to find one in India.
- - In my tests it was not necessary to use EMI filters, but it may be necessary to use short cables.

> Is there another solution (other module or sth) to the problem of being able to play songs and read/write to the card reliably?

- - I currently believe that using ESP32 with an audio card is the best alternative, see:
https://github.com/schreibfaul1/ESP32-audioI2S

jskrnbindra, 2022-01-17:
Thanks for the quick reply @rtek1000. Really appreciate your knowledge sharing and efforts.

I understand that I must read more basics so that I'm better able to grasp open knowledge on such forums. I'm improving, reading this issue 5 days later made me understand a lot more than earlier. Just that you're here I'd like to use the opportunity to describe my full motive and make sure I'm headed in the right direction.

My end goal is to create a device like following:

1. The device should take in a micro SD card (or a pen drive)
2. Read at max 10 sound files (say mp3)
3. Display the read files to a mobile app via bluetooth (not internet)
4. Accept a play command (say file name/number) from mobile app via bluetooth
5. play the selected sound file

What is the best possible way to design this device keeping the total cost <= $15?
Another constraint is that the user should be able to give commands while the audio is playing and the commands should be served simultaneously.

jskrnbindra, 2022-01-17:
I played around with BLE modules and I think that is straight forward.

I'm stuck at trying to build on top of DFPlayer as opposed to using an external DAC + Amp to replicate what DFplayer is doing.