felis/USB_Host_Shield_2.0 #435: Xbox One S(2018) USB new generation issue

nathanRamaNoodles, 2018-12-08:
It appears the Xbox one S(2018) controllers had a new hardware update and this library is conflicted again. :(
I know this because I followed this [issue](https://github.com/felis/USB_Host_Shield_2.0/issues/216#issuecomment-201823671), and got a different hardware version.
Here's what i got when I ran [XboxOne.ino](https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Xbox/XBOXONE/XBOXONE.ino) with the debugger.
```
XBOX USB Library Started
Xbox One Controller Disconnected

Unknown Device Connected - VID: 045E PID: 04D4
Xbox One Init Failed, error code: D1

Xbox One Controller Disconnected
```
And here's the USB_desc:
```
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	47
Device Protocol:	D0
Max.packet size:	40
Vendor  ID:		045E
Product ID:		04D4
Revision ID:		0602
Mfg.string index:	02
Prod.string index:	04
Serial number index:	06
Number of conf.:	02

Configuration descriptor:
Total length:		0060
Num.intf:		03
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FE
Intf. Subclass:		1F
Intf. Protocol:		40
Intf.string:		00
Unknown descriptor:
Length:		1C
Type:		14
Contents:	080D0000101C16080D0000102410040003FD1F4000241004040BFA3E
Unknown descriptor:
Length:		FA
Type:		3E
Contents:	80003828180F200008382C180A0000080904020000FF47D0000904020102FE1F40001C1410090000001C161009000000000000000DDA0D2B026F1112000AB0D4000223A3000D4A017A008000000ECF01FA00D10371083B0C300904020000FF47D0000904020102FE1F40001C1410090000001C1610090000002410040003FD1F4000241004040BFD1F40001C0C530BA4015B0A0B08320001000063E3000BBF000008B400010B2C089308B3000201010DC612001201FA0723000000000000000C530BA4015B0C0D010A0002000063E300600008C600020000010100000012016E045E0AAB12010002FF47D0405E04D40402060204060208E60000

Configuration descriptor:
Total length:		0060
Num.intf:		03
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FE
Intf. Subclass:		1F
Intf. Protocol:		40
Intf.string:		00
Unknown descriptor:
Length:		1C
Type:		14
Contents:	080D0000101C16080D0000102410040003FD1F4000241004040BFA3E
Unknown descriptor:
Length:		FA
Type:		3E
Contents:	80003828180F200008382C180A0000080904020000FF47D0000904020102FE1F40001C1410090000001C161009000000000000000DDA0D2B026F1112000AB0D4000223A3000D4A017A008000000ECF01FA00D10371083B0C300904020000FF47D0000904020102FE1F40001C1410090000001C1610090000002410040003FD1F4000241004040BFD1F40001C0C530BA4015B0A0B08320001000063E3000BBF000008B400010B2C089308B3000201010DC612001201FA07230000000C530BA4015B0001010A0002000063E3000BC10001600008C600020001010000000012016E045E0AAB12010002FF47D0405E04D40402060204060208E60000
Request error. Error code:	05

Addr:1(0.0.1)
```

gdsports, 2018-12-09:
Perhaps a change similar to the one below will handle the new PID.

https://github.com/felis/USB_Host_Shield_2.0/pull/430/commits/80e6822672aa7199613c56e8de00d4d3b2c8875f

nathanRamaNoodles, 2018-12-09:
Woah, this is weird.  **I supplied my Arduino Uno with more power by adding a reliable 9V battery pack to the DC jack** instead of my Computer's 5v via USB, and the xbox controller uses the Arduino's regulated 5V to supply it's power.  Now I am getting **more reliable** error codes.

```
Xbox One Controller Disconnected
NC:01

Unknown Device Connected - VID: 045E PID: 02EA
Xbox One Init Failed, error code: D1

Xbox One Controller Disconnected

Xbox One Controller Disconnected
NC:01

Unknown Device Connected - VID: 045E PID: 02EA
Xbox One Init Failed, error code: D1

Xbox One Controller Disconnected
```
And here's the new USB_desc:
```
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	47
Device Protocol:	D0
Max.packet size:	40
Vendor  ID:		045E
Product ID:		02EA
Revision ID:		0301
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		0060
Num.intf:		03
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FE
Intf. Subclass:		1F
Intf. Protocol:		40
Intf.string:		00
Unknown descriptor:
Length:		1C
Type:		14
Contents:	080D0000101C16080D0000102410040003FA3E0000200040407F47D0
Unknown descriptor:
Length:		47
Type:		D0
Contents:	0007050301E4000107050602000004240904020000FF47D0000904020102FF47A0000E0A08040000001C161009000000000000000DF90D4A026F1617000A80C9010163E3000D67

Addr:1(0.0.1)
```
The strange thing is, the PID is already listed in the xboxone.h file.  And, thanks for the suggestion @gdsports, but i guess i don't need it now after that reliable battery pack.

So it's still not working, even though the library supports this hardware model :(

nathanRamaNoodles, 2018-12-09:
I FIGURED IT OUT!!!!
The issue is the Arduino's voltage regulator.  Apparently, it can't handle the Xbox's controllers load.  So I simply powered the controller on(this will make sure the controller relies on its own battery supply), then I connected it to the USB host shield via a USB cable.  Finally, i can use this library, Thanks yall.

P.S. don't connect any power other than the USB power to the Arduino, otherwise, this won't work.

closed 2018-12-09 by nathanRamaNoodles