felis/USB_Host_Shield_2.0 #436: Xbox One S Controller Bluetooth (Connect without Password/Pin)

nathanRamaNoodles, 2018-12-12:
I am having trouble connecting the xbox one s controller to my usb host shield.  I tried the BTHID.ino sketch, but it says this:
```
HID Bluetooth Library Started
Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:13
Please enable discovery of your device
Gamepad found
HID device found
Now just create the instance like so:
BTHID bthid(&Btd);
And then press any button on the device
Connecting to HID device
Connected to HID device
Received Key Request
Bluetooth pin is set too: 1111
Pairing Failed: 06
HCI Disconnected from Device
Wait For Incoming Connection Request
```
I tried passKeys `0000` and `1111` and `1234`, but none of them work.  I also realized, that the xbox one controllers don't require a passkey for connection, but then how do they connect to other devices?  Is there a way to connect without a passkey?

Lauszus, 2019-01-06:
Since I don't own a Xbox One controller I can not debug it. I'll work on it if anyone is kind enough to donate one.

nathanRamaNoodles, 2019-01-06:
I decided to make an [android app](https://github.com/nathanRamaNoodles/Tinker-Controller) that connects any game controller to arduino/esp8266.  This way, we don't have to keep making a new library driver for every new controller out there.

Lauszus, 2019-01-06:
@nathanRamaNoodles cool app. Good luck with your project :)

closed 2019-01-13 by nathanRamaNoodles