felis/USB_Host_Shield_2.0 #437: USB HID - PS3 Clone gamepad (RF Dongle)

britishbadger, 2019-01-03:
I have a Gioteck VX3 Wireless PS3 Controller - Black (https://www.argos.co.uk/product/6319489) and have been trying to get the USBHidJoystick example working. 

The report descriptor is very similar to that on:  https://www.circuitsathome.com/mcu/hid-joystick-code-sample/ but not quite, :

```
0000: 05 01 09 04 A1 01 A1 02 75 08 95 04 15 00 26 FF 
17:26:59.770 -> 0010: 00 35 00 46 FF 00 09 32 09 35 09 30 09 31 81 02 
17:26:59.770 -> 0020: 75 04 95 01 25 07 46 3B 01 65 14 09 39 81 42 65 
17:26:59.770 -> 0030: 00 75 01 95 0C 25 01 45 01 05 09 19 01 29 0C 81 
17:26:59.770 -> 0040: 02 06 00 FF 75 01 95 08 25 01 45 01 09 01 81 02 
17:26:59.770 -> 0050: C0 A1 02 75 08 95 04 46 FF 00 26 FF 00 09 02 91 
17:26:59.770 -> 0060: 02 C0 C0 
17:26:59.770 -> Usage Page Gen Desktop Ctrls(01)
17:26:59.803 -> Usage Game Pad
17:26:59.803 -> Collection Application
17:26:59.803 -> Collection Logical
17:26:59.803 -> Report Size(08)
17:26:59.803 -> Report Count(04)
17:26:59.803 -> Logical Min(00)
17:26:59.803 -> Logical Max(FF00)
17:26:59.803 -> Physical Min(00)
17:26:59.803 -> Physical Max(FF00)
17:26:59.803 -> Usage Z
17:26:59.803 -> Usage Rz
17:26:59.803 -> Usage X
17:26:59.803 -> Usage Y
17:26:59.803 -> Input(00000010)
17:26:59.803 -> Report Size(04)
17:26:59.803 -> Report Count(01)
17:26:59.803 -> Logical Max(07)
17:26:59.803 -> Physical Max(3B01)
17:26:59.803 -> Unit(14)
17:26:59.803 -> Usage Hat Switch
17:26:59.803 -> Input(01000010)
17:26:59.803 -> Unit(00)
17:26:59.803 -> Report Size(01)
17:26:59.836 -> Report Count(0C)
17:26:59.836 -> Logical Max(01)
17:26:59.836 -> Physical Max(01)
17:26:59.836 -> Usage Page Button(09)
17:26:59.836 -> Usage Min(01)
17:26:59.836 -> Usage Max(0C)
17:26:59.836 -> Input(00000010)
17:26:59.836 -> Usage Page Undef(00)
17:26:59.836 -> Report Size(01)
17:26:59.836 -> Report Count(08)
17:26:59.836 -> Logical Max(01)
17:26:59.836 -> Physical Max(01)
17:26:59.836 -> Usage
17:26:59.836 -> Input(00000010)
17:26:59.836 -> End Collection
17:26:59.836 -> Collection Logical
17:26:59.836 -> Report Size(08)
17:26:59.836 -> Report Count(04)
17:26:59.836 -> Physical Max(FF00)
17:26:59.836 -> Logical Max(FF00)
17:26:59.836 -> Usage
17:26:59.836 -> Output(00000010)
17:26:59.836 -> End Collection
17:26:59.836 -> End Collection

```
Connecting and pairing the RF dongle under Linux I can get the following output from lsusb -v: 

```
us 001 Device 044: ID 25f0:c131  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
[... 149 lines cut ...]

```
21:51:37.149 -> Start
started
HU Init
Addr:01
21:51:38.009 -> NC:01
21:51:38.009 -> Cnf:01
21:51:38.009 -> HU configured

```
I'm really stuck, so any help appreciated. I've seen the following comment on the original blog, but I don't understand what is going on, is this where I should be concentrating.

> uint8_t feature_F4_report[4] = { 0x42, 0x0c, 0x00, 0x00 };
uint8_t startDelay = millis() + 5000;

void loop()
{
Usb.Task();

if ((Usb.getUsbTaskState() == USB_STATE_RUNNING) && startDelay && (startDelay <= millis())) {
Hid.SetReport(0, 0, 3, 0xf4, 4, feature_F4_report);
startDelay = 0;
}
}

mashedkeyboard, 2020-11-15:
Did you by any chance get anywhere with this? I'm trying to make the same gamepad work, with little success.

hebeallrusty, 2021-02-20:
I managed to get this joystick working on Retropie on a Pi 4 however at first the analog sticks would mirror the d pad and the buttons on the right. Using jstest I could see there were buttons that the left analogue stick axis were either plus or minus fully on when active. By chance I pressed the big circle button on the centre of joystick so that two lights lit up on the controller instead of one, and everything worked from there. Reading the manual, the button was meant to change the frequency with the RF dongle, but it seems to activate full analogue (with a range of values on the analogue stick instead of fully on or off).