felis/USB_Host_Shield_2.0 #440: Broadcom BCM20702 not working

rofearg, 2019-01-21:
I'm trying use a dongle with the Broadcom BCM20702 chipset in it. Is this chipset supported yet?

Sample dongle sourced from here:
https://www.amazon.co.uk/gp/product/B007MKMJGO/ref=ppx_yo_dt_b_asin_title_o02__o00_s00?ie=UTF8&psc=1

Lauszus, 2019-01-21:
@rofearg please enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging. What does it print when you plug it in?

rofearg, 2019-01-22:
> 
> 
> @rofearg please enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging. What does it print when you plug it in?

I've done this but there is no output beyond "PS4 Bluetooth Library Started" or "HID Bluetooth Library Started"
The led in the dongles does not light up when I use the PS4BT or BTHID examples, but does if I use the USBHIDBootMouse example the led switches on and I see some data being returned.

Should I have seen more debug after switching on #define ENABLE_UHS_DEBUGGING 1

Lauszus, 2019-02-04:
@rofearg yes you should data being printed to the screen. Please try to enable extra debugging as well: https://github.com/felis/USB_Host_Shield_2.0/blob/148cf2629feade642070fbcd8182ff2eb0518fbb/BTD.cpp#L20