felis/USB_Host_Shield_2.0 #444: Sending commands to an Unihedron SQM device

rtohme, 2019-01-31:
I'm trying to send serial commands and read the response from a SQM-LU device from Unihedron (http://www.unihedron.com/projects/sqm-lu). It's a photometer with a USB port that uses the FT232R chipset. I assumed using the Arduino USB Host Shield would allow me to do that, however, I'm using the code in your FTDI loopback example as a base, and I can't get Usb.getUsbTaskState() to return USB_STATE_RUNNING. Instead, it returns a number: it starts at 32 and goes to 104 quickly, then it goes down to 18 when I disconnect the SQM-LU device from the USB Host Shield, but it returns to 104 when I plug it back. I've run the board_qc sketch and my shield seems to be ok. What can be happening?

gdsports, 2019-01-31:
What is the PID of the device? The only one accepted is 0x6001. But other PID values can be passed on the constructor.

rtohme, 2019-02-04:
> What is the PID of the device? The only one accepted is 0x6001. But other PID values can be passed on the constructor.

Here is a dump from "dmesg" on Linux after connecting the SQM-LU to a computer. As you can see in the second line, PID is indeed 0x6001:

[Feb 4 10:54] usb 8-1: new full-speed USB device number 34 using uhci_hcd
[  +0.209982] usb 8-1: New USB device found, idVendor=0403, idProduct=6001
[  +0.000003] usb 8-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000003] usb 8-1: Product: MM232R USB MODULE
[  +0.000002] usb 8-1: Manufacturer: FTDI
[  +0.000002] usb 8-1: SerialNumber: FTF4FO08
[  +0.007061] ftdi_sio 8-1:1.0: FTDI USB Serial Device converter detected
[  +0.000071] usb 8-1: Detected FT232RL
[  +0.002012] usb 8-1: FTDI USB Serial Device converter now attached to ttyUSB1

rtohme, 2019-02-04:
> @rtohme please also enable serial debugging for the library: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging, so you can tell what is going on.

I get this when I connect the SQM device to the USB Host Shield:

FTDI Init
Addr:01
NC:01
Conf.Val: 01
Iface Num: 00
Alt.Set: 00
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
Conf.Val: 01
Iface Num: 00
Alt.Set: 00
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
NumEP:03
Conf:01
baud_value:001A
baud_index:0000
FTDI configured