felis/USB_Host_Shield_2.0 #447: usbhost + esp8266

ihebslimen, 2019-02-05:
My project is about getting data from usb shield then send it to a firebase database using esp8266 arduino code .
but when i try to compile i got many compiling issues that are related to the library (see below)
can you help me make usbhost library compatible with  esp8266 arduino core.
thanx
**erreurs**
In file included from C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/settings.h:154:0,

                 from C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/Usb.h:34,

                 from C:\Arduino\arduino-1.8.4\libraries\glucoduino\examples\glucoduino_read\glucoduino_read.ino:15:

C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/SPI.h: In static member function 'static byte SPIClass::transfer(byte)':

C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/SPI.h:56:3: error: 'SPDR' was not declared in this scope

   SPDR = _data;

   ^

C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/SPI.h:57:12: error: 'SPSR' was not declared in this scope

   while (!(SPSR & _BV(SPIF)))

            ^

In file included from C:\Users\CD-HOUSE\AppData\Local\Temp\arduino_build_104986\sketch\glucoduino_read.ino.cpp:1:0:

C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/SPI.h:57:23: error: 'SPIF' was not declared in this scope

   while (!(SPSR & _BV(SPIF)))

                       ^

C:\Users\CD-HOUSE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266/Arduino.h:189:25: note: in definition of macro '_BV'

 #define _BV(b) (1UL << (b))

                         ^

In file included from C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/settings.h:154:0,

                 from C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/Usb.h:34,

                 from C:\Arduino\arduino-1.8.4\libraries\glucoduino\examples\glucoduino_read\glucoduino_read.ino:15:

C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/SPI.h: In static member function 'static void SPIClass::attachInterrupt()':

C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/SPI.h:63:3: error: 'SPCR' was not declared in this scope

   SPCR |= _BV(SPIE);

   ^

In file included from C:\Users\CD-HOUSE\AppData\Local\Temp\arduino_build_104986\sketch\glucoduino_read.ino.cpp:1:0:

C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/SPI.h:63:15: error: 'SPIE' was not declared in this scope

   SPCR |= _BV(SPIE);

               ^

C:\Users\CD-HOUSE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266/Arduino.h:189:25: note: in definition of macro '_BV'

 #define _BV(b) (1UL << (b))

                         ^

In file included from C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/settings.h:154:0,

                 from C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/Usb.h:34,

                 from C:\Arduino\arduino-1.8.4\libraries\glucoduino\examples\glucoduino_read\glucoduino_read.ino:15:

C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/SPI.h: In static member function 'static void SPIClass::detachInterrupt()':

C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/SPI.h:67:3: error: 'SPCR' was not declared in this scope

   SPCR &= ~_BV(SPIE);

   ^

In file included from C:\Users\CD-HOUSE\AppData\Local\Temp\arduino_build_104986\sketch\glucoduino_read.ino.cpp:1:0:

C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master/SPI.h:67:16: error: 'SPIE' was not declared in this scope

   SPCR &= ~_BV(SPIE);

                ^

C:\Users\CD-HOUSE\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266/Arduino.h:189:25: note: in definition of macro '_BV'

 #define _BV(b) (1UL << (b))

                         ^

Utilisation de la bibliothèque USB_Host_Shield_2.0-master version 1.3.2 dans le dossier: C:\Arduino\arduino-1.8.4\libraries\USB_Host_Shield_2.0-master 
Utilisation de la bibliothèque glucoduino prise dans le dossier : C:\Arduino\arduino-1.8.4\libraries\glucoduino (legacy)
exit status 1
Erreur de compilation pour la carte Adafruit Feather HUZZAH ESP8266

Lauszus, 2019-02-06:
@ihebslimen please update your hardware plugin, as it seems to be quite old.

MMushovic, 2019-02-06:
> please update your hardware plugin, as it seems to be quite old.

Could you provide more details on how to do this?  I am seeing the same/similar compilation errors. How does one update their hardware plugin... or even know what "version" they have installed...?

Thank you in advance for the support!

gdsports, 2019-02-06:
@MMushovic To install the ESP8266 hardware board package, see the instructions at https://github.com/esp8266/arduino

But downgrade the version to 2.4.2 for now. There is a problem with the latest version 2.5.0.

MMushovic, 2019-02-07:
OK, I'll give that a try tonight...  and thank you for the reply.

Is email the best way to maintain a discussion with you (assuming you are
interested in that and/or my project :^)

MMushovic, 2019-02-18:
Update:  worked great, thank you

Board and project are working great!

Thank you for your help

Email sent via iPhone- please excuse brevity & errors

closed 2019-02-28 by Lauszus