felis/USB_Host_Shield_2.0 #454: Library files for Epson LQ-310 Thermal Printer

Vamshi253, 2019-03-18:
Hi guys,

I am looking to drive a Epson LQ-310 thermal printer with Arduino Mega ADK through USB. Are there any library files which are compatible for this application.

I connected printer to Mega ADK and obtained the following descriptor parameters:
`Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0110
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	08
Vendor  ID:		04B8
Product ID:		0047
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		0020
Num.intf:		01
Conf.value:		01
Conf.string:		00
Attr.:			C0
Max.pwr:		01

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		07
Intf. Subclass:		01
Intf. Protocol:		02
Intf.string:		00

Endpoint descriptor:
Endpoint address:	01
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	82
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)
`
Could anyone please help me regarding this on how to achieve this?

Thank You.

xxxajk, 2019-03-18:
Not even showing a product for that VID:PID except one, and that is referencing an impact dot matrix printer, specifically the Epson LQ-350.
Thermal printers are normally for Point of Sale. Is that what this is for?

Vamshi253, 2019-03-18:
Oh I am sorry, it was my mistake.
It's actually a dot matrix printer.

xxxajk, 2019-03-18:
I won't be able to look into this at all until next week. 
I know for sure I can implement this in UHS3 without any problems. UHS2, while could be made to work, probabbly won't work as easily, or as well. Don't expect to be able to do much with the limited RAM on the atmel mega chips. Text should be OK, graphics-- not so much, since it can take quite a bit of memory to buffer. Rasters on esc P printers are chopped into vertical strips, instead of horizontal, which means if you are going to do graphics on a MEGA, you would need to either have something pre-processed, or be able to generate a rather large array rotated from the original.

This is how much memory you would need in order to do graphics rotation just for the raster array, and does not include any stream buffering, USB packet buffering, or device house-keeping:

180DPI will take 2880 bytes of RAM. (more than a quarter of available memory)
360DPI will take 17280 bytes of RAM. (won't fit even with tricks, 2560 only has 8K RAM)

You *COULD* do a few tricks to lower the memory requirements and be able to use the horrible 180DPI, but the problem is that it would use more flash, and have very complicated code to do it. On older computers the code to do this always had various issues, and would actually slow down the printing process some times. Yup, Imagine that, a printer that could out-pace the pc.
Something pre-processed in FLASH could use flash up very quickly too.
You are way better off getting a Teensy 3.6, use the native USB port on it, and already supported on UHS3. It has abundant resources:
32 bit 180 MHz ARM Cortex-M4 processor with floating point unit
1M flash 256K RAM
$29.25
if doing graphics, and I highly recommend it. I don't bother with the AVR platform for any new projects anymore. The ARM offerings are less expensive and offer more.

Vamshi253, 2019-03-18:
Thank you very much for the detailed clarification.
Actually, I am not into graphics printing.
I just want to print the text that is coming to the serial port of Arduino.

Vamshi253, 2019-03-20:
Hi,

Based on adk header files in the library, I am developing the library files for printer.
Can any one of you suggest me regarding on what to assign for bAddress, bConfNum, bNumEP considering printer?

Thank You.

sudha100, 2019-06-13:
hi vamshi iwat basic code for usb dot matrix printer...

and the arduino must and should interface with prnter using usb cable only