felis/USB_Host_Shield_2.0 #459: Replacement for Usb.setReport()?

JoeMalt, 2019-04-26:
I'm trying to toggle the state of some LEDs (Caps Lock indicator etc.) on a USB keyboard.

An [example](http://webcache.googleusercontent.com/search?client=ubuntu&channel=fs&q=cache%3Ahttps%3A%2F%2Fwww.circuitsathome.com%2Fmcu%2Fhow-to-drive-usb-keyboard-from-arduino&ie=utf-8&oe=utf-8) (linked to cached version since site seems to be down) calls `Usb.setReport()`, but this method doesn't seem to exist.

There seems to be a SetReport() method in usbhid.cpp, I'm trying to use that instead (after creating a USBHID object), but I'm not sure if it's a replacement or just something with a similar name.

Am I on the right track?

DanielGibson, 2021-01-01:
(I know this is late, but maybe you still wonder how this works or someone gets here via google or wahtever)
Take a look at https://github.com/felis/USB_Host_Shield_2.0/blob/08121e1701908006cd646ad51b8c93cf7fe4ce7a/hidboot.h#L151-L172 to see how to set the keyboard LEDs.
Note that this doesn't work reliably with the "current" release (1.3.2 which is almost 3 years old now) you'd get from the Arduino Library Manager, but it does work with the latest git code (see https://github.com/felis/USB_Host_Shield_2.0#manual-installation)

Lauszus, 2021-02-07:
This has been fixed with the 1.4.0 release.

closed 2021-02-07 by Lauszus