felis/USB_Host_Shield_2.0 #464: DATALOGIC scanner does not initialize

WRelihan, 2019-06-02:
Hi
I am trying to use a DATALOGIC Quickscan Lite QW2420 barcode/QR scanner to work with the USB host shield.  I am using a ESP32 for the micro.
The setup works fine with a USB keyboard.
The scanner works fine with a PC.  I can set it up to work as either a keyboard or a comport.
The problem is that the scanner does not initialize. I have put a scope on the D+ and D- signals and I see that the 1ms data frames is present.  The comms seems to be around 5Mhz for a single bit.
The waveform does seem somewhat distorted. I dont know if this could be a problem.
[image]
I have also checked with a thumbdrive, which operates at the same speed and the waveforms looks a lot better:
[image]

Is it possible that the max 3421 is sensitive to this type of distortion?  I got the scanner to work a few times with USB_desc.  And therereafter with USBHID_desc it actually made the beep sounds when the scanner is initialized.  I could then proceed to start scanning.  But I was just starting to do some code for parsing the input when it all stopped.  I could not get anywhere with it again.

I  changed USB_desc to print the USB_state as it powers up.
I get the following:
0x12  - Detached waiting for device
0x20  -Attached - substate settle
0x40 - Attached - substate wait reset complete
0x51 - Attached - substate wait reset
and then
0xA0 - USB state error

I have traced this error to USB::Configuring in Usb.cpp.  There is a printf that was commented out which I re-enabled and  changed to print the rcode ( around line731):
///////////// 
if(rcode) {
                printf("Configuring error: Can't get USB_DEVICE_DESCRIPTOR:  0x%02X \r\n",rcode);
                
                return rcode;
        }
////////////

This prints the message:
                    Configuring error: Can't get USB_DEVICE_DESCRIPTOR:  0x0D 

I am at a loss what to do next.  I think that the 5m cable that comes with the scanner is maybe to long for the data rate, and that is why the wave-forms are distorted, but it does work with my PC.

I am prepared to buy another brand of barcode/qrscanner if someone can recommend one to me that works.

PS:  THANKS for this great library.  I am struggling to understand it as I am new to object orientated programming and classes etc.  But by digging into this library I have already learned a lot. 

Regards

Willem

xxxajk, 2019-06-02:
Are you using USB power only? If so, use a 9VDC adapter instead.

WRelihan, 2019-06-02:
No. I am supplying 5vdc to the usb_host shield via a 12 to 5v dc/dc. I also
monitored the 5v on the usb plug and it is steady at 5v.
Thanks for your help.

foxabilo, 2019-07-13:
What kind of output filtering have you got on that dc/dc noisemaker?

sylvioalves, 2019-12-20:
@WRelihan, have you found a solution for this issue?