felis/USB_Host_Shield_2.0 #468: display via gpout (max_lcd example)

lokkiikkol, 2019-06-28:
hi,

i try to use the usb-host shield with an 20x4 HD44780 display. the same display works fine with the liquidcrystal example. when i compile and run the max_LCD example from the usb host library, i can only see solid squares on line 1 and 3 of the display. should 20x4 displays be supported by the max_LCD library? any hints on working this out? (i have the connections as per the text in max_LCD.ino (all hooked up to GPOUT)

Lauszus, 2019-07-02:
Please make sure that you have set the number of columns and rows like so:

```C++
lcd.begin(20, 4);
```

Also try using a different ```charsize```: https://github.com/felis/USB_Host_Shield_2.0/blob/master/max_LCD.h#L71.

lokkiikkol, 2019-07-02:
thanks for your suggestions, i already tried with the lcd.begin(20, 4); no difference. i will try a different charsize and report back. the sparse info i found on the archive.org site (since circuitsathome is down) suggests a 3.3v lcd display. mine is 5v, but the 3.3v from the gpout should still suffice to read as logic high...

lokkiikkol, 2019-07-03:
i tried with those suggestions and could not get it to work correctly. (only squares displayed) i have now switched to a shift register approach, so i only need 3 arduino pins to drive the hd44780. i moved some buttons to the gpin section of the usb-host shield and all works...