felis/USB_Host_Shield_2.0 #472: confused using PS4.setLed

Redfearn, 2019-07-01:
Hello, 

I'm using a PS4 controller connected via bluetooth to control a stepper motor.

I want to use setLed to indicate when a particular operation is happening, e.g. turn stepper motor 180 deg clockwise.

I have written my code as follows:

```
// all the usual stuff

void loop() {
  Usb.Task();

  if (PS4.connected()) {

    PS4.setLed(White); // to indicate safe mode

    if (PS4.getButtonClick(CROSS)) {
      PS4.setLed(Red); // Warning, motors running
      delay(2000); // to simulate running a funciton that take 2 s
    }
  }
}
```

Everything works except the LED change. It seems to be an issue with the first LED color change, if i remove that then the LED does change to red, however this happens after the delay ...

After fiddling about and looking at the provided example PS4BT the same thing happens if I put an initial colour change in the example code, so I must be missing something.

Thanks for any help,

Andrew

Lauszus, 2019-07-02:
@Redfearn please make sure that your code is not blocking anywhere i.e. do not use ```delay```. Please see: https://github.com/felis/USB_Host_Shield_2.0/issues/339#issuecomment-356759315.

Redfearn, 2019-07-02:
@Lauszus Thanks for your quick response. The function I intend to call by pressing the CROSS button contains delays (it is sending a pulse to a stepper driver at regular intervals in order to move it a set distance) so maybe it's not possible to do what I want if the Usb.Task function needs to be called at regular intervals.

Lauszus, 2019-07-02:
@Redfearn and it's not possible to rewrite the code to use micros/millis and a state machine instead of delays?

Redfearn, 2019-07-02:
Not sure, here is the code with one of the functions I would like to run:

```
void loop() {
  
  Usb.Task();
  PS4.setLed(Green);

  
  if (PS4.getButtonClick(CIRCLE)) {
    PS4.setLed(Red);
  }

}

// movement function // INPUTS: pulse per rev, movement dist in mm, speed in mm/s (27 max)

void movement(int PPR, float dst1, float duration) {

  //float spd1 = dst1 / duration; // converts duration input to speed based on tongue in rig
  pls1a = (dst1 / 8) * 1.2 * PPR; // 1.2 is the gear ratio
  pls1 = abs((int) pls1a);
  dly1 = (60 / ((abs(dst1 / (duration / 1000)) / 8) * 1.2 * PPR * 0.00012)) - 10; // 10 microsecond to compensate for time spent doing digital write

  if (dst1 < 0) // Works out movement direction
  {
    digitalWrite(dirPin, LOW);
  } else {
    digitalWrite(dirPin, HIGH);
  }

  for (int i = 0; i < pls1; i++) // Moves stepper
  {
    digitalWrite(stepPin, HIGH);
    delayMicroseconds(dly1);
    digitalWrite(stepPin, LOW);
    delayMicroseconds(dly1);
  }

  digitalWrite(stepPin, LOW); // Puts motor into stationary holding mode
}
```

The movement function can be running for a few seconds depending on the chosen speed. That's why I was simulating it with a delay.