felis/USB_Host_Shield_2.0 #475: HSPI instead of VSPI on ESP32

foxabilo, 2019-07-08:
Hi all,

I wanted to change over the SPI pins from VSPI to HSPI on the ESP32, I had a look at a guide on the ESP8266 and the chap there changed a few pins around in usbcore.h my change requires a bit more than that so I changed the pin definitions in avrpins.h (see below) but this has not done the trick, it's still on the VSPI bus pins and as a small aside, the USB.Init() picks my SD card up (on VSPI) as a valid MAX3421Edevice and passes the Init check(minor point but worthy of note), I'll keep on plugging at it to refine the pins any help gratefully appreciated. 

```
#elif defined(ESP32)

// Workaround strict-aliasing warnings
#ifdef pgm_read_word
#undef pgm_read_word
#endif
#ifdef pgm_read_dword
#undef pgm_read_dword
#endif
#ifdef  pgm_read_float
#undef pgm_read_float
#endif
#ifdef  pgm_read_ptr
#undef pgm_read_ptr
#endif

#define pgm_read_word(addr) ({ \
  typeof(addr) _addr = (addr); \
  *(const unsigned short *)(_addr); \
})
#define pgm_read_dword(addr) ({ \
  typeof(addr) _addr = (addr); \
  *(const unsigned long *)(_addr); \
})
#define pgm_read_float(addr) ({ \
  typeof(addr) _addr = (addr); \
  *(const float *)(_addr); \
})
#define pgm_read_ptr(addr) ({ \
  typeof(addr) _addr = (addr); \
  *(void * const *)(_addr); \
})

// Pinout for ESP32 dev module
MAKE_PIN(P0, 0);
MAKE_PIN(P1, 1); // TX0
MAKE_PIN(P10, 10); // TX1
MAKE_PIN(P3, 3); // RX0
MAKE_PIN(P21, 21); // SDA
MAKE_PIN(P22, 22); // SCL
MAKE_PIN(P19, 12); // MISO HSPI
MAKE_PIN(P23, 13); // MOSI HSPI
MAKE_PIN(P18, 14); // SCK HSPI
MAKE_PIN(P5, 15); // SS HSPI
MAKE_PIN(P17, 17); // INT

#endif
```

foxabilo, 2019-07-09:
Actually, this issue is that it is using the default SPI object and that gets redefined by the SD card library when it starts using the same default object, so I think just creating a new SPI class instance will be the solution, I'll post back with the results.

xxxajk, 2019-07-09:
I will soon be adding support for esp8266/esp32 on UHS3. 
It should be able to avoid these issues. 
Bonus: UHS3 already supports SD cards :-)

foxabilo, 2019-07-09:
Sounds interesting, I think my main issue is trying to create two SPI devices running one on HSPI pins and one of HSPI pins, SD card on one, USB HOST on the other, as the USB host can speak at any time I need to use completely separate pins for each, just struggling to work out the correct syntax to create two SPI objects and get each library to use them

closed 2019-07-11 by foxabilo

suremicro, 2020-09-09:
@foxabilo Did you manage to het the HSPI pins working? I have made the same changes but cannot get to init (switching back the pins immediately works). Thanks

foxabilo, 2020-09-09:
[sd-spi-lib.zip](https://github.com/felis/USB_Host_Shield_2.0/files/5195925/sd-spi-lib.zip)
I had to change a bunch of pin defines and change some stuff around, don't quite remember exactly what I did but it did end up working, I've included a copy of my SD and SPI modified libs in this post, have at em. I took a quick look and I'd need a week to get back into it.

suremicro, 2020-09-09:
Thank you for getting back so quickly. That'll give me a good idea of what I've missed!

foxabilo, 2020-09-09:
if (SD.begin(4, 15000000)) // SD Card speed , should be calculated from tests?
	{						   //,SD_SCK_MHZ(15))) {
		Serial.println("SD Card initialized.");
		hasSD = true;
	}

is my init call to the SD lib and you can ignore the vspi folder it's not used

Good luck

foxabilo, 2020-09-09:
Hmmm, well in this case I'm using VSPI for the USB controller, is that what you are trying to do?

suremicro, 2020-09-09:
Ah no, I'm trying to use the HSPI with the host controller - the reason being I'd like it to work with an ESP32-CAM module that only has those pins.

foxabilo, 2020-09-09:
so in your setup do you have a 

`hSPI = new SPIClass(HSPI);`

foxabilo, 2020-09-09:
the only difference I can find is in usbhost.h 

[image]

foxabilo, 2020-09-09:
Something tickles the back of my mind about me adding those pin numbers to the USB_SPI.begin()

suremicro, 2020-09-09:
Bingo! The original library doesn't have any pins defined, so the constants are only for the USB library. Many thanks ;)

foxabilo, 2020-09-09:
Awesome!