felis/USB_Host_Shield_2.0 #476: two USB HOST working on same ESP32

fotosettore, 2019-07-12:
hi
i'm using a ESP32.
i'd like to install two USB HOST device on the same esp32 --> https://i.ebayimg.com/images/g/UUQAAOSwFyhaF6Pj/s-l1600.jpg

I saw that, as default, usbh_midi refers to VSPI (18,19,23,5)
what i must do to use HSPI (14,12,13,15) also (in case it were possible to make them work together), or better more add another SS to use two usb host on same line (VSPI or HSPI) ?

here the code that works fine on VSPI with only one usb host shield
``` 
#include <usbh_midi.h>
USB Usb;
USBH_MIDI  Midi(&Usb);

//****
void setup() 
{
  Serial.begin(115200);
  if (Usb.Init() == -1) 
  {
    Serial.println("USB did not start.");
    while(1);
  }  
  else
  {
    Serial.println("Started USB :-)");
  }

/*
 ---> here i'd need to start another usb host  (ex.  usb_2) 
*/
}

```

i think i must refer to **USB_MIDI_converter_multi.ino** but i did not understand how to connect the pin SS of second usb host.
I have been looking for an example for two days but found nothing.

many thanks for help

gdsports, 2019-07-12:
Maybe use a USB hub for more than USB MIDI devices. But you may need the fixes in PR #414 for hubs to work.

fotosettore, 2019-07-13:
thanks for your reply
i made all modifications but i'm not skilled to understand what kind of code i must add in my sketch to obtain the second usb host.
may you help me ?
thanks

fotosettore, 2019-07-22:
no info, no help ?

xxxajk, 2019-07-22:
Not practical with this library, I still need to support espressif MCU in
uhs3, which can do more than 1...

fotosettore, 2019-07-22:
thanks ... i will wait and i will try to understand better
it should be very simple. we'd must change the SS pin
for example in the UsbCore.h i try to change this line
`#elif defined(ESP8266)
typedef MAX3421e<P2, P5> MAX3421E; // ESP8266 boards      //P4=D2    P0=D3   P2=D4 `
and it works fine if i change first P
but i must to define two P, one for first host and another for second one.

xxxajk, 2019-07-22:
Just wait for UHS3 to do this, as you won't have to worry so much on what to do.
I'll try to get to it this weekend... No promises though, I am quite busy.

fotosettore, 2019-07-22:
really really many thanks ...

CodeOhms, 2019-07-31:
If this has not been implemented when I get two shields in a month, I will try to help get this done.

fotosettore, 2019-09-17:
hi everyone ...
no news about this problem ?

fotosettore, 2020-04-09:
i think i have no hope for this issue :-(

fusionstream, 2020-05-02:
Duplicate the library and change the pins as you have already done. Use the "different" library to control the other host.