felis/USB_Host_Shield_2.0 #478: Error in Aruino IDE but not on WEB Creator!?!?

TDuxis, 2019-07-15:
Hi

I am running an Arduino Mega ADK rev3d.

1.
I am trying this library out. I noticed that when I run it using Arduino IDE on my PC, I get OSCOKIRQ error, but when I run it from the Arduino Creator Web interface, I do not get this problem. Tried several of the examples, same thing on all.

Using latest version 1.3.2 in both cases. What could be the problem?

2. When trying acm_terminal out, I notice it hangs on acm.isReady(). Any suggestions on what could be the problem here? I am trying to connect to a Huawei 3G USB modem. Hub_demo works nice and reports back Huawei as the manufacturer. So I know the USB connection is working ok in that sense.

Thanks!

gdsports, 2019-07-15:
I do not know if I can help but it would be helpful for you to post the output of USB_desc. Also turn [debugging](https://github.com/felis/USB_Host_Shield_2.0#enable-debugging) and post the output.

TDuxis, 2019-07-16:
Hi

Here's the output from Hub_demo, see below.

I noticed that when running acm_terminal, inside ACM.init, pAsync->OnInit(this); returns 5.

rcode = pAsync->OnInit(this);

I will turn debugging on and get back with more info.

`Start

01
--
String Descriptors:
Manufacturer:		Huawei Technologies
Product:		HUAWEI Mobile

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		12D1
Product ID:		1446
Revision ID:		0000
Mfg.string index:	03
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0037
Num.intf:		02
Conf.value:		01
Conf.string:		01
Attr.:			C0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		08
Intf. Subclass:		06
Intf. Protocol:		50
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	01
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		08
Intf. Subclass:		06
Intf. Protocol:		50
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	82
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)
`

gdsports, 2019-07-16:
The device look like a mass storage device. If you plug it into a Linux box such as a raspberry pi, you will see it looks like a USB hard drive or CDROM drive. Usually the drive contains Windows and Mac drivers for the modem. The Linux community has figured out how to switch these kinds of devices into modem mode without Windows drivers. This is a complex device so I would not consider it for use with Arduino.

```
Intf. Class: 08         Mass Storage
Intf. Subclass: 06      SCSI transparent command set
Intf. Protocol: 50      BBB (Bulk only)

```

TDuxis, 2019-07-16:
Thanks for the info!

Setting the modem mode is done via AT commands, at least that's the only info I've found, and I know how to do that. I've also sent SMS by AT commands, but not via the arduino, yet :)

So, right now the focus is on how get the arduino to talk to the Huawei device in terminal mode. I've seen others use acm_terminal to send AT commands to Huawei mobile phones, without any changes made to the acm-terminal SW. So I figure it should be more or less the same with a modem. It's basically a phone without the HMI ;)

TDuxis, 2019-07-16:
Ok, so I just noticed that pacm->SetControlLineState(3); returns 5, which is an error. What does that mean?

`uint8_t ACMAsyncOper::OnInit(ACM *pacm)
{
    uint8_t rcode;
    // Set DTR = 1 RTS=1
    rcode = pacm->SetControlLineState(3);`

xxxajk, 2019-07-16:
Can you try this device on UHS3? It might be able to enumerate it.

TDuxis, 2019-07-16:
Sorry, but what is UHS3? :)

xxxajk, 2019-07-16:
Version 3.0 of this library.

TDuxis, 2019-07-16:
Whoops, sorry, of course it is :)  I wasn't aware of its existense. Will try right away. Thanks.

xxxajk, 2019-07-16:
Should work on mega adk without modifications. If it's showing up as mass
storage, include the mass storage driver. You can also send whatever you
need to endpoint 0 to switch it to the different mode.

TDuxis, 2019-07-16:
Ok. How do I send something to endpoint 0?

But still, as far as I understand, its the setup of the arduino that I am not getting through with in the acm_terminal, I never get "ok" on acm.isReady, and it seems its the setup of DTS/RTS that is not ok. I commented it out and then the same thing on Line Coding. So even if I switch mode, I will still have the same issue with acm, won't I?

TDuxis, 2019-07-17:
Btw, I have set the unit to Modem + PCUI mode only (AT^SETPORT="A1,A1;1,2")

But still, this shouldn't be causing the issue with the acm-terminal as described above, right?

xxxajk, 2019-07-17:
Discuss on UHS3. I need more details from you.

TDuxis, 2019-07-17:
Here's the output in the Serial window with Debug turned on. Rows with "TRN" in them are extras I added. Plus I commented out the part with No endpoints as mentioned in another issue. Otherwise that will cause a fail as there's only 1, but required to be at least 4.

`Start

USB Init done!
ACM Init
Addr:01
NC:01
1 = bNumEP TRN
Conf:00
SetControlLineState TRN
SetControlLineState: 05

SetControlLineState TRN rcode = 5
OnInit:05`

xxxajk, 2019-07-17:
https://github.com/felis/UHS30/issues go here, open a new issue.

TDuxis, 2019-07-17:
Sorry, it seemed to be a bit of manual work to get UHS3 so I am still on 2.0
Will try UHS3 later today and swap over.

But I am still wondering if the issue with SetControlLineState could be due to the USB device, or if it is an Arduino issue?

gdsports, 2019-07-17:
The USB descriptor dump shows the 3G device is in USB mass storage mode so the ACM driver is doing nothing except ignoring the device.

TDuxis, 2019-07-17:
Aha, thank you! 👍
Which modes does acm accept?
I have switched the modem to modem only mode. Don't know why it still reports as mass storage.

xxxajk, 2019-07-17:
Because of how UHS2 enumerates, that's why. It can't handle complex enumeration. 
UHS3 can enumerate complex devices easily.

The problem is that UHS2 sees what is plugged in as a single item. If the first descriptor matches, the remaining ones are ignored.

UHS3 doesn't do this shortcut. Instead each descriptor is examined for a match, and can do multiple matches.

TDuxis, 2019-07-18:
Hmmm, so, if I figure out which endpoint is the serial terminal for the modem, then I could actually hardcode the right parameters for the CtrlReq's in the library, right?

TDuxis, 2019-07-18:
Ok, so now I've setup the USB Modem so that the serial terminal on the modem is the one and only interface available, interface nr 0. It's got two endpoints 81 and 01. The PID is 1506 which is Huaweis PID for USB Modem. The interface class is FF, which means vendor specific.

I am still getting rcode 5 on SetControlLineState, see below.

Is this because the interface class is FF?
How can I change the code so it does not get hung up on the interface class?

Thanks!

USB_Desc  output
------------------------
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		12D1
Product ID:		1506
Revision ID:		0000
Mfg.string index:	03
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0020
Num.intf:		01
Conf.value:		01
Conf.string:		01
Attr.:			C0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		02
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			02
Max.pkt size:		0040
Polling interval:	20

Endpoint descriptor:
Endpoint address:	01
Attr.:			02
Max.pkt size:		0040
Polling interval:	20

Addr:1(0.0.1)
------------------------

acm_terminal output
-------------------------
Start

USB Init done!
ACM Init
Addr:01
NC:01
1 = bNumEP TRN
Conf:00
SetControlLineState TRN
SetControlLineState: 05

SetControlLineState TRN rcode = 5
OnInit:05
---------------------

TDuxis, 2019-07-19:
> 
> 
> Hmmm, so, if I figure out which interface/endpoint is the serial terminal for the modem, 
> then I could actually hardcode the right parameters for the CtrlReq's in the library, right?
>

Yes, and that's exactly how I solved it :)
But using UHS3 instead. However, 2.0 would probably have worked well too.

closed 2019-07-19 by TDuxis