felis/USB_Host_Shield_2.0 #480: Issue with chinese wireless USB Controller

gneraudeau, 2019-07-20:
Hello, 

I recently bought a wireless Chinese controller with dongle (maybe RF ou BT)

[image]

I correctly installed USB Host shield with Manual, I uploaded many examples on the board, but I have only return (bad) with XBOXUSB.ino I can see some pressed buttons but, even if I don’t press any button, I have response like this:

[image]

I tried many other examples but no response from the others.

I tried USB_des.ino and it returned me this:

14:49:21.563 -> St	02
14:49:21.563 -> Attr.:			03
14:49:21.563 -> Max.pkt size:		0020
14:49:21.563 -> Polling interval:	08
14:49:21.563 -> 
14:49:21.563 -> 
14:49:21.563 -> Addr:1(0.0.1)
14:49:21.563 -> l:	04
14:49:21.563 -> 
14:49:21.563 -> Endpoint descriptor:
14:49:21.563 -> Endpoint address:	02
14:49:21.563 -> Attr.:			03
14:49:21.563 -> Max.pkt size:		0020
14:49:21.563 -> Polling interval:	08
14:49:21.563 -> 
14:49:21.563 -> 
14:49:21.563 -> Addr:1(0.0.1)
14:49:21.563 -> l:	04
14:49:21.563 -> 
14:49:21.563 -> Endpoint descriptor:
14:49:21.563 -> Endpoint address:	02
14:49:21.563 -> Attr.:			03
14:49:21.563 -> Max.pkt size:		0020
14:49:21.563 -> Polling interval:	08
14:49:21.563 -> 
14:49:21.563 -> 
14:49:21.563 -> Addr:1(0.0.1)
14:49:21.563 -> Start
14:49:22.330 -> 
14:49:22.330 -> 
14:49:22.330 -> 01
14:49:22.330 -> --
14:49:22.330 -> 
14:49:22.330 -> Device descriptor: 
14:49:22.330 -> Descriptor Length:	12
14:49:22.330 -> Descriptor type:	01
14:49:22.330 -> USB version:		0200
14:49:22.330 -> Device class:		FF
14:49:22.330 -> Device Subclass:	FF
14:49:22.366 -> Device Protocol:	FF
14:49:22.366 -> Max.packet size:	08
14:49:22.366 -> Vendor  ID:		045E
14:49:22.366 -> Product ID:		028E
14:49:22.366 -> Revision ID:		0110
14:49:22.366 -> Mfg.string index:	01
14:49:22.366 -> Prod.string index:	02
14:49:22.366 -> Serial number index:	03
14:49:22.366 -> Number of conf.:	01
14:49:22.366 -> 
14:49:22.366 -> Configuration descriptor:
14:49:22.366 -> Total length:		0030
14:49:22.366 -> Num.intf:		01
14:49:22.366 -> Conf.value:		01
14:49:22.366 -> Conf.string:		00
14:49:22.366 -> Attr.:			80
14:49:22.366 -> Max.pwr:		FA
14:49:22.366 -> 
14:49:22.366 -> Interface descriptor:
14:49:22.366 -> Intf.number:		00
14:49:22.366 -> Alt.:			00
14:49:22.366 -> Endpoints:		02
14:49:22.366 -> Intf. Class:		FF
14:49:22.366 -> Intf. Subclass:		5D
14:49:22.403 -> Intf. Protocol:		01
14:49:22.403 -> Intf.string:		00
14:49:22.403 -> Unknown descriptor:
14:49:22.403 -> Length:		10
14:49:22.403 -> Type:		21
14:49:22.403 -> Contents:	10010124811403000313020003000705
14:49:22.403 -> 
14:49:22.403 -> Endpoint descriptor:
14:49:22.403 -> Endpoint address:	81
14:49:22.403 -> Attr.:			03
14:49:22.403 -> Max.pkt size:		0020
14:49:22.403 -> Polling interval:	04
14:49:22.403 -> 
14:49:22.403 -> Endpoint descriptor:
14:49:22.403 -> Endpoint address:	02
14:49:22.403 -> Attr.:			03
14:49:22.403 -> Max.pkt size:		0020
14:49:22.403 -> Polling interval:	08
14:49:22.403 -> 
14:49:22.403 -> 
14:49:22.403 -> Addr:1(0.0.1)

Thank you very much for your help

gdsports, 2019-07-20:
That looks like a PS3 controller, not Xbox. Try the USB PS3 or PS4 examples to determine which controller you have. Then try the wireless device.

gneraudeau, 2019-07-22:
Hello, 

I tried with "PS3USB0.ino", no reaction:
[image]

I tried with "PS4USB.ino", no reaction too:
[image]
 
The only way to have a reaction (bad), it's with "XBOXUSB.ino"

For information this controller works very well with linux distribution as libreelec, or recalbox on a raspberry pi.

Thank you for your help

gdsports, 2019-07-24:
This is a clone of a Microsoft xbox360 controller. The vendor ID is for Microsoft and the product ID is for xbox360 controller.

```
14:49:22.366 -> Vendor ID: 045E
14:49:22.366 -> Product ID: 028E
```
Something is wrong with your test program. Use PS3USB.ino. On a PS3 controller, getAnalogHat() returns a uint8_t so your program should be not printing such large numbers. The values should be 0..255.

gneraudeau, 2019-07-24:
Thank you for your answer.
Sorry if my programming level is bad.
Here is the content of the XBOXUSB.cpp:
[image]
readReport function is active, but how can i see "report" befor int16_t conversion?

gdsports, 2019-07-24:
Turn on [USB host debugging](https://github.com/felis/USB_Host_Shield_2.0#enable-debugging).

Add code like this. I do not have time to test this code so you may need to make changes.

```
for (uint8_t i=6; i < 14; i++) {
    USBTRACE2("readBuf ", readBuf[i]);
}

hatValue[LeftHatX] = (int16_t)(((uint16_t)readBuf[7] << 8) | readBuf[6]);

```

ABRUPTLY, 2019-07-25:
привет! помогите мне с моим USB BT адаптером купленным  https://ru.aliexpress.com/item/32384484446.html?spm=a2g0s.9042311.0.0.3ed633edrbhW8l
при включенной отладке выдает следующее сообщение
[image]
что я могу сделать что бы мой USB BT заработал ?

gneraudeau, 2019-07-26:
Thank you, It seems to be better, but when I turn on the controller, i have continuously:

_RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767
RightHatX: -32768	RightHatY: 32767_

If I press A button:

_readBuf 00
readBuf 00
readBuf 00
readBuf 00
readBuf 00
readBuf 80
readBuf FF
readBuf 7F
RightHatX: -32768	RightHatY: 32767
A_

If I press B button:

_readBuf 00
readBuf 00
readBuf 00
readBuf 00
readBuf 00
readBuf 80
readBuf FF
readBuf 7F
RightHatX: -32768	RightHatY: 32767
B_

Finally, it seems to work but the code is contaminated by "RightHatX: -32768	RightHatY: 32767". Is that possible to hide this?

gneraudeau, 2019-07-26:
I removed "RightHatX" and "RightHatY" from the code and it's better for me, i thing it will be usable for me:

[image]