felis/USB_Host_Shield_2.0 #485: Bluetooth Dongle Init Failed, error code: D1, running PS3BT

Tychka77, 2019-08-22:
HI,

So, I want to connect arduino to PS3, but when I try to start PS3BT.ino, I get the following error in the serial monitor:

BTD ConfigureDevice
BTD Init
Addr: 01
Unknown Device Connected - VID: 0A12 PID: 0001
BTD Init Failed, error code: D1

Thanks in advance,
Tychka77

Lauszus, 2019-09-02:
@Tychka77 you can not use this library to connect to other USB host. This library is for connecting USB devices i.e. mice, keyboard, Bluetooth dongles, game controllers etc for an Arduino.

Check out UnoJoy, which does exactly what you want: https://github.com/AlanChatham/UnoJoy.

Or did I misunderstand your question?

Tychka77, 2019-09-05:
@Lauszus you did not understand my question.

xxxajk, 2019-09-07:
Some require firmware to be upoloaded in order to work, and it forgets it when you unplug or reset it.