felis/USB_Host_Shield_2.0 #487: Links to hardware manual are not working

maziac, 2019-08-31:
The links like
https://www.circuitsathome.com/usb-host-shield-hardware-manual
are not working.

katadelos, 2019-09-03:
The majority of the content from the circuitsathome website can be found using the Wayback Machine - it appears that the library authors no longer have control over the domain.

The page that you linked to can be found at [https://web.archive.org/web/20140214051130/https://www.circuitsathome.com/usb-host-shield-hardware-manual](url).

maziac, 2019-09-03:
Thanks very much for the link. I didn't thought of looking into archive.org...

Lauszus, 2019-09-04:
Thanks for the feedback. I talked to the guy who used to own the site and is trying to come up with a solution.