felis/USB_Host_Shield_2.0 #488: Not possible to measure a "Zero Delay USB Encoder, Dragon Rise Inc."

maziac, 2019-08-31:
Hi,
I really like the SW it is great.
I tested a couple of USB controllers without a problem but now I wanted to test a
"Zero Delay USB Encoder, Dragon Rise Inc."
but it does not work.

The device is somewhat recognized:
~~~
Start
Interval:		10ms
HID Controller initialized
784.26
~~~

But if I press any connected button I do not get any serial output (like with the other controllers).
When I use 't' I only get a bunch of:
~~~
Launching test
Input was dropped!
Input was dropped!
Input was dropped!
Input was dropped!
Input was dropped!
Input was dropped!
Input was dropped!
Input was dropped!
...
~~~

I can use the device normally on e.g. Linux, so it's not broken.

Any idea what the problem could be?

Lauszus, 2019-09-02:
@maziac it does not seems like you are using this library based on the output? Are you using a fork?

maziac, 2019-09-02:
Sorry, this was indeed the wrong repository.
My issue is for: https://gitlab.com/loic.petit/usblag

closed 2019-09-02 by maziac