felis/USB_Host_Shield_2.0 #491: Print a character continuously by pressing and holding a key on keyboard

Vamshi253, 2019-09-18:
Hi,

What I have observed in this version is the keyboard inputs only one character when a certain key is pressed. What am I looking is the character should be printed continuously and sequentially if a certain key is pressed and hold for a moment. I am requesting you all to help me in this regard.

Thank You.

Regards,
Vamshi

JelleHaas, 2019-12-20:
Can you show me how you got the keyboard inputs in? Are you using an arduino? I personally am struggeling to get the host work

Jonas-commits, 2020-12-18:
Hi, @Vamshi253 not sure what is your application exactly and if after a year you are still interested in a solution. Digging into the HID protocol for another issue I had, I can tell you, that you can only read out the pressed keys. Repeating the key is not a part of HID protocol, for this behavior you have to write your own logic.