felis/USB_Host_Shield_2.0 #493: Reseting Arduino causes USB shield to stop responding.

RicardoGlobal, 2019-10-02:
I have an arduino UNO and an USB Shield. I am using the USBHIDBootKbd.ino sample program without modifications. I am using a bar code scanner connected to the USB Shield.
Once I download the compiled program into the arduino. I open the Serial Monitor of the Arduino IDE, and everything works fine. Then I just close the Serial monitor, open it again (wich causes a Reset to the Arduino) and the program may or may not work. Either case, I get the "Start" message in the monitor, but half the time I would get no information from the barcode scanner.

Instead the barcode reader I tried a USB keyboard, and I got similar results. Is very easy to replicate the problem.

I think the Usb.Init() is missing some initialization, because if I unplug the Barcode scanner and plug it again, it will work ok.

In any case, (even if I am wrong) I am looking for a way to simulate unplug and plug back in software, so I can include it in my setup rutine, to be sure that the barcode will work always.

Related issues:
Stops detecting USB device insertion #425
PS3BT, PS3 controller won't reconnect #461

Congratulations for your library and thanks in advance for any suggestion.

Lauszus, 2019-10-03:
Please see: https://chome.nerpa.tech/mcu/vbus-power-control-on-usb-host-shield/.

closed 2019-10-03 by Lauszus

RicardoGlobal, 2019-10-03:
Thank you for your answer. Sorry to disagree with you, this is not power related. Is there anya library call I can use to disconnect and connect back?
Is there any way I can trigger the plugged interrupt by software?

Thanks.

xxxajk, 2019-10-03:
You could try a BUSRST, but those won't work if the device is wedged/crashed.

tmk, 2019-10-04:
I had similar problem on device enumeration with a keyboard, which has a glitch, or unwanted tens us width pulse on D+ line just after plugin and bus reset. In my case the glitch is caused probably by keyboard firmware bug or improper handling on D+ pin at least, but such glitch can be happened also in some situations like unstable mate of connectors during plugin, I think. The glitch often causes MAX3421 and the library to miss detection of subsequent bus changes occured in that short period properly.(CONDETIRQ)

I'm not sure your problem is related to this but you may want to try this fix, which ignores bus activities for 200ms(USB_SETTLE_DELAY) after first detection, this works as debouncing on bus lines.
https://github.com/tmk/USB_Host_Shield_2.0/commit/227f3aab715bf7f04fc616558b80e7fa69d2b914

RicardoGlobal, 2019-10-04:
Thank you for your comment tmk and xxxajk. Finally solved the problem disabling DTR -> Reset to the arduino. Just opening a trace in the arduino. So, whenever the computer sends DTR signal, the Arduino will not Reset and the USB device will continue working. This solved my problem. Thank you all.