felis/USB_Host_Shield_2.0 #494: What variable contains complete buffer?

casxsja, 2019-10-12:
Hi,

I am working in a project where I connect my host shield and Arduino MEGA (both official) with a customized USB Hub Device (**Device Class Interface: HID (0x03)** and FullSpeed) which just sends 64B packages. I just want to watch on Serial Monitor of Arduino IDE the data buffer that the device send, so I have been studying HIDUniversal and HIDComposite library (and all their hereditance) for months and using [USBHIDMultimediakbd.ino](https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHIDMultimediaKbd/USBHIDMultimediaKbd.ino) .
I try to modify this example for my purpose. I think I have to use HIDUniversal library because it has an interesting void function calle _SaveBuffer_ .

My question is: **What variable contains the buffer of data that Device sends?**
It seems to my that are two variables that could contain all the buffer: _buf_ from **hiduniversal.Init** or **hidcomposite.Init** and _data_ from **Usb.InTransfer** .

If somebody has more documentation about the variables of the libraries, it could be helpful too =).

Thank you so much.

Cas

casxsja, 2019-11-07:
Nice... I found the error that appears when I run H**ID_Desc** example (again) is "**FailGetConfDescr**" from **HIDUniversal::Init** and the collection does not shows. Although it is fixed when I unplug and plug again the USB OTG cable that connects Shield and my Hub device (HID type, sure). When it happens, **all collection** shows in Arduino's monitor and after of that list of information, all the packet that I want to get appears in my monitor all time.

Nevertheless, I can not get this data pointer, I tried to generate it modifying many library examples but it does not work.
**Any ideas guys?** I think must there be a easy solution, but I am not enable to see it yet. I was studying these libraries and USB protocol docs from Fabruary and I ca not answer the question that I made. I need some help. In internet there are not solutions or information to face it and I do not know anybody that knows about USB communication.

I hope somebody could answer this issue.

**CAS**

casxsja, 2019-11-08:
Debugging deeper I found a new error that I overlooked. I think because of it I can not get my package directly (without unplug and plug).
The error is in Usb.cpp library, in InTransfer function: "Problem! dispatchPkt" and rcode = 5, so there is a STALL error. I tried to resolve it changing timeout from dispatchPkt function but it does not turn out to be a success.
I can solve it when I unattach and attach my Device, but it is not so professional.

**Did somebody solve these dispatchPkt and STALL errors before?** I think I am so near of the solution. Perhaps do somebody of developer team know why this error happen?
@felis @Lauszus I read some of your answers in similar user issues, so I think you could know it.

Thanks

CAS

**My Arduino code**: (Is based on SRWS1 example of Kristian Lauszus and it is not finished yet).

` //#include <usbhid.h>
//#include <usbhub.h>
//#include <hidescriptorparser.h>
#include <hiduniversal.h>
#include <SPI.h>

/* CLASES */

#define AmfiHUB_VID       0x0C17
#define AmfiHUB_PID       0x0D12

class ArduPREMO : public HIDUniversal {
public:
        ArduPREMO(USB *p) : HIDUniversal(p) {};
        bool connected() {
                Serial.println("");
                Serial.print("VID Hub es:  ");Serial.println(HIDUniversal::VID, HEX);
                return HIDUniversal::isReady() && HIDUniversal::VID == AmfiHUB_VID && HIDUniversal::PID == AmfiHUB_PID;
        };            /*linea93 hiduniversal.h*/
        
//        ArduPREMOData ardupremoData;

private:
                
        void ParseHIDData(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf); // Called by the HIDUniversal library
        
        uint8_t OnInitSuccessful() { // Called by the HIDUniversal library on success (BOOLEAN)
                if (HIDUniversal::VID != AmfiHUB_VID || HIDUniversal::PID != AmfiHUB_PID){ // Make sure the right device is actually connected
                        return 0;
                }
                
                return 1;
        };

public:
        const uint8_t* Getbuf(uint8_t *buffealo){
          return buffealo;
        }

};

/*OBJETOS Y VARIABLES GLOBALES*/
USB Usb;
ArduPREMO premo(&Usb);
HIDUniversal  hidd(&Usb);

bool printTilt;

uint8_t* pBUFF(NULL);
/*
//USB* pUsb;
//USBHID usbhid(&Usb);
//USBHID *hid;
//MultiByteValueParser multi;
//ReportDescParserBase base;
//UniversalReportParser basUni;
//UsbDeviceAddress adr;
[... 150 lines cut ...]
Endpoints:		02
Intf. Class:		03
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		21
Contents:	110100012236000705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0040
Polling interval:	01

Endpoint descriptor:
Endpoint address:	01
Attr.:			03
Max.pkt size:		0040
Polling interval:	01

Addr:1(0.0.1)

------------------

xxxajk, 2019-11-09:
Stall error means different things for different devices. If you have the
VBUS chip on the board, you can powercycle the stuck device, but only if it
isn't on a powered hub.

xxxajk, 2019-11-09:
More information...

The STALL packet indicates that the endpoint has halted, or a control pipe
does not support a certain request.

A function uses the STALL handshake packet to indicate that it is unable to
transmit or receive data. Besides the default control pipe, all of a
function's endpoints are in an undefined state after the device issues a
STALL handshake packet. The host must never issue a STALL handshake packet.

Typically, the STALL handshake indicates a functional stall. A functional
stall occurs when the halt feature of an endpoint is set. In this
circumstance, host intervention is required via the default control pipe to
clear the halt feature of the halted endpoint. Less often, the function
returns a STALL handshake during a SETUP or DATA stage of a control
transfer. This is called a protocol stall and is resolved when the host
issues the next SETUP transaction.

casxsja, 2019-11-11:
First of all, thanks @xxxajk for your answers. I have reseach about VBus chip, but I have not idea about my device's hardware because lake of documentation. So, I have written an email to the company asking about it. And the info of Stall was nice, I have read about it before, but I think I have a nice idea now.

About powercycle, yes... I think the action of unplugging and plugging again my device from Arduino shield makes a "reset" of the communication and the Stall error goes. When it happens, the buf pointer of OnParseItem (of hiduniversalparser.cpp) is showed on my monitor.
Following you advice I tried to use the functions: **vbusPower(VBUS_t state) and Init(int mseconds)** of usbhost.h in my Arduino code and inside USB.cpp (different tryings). They does not work. I tryed it with part of the example of Lightweight post of Oleg ([link here](https://web.archive.org/web/20171001020402/https://www.circuitsathome.com/mcu/lightweight-usb-hostpart-1/)).

**Do I have to use the two functions of usbhost.h library for powercycling? Do you have a better example?** I just need to get the package of 64B that this devices send everytime... **Do you know a better option for this goal based on my last code or library's examples?**

Thanks again for your time xxxajk .

CAS

xxxajk, 2019-11-11:
This is what you want:

https://web.archive.org/web/20160821185145/https://www.circuitsathome.com/mcu/vbus-power-control-on-usb-host-shield

casxsja, 2019-11-11:
I apologise @xxxajk . I was wrong when I said  "_I tryed it with part of the example of Lightweight post of Oleg (link here)._". I meant that I tryed it with part of the code example of Vbus power control of Oleg (the link you just passed me).
My questions of my last paragraph were due to I read Vbus power article and **tried the functions that Oleg used and I wrote them in different parts of my code and of the library** without success.

i.e. I wrote these functions (first vbusPower(), it failed so later I tryed Init(int ms) in my own loop and tryed:

void loop()
{
    Serial.println("----Tasking----");
    Usb.Task```
();    //Enumeración --- Llega a --> Estado Funcionamiento de USB
    Serial.println("----Termino Task----");

    
    if(Usb.getUsbTaskState() == USB_STATE_RUNNING){
      Serial.println("---USB_STATE_RUNNING---");
      if(premo.connected()){
        Serial.println("connected y Universal está Ready");
       
        Inicializa_ArduPREMO(); //Comienza comunicación USB
      } Serial.println("--Termino Connected--");
    } Serial.println("-Termino getUsbTaskStateRunning-");

    #if 0

    if(repite){
      Usb.Init(3000);
      repite = false;
    }
    #endif
    
    #if 0
    if(repite){
      Usb.vbusPower(vbus_off);
      delay(2000);
      Usb.vbusPower(vbus_on);
      delay(100);
      repite = false;
    }
    #endif

   Serial.println("---Termino loop---");
    
}
```
So... I would like to ask you again. Do I have to use the two functions of usbhost.h library for powercycling? Do you have a better example? I just need to get the package of 64B that this device sends everytime... Do you know a better option for this goal based on my last code or library's examples?

Thank you again =)

CAS

xxxajk, 2019-11-12:
1: **Requires** physically added part (the switch chip, jumper wires etc.)
2: Use the built-in functions.
3: No need to re-call Init().
like so:

```

// Loop this many times and trigger a USB power cycle. ~20 seconds in this case.
#define FAKE_TRIGGER_LOOPS 20000
// Count down to trigger USB power cycle event.
int fake_issue = FAKE_TRIGGER_LOOPS; 
// Actual power_cycle trigger.
volatile boolean power_cycle = false;
// Count down for how long to have Vbus off.
int off_time = 0;

void loop()
{

  Usb.Task(); // Always service USB host

  if(power_cycle) { // Power cycle requested?
    Usb.vbusPower(vbus_off);
    power_cycle = false;
    off_time = 500;
  }

  if(off_time > 0) { // Power cycling?
    delay(1);
    off_time--;
    if(off_time == 0) Usb.vbusPower(vbus_on);
  } else {
    // Do other processing here...
    // Pretend work...
    delay(1);

    // Fake issue trigger. Here we pretend something bad happened.
    fake_issue--;
    if(fake_issue == 0) {
      fake_issue=FAKE_TRIGGER_LOOPS; // fail again in ~ FAKE_TRIGGER_LOOPS milliseconds
      power_cycle = true;
    }
  }
}

```

casxsja, 2019-11-12:
Ok! Thank you so much for your support! =) It was very clear. I will try that this weekend, if I have any problem or advance I will comment here.

Cas

xxxajk, 2019-11-12:
You are very welcome. :-)