felis/USB_Host_Shield_2.0 #499: Mini Keyboard Wireless Touchpad: touchpad is not working.

rtek1000, 2019-11-04:
I would like to inform you that the touchpad is not working.

I tried:
- USBHIDBootKbdAndMouse
- USBHIDBootMouse

(the wired mouse works)

USB_desc.ino output:

> Start
> 
> 
> 01
> --
> 
> Device descriptor: 
> Descriptor Length:	12
> Descriptor type:	01
> USB version:		0110
> Device class:		00
> Device Subclass:	00
> Device Protocol:	00
> Max.packet size:	08
> Vendor  ID:		0513
> Product ID:		0318
> Revision ID:		0100
> Mfg.string index:	00
> Prod.string index:	00
> Serial number index:	00
> Number of conf.:	01
> 
> Configuration descriptor:
> Total length:		003B
> Num.intf:		02
> Conf.value:		01
> Conf.string:		00
> Attr.:			A0
> Max.pwr:		32
> 
> Interface descriptor:
> Intf.number:		00
> Alt.:			00
> Endpoints:		01
> Intf. Class:		03
> Intf. Subclass:		01
> Intf. Protocol:		01
> Intf.string:		00
> Unknown descriptor:
> Length:		09
> Type:		21
> Contents:	100100012240000705
> 
> Endpoint descriptor:
> Endpoint address:	81
> Attr.:			03
> Max.pkt size:		0008
> Polling interval:	08
> 
> Interface descriptor:
> Intf.number:		01
> Alt.:			00
> Endpoints:		01
> Intf. Class:		03
> Intf. Subclass:		01
> Intf. Protocol:		02
> Intf.string:		00
> Unknown descriptor:
> Length:		09
> Type:		21
> Contents:	100100012277000705
> 
> Endpoint descriptor:
> Endpoint address:	82
> Attr.:			03
> Max.pkt size:		0008
> Polling interval:	08
> 
> 
> Addr:1(0.0.1)
> 

[image]

xxxajk, 2019-11-04:
might be proprietary, and not support HID BOOT?

rtek1000, 2019-11-05:
Well, how can it be proprietary if it works on Linux, Android, Windows without having to install anything additional? And on the MAX3421E works the keyboard part.

Is it possible to perform some verification through Linux/Windows?

rtek1000, 2019-11-05:
I found this:

> Usbhid-dump is a USB HID dumping utility based on libusb 1.0. It dumps USB HID device report descriptors and reports themselves as they are being sent, for all or specific device interfaces.

> Warning: please be careful running usbhid-dump as a superuser without limiting your device selection with options. Usbhid-dump will try to dump every device possible and If you're using a USB keyboard to control your terminal, it will be detached and you will be unable to terminate usbhid-dump and regain control.

https://github.com/DIGImend/usbhid-dump

xxxajk, 2019-11-05:
Without additional hardware, it gets complicated, but can be done using Linux.
You will have to track and trace USB URBs.
http://www.linux-usb.org/USBMon/dissertation/USB-dissertation.htm
Prepare to get neck deep in information.

rtek1000, 2019-11-05:
Nice!!!!!! Lib 3.0 works all functions!!!

https://github.com/felis/UHS30/tree/master/libraries/UHS_host/UHS_HID/examples/USB_HOST_SHIELD/UHS_HID_RAW

closed 2019-11-05 by rtek1000

xxxajk, 2019-11-05:
Not surprised. UHS3.0 enumerates in a way that isn't as hackish.

rtek1000, 2019-11-05:
It seems that they had not yet invented the ASCII standard before making the keyboard map in this strange sequential order:

```
(Keyboard)
Length 8 RAW input 8 bytes interface 0, Subclass 01, Protocol 01
Data: 00 00 00 00 00 00 00 00

0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 "Key released"
0x00 0x00 0x04 0x00 0x00 0x00 0x00 0x00 A
0x00 0x00 0x05 0x00 0x00 0x00 0x00 0x00 B
0x00 0x00 0x06 0x00 0x00 0x00 0x00 0x00 C
...       0x07                          D
...       0x1b                          X
0x00 0x00 0x1c 0x00 0x00 0x00 0x00 0x00 Y
0x00 0x00 0x1d 0x00 0x00 0x00 0x00 0x00 Z
0x00 0x00 0x1e 0x00 0x00 0x00 0x00 0x00 1
0x00 0x00 0x1f 0x00 0x00 0x00 0x00 0x00 2
...       0x20                          3
...       0x25                          8
0x00 0x00 0x26 0x00 0x00 0x00 0x00 0x00 9
0x00 0x00 0x27 0x00 0x00 0x00 0x00 0x00 0
0x00 0x00 0x29 0x00 0x00 0x00 0x00 0x00 Esc
0x00 0x00 0x2a 0x00 0x00 0x00 0x00 0x00 Back
0x00 0x00 0x2b 0x00 0x00 0x00 0x00 0x00 Tab
0x00 0x00 0x2d 0x00 0x00 0x00 0x00 0x00 -
0x00 0x00 0x2e 0x00 0x00 0x00 0x00 0x00 =
0x00 0x00 0x2f 0x00 0x00 0x00 0x00 0x00 ´
0x00 0x00 0x30 0x00 0x00 0x00 0x00 0x00 [
0x00 0x00 0x31 0x00 0x00 0x00 0x00 0x00 ]
0x00 0x00 0x33 0x00 0x00 0x00 0x00 0x00 Ç (Layout PtBr ABNT)
0x00 0x00 0x34 0x00 0x00 0x00 0x00 0x00 ~
0x00 0x00 0x35 0x00 0x00 0x00 0x00 0x00 '
0x00 0x00 0x36 0x00 0x00 0x00 0x00 0x00 ,
0x00 0x00 0x37 0x00 0x00 0x00 0x00 0x00 .
0x00 0x00 0x38 0x00 0x00 0x00 0x00 0x00 ;
0x00 0x00 0x39 0x00 0x00 0x00 0x00 0x00 Caps
0x00 0x00 0x46 0x00 0x00 0x00 0x00 0x00 PrtSc
0x00 0x00 0x4b 0x00 0x00 0x00 0x00 0x00 PgUp
0x00 0x00 0x4c 0x00 0x00 0x00 0x00 0x00 Del
0x00 0x00 0x4e 0x00 0x00 0x00 0x00 0x00 PgDn
0x00 0x00 0x4f 0x00 0x00 0x00 0x00 0x00 Arrow R
0x00 0x00 0x50 0x00 0x00 0x00 0x00 0x00 Arrow L
0x00 0x00 0x51 0x00 0x00 0x00 0x00 0x00 Arrow Up
0x00 0x00 0x52 0x00 0x00 0x00 0x00 0x00 Arrow Up

0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 Ctrl
0x01 0x00 0x06 0x00 0x00 0x00 0x00 0x00 Ctrl+C
0x01 0x00 0x19 0x00 0x00 0x00 0x00 0x00 Ctrl+V
0x01 0x00 0x1b 0x00 0x00 0x00 0x00 0x00 Ctrl+X
0x01 0x00 0x1c 0x00 0x00 0x00 0x00 0x00 Ctrl+Y
0x01 0x00 0x1d 0x00 0x00 0x00 0x00 0x00 Ctrl+Z

0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00 Shift
0x02 0x00 0x1e 0x00 0x00 0x00 0x00 0x00 ! (Shift+1)
0x02 0x00 0x1f 0x00 0x00 0x00 0x00 0x00 @ (Shift+2)
0x02 0x00 0x20 0x00 0x00 0x00 0x00 0x00 # (Shift+3)
0x02 0x00 0x21 0x00 0x00 0x00 0x00 0x00 $ (Shift+4)
0x02 0x00 0x22 0x00 0x00 0x00 0x00 0x00 % (Shift+5)
0x02 0x00 0x23 0x00 0x00 0x00 0x00 0x00 ¨ (Shift+6)
0x02 0x00 0x24 0x00 0x00 0x00 0x00 0x00 & (Shift+7)
0x02 0x00 0x25 0x00 0x00 0x00 0x00 0x00 * (Shift+8)
0x02 0x00 0x26 0x00 0x00 0x00 0x00 0x00 ( (Shift+9)
0x02 0x00 0x27 0x00 0x00 0x00 0x00 0x00 ) (Shift+0)
0x02 0x00 0x2d 0x00 0x00 0x00 0x00 0x00 _ (Shift+-)
0x02 0x00 0x2e 0x00 0x00 0x00 0x00 0x00 + (Shift+=)
0x02 0x00 0x2f 0x00 0x00 0x00 0x00 0x00 ` (Shift+´)
0x02 0x00 0x30 0x00 0x00 0x00 0x00 0x00 { (Shift+[)
0x02 0x00 0x31 0x00 0x00 0x00 0x00 0x00 } (Shift+])
0x02 0x00 0x34 0x00 0x00 0x00 0x00 0x00 ^ (Shift+~)
0x02 0x00 0x35 0x00 0x00 0x00 0x00 0x00 " (Shift+')
0x02 0x00 0x36 0x00 0x00 0x00 0x00 0x00 < (Shift+,)
0x02 0x00 0x37 0x00 0x00 0x00 0x00 0x00 > (Shift+.)
0x02 0x00 0x38 0x00 0x00 0x00 0x00 0x00 : (Shift+;)

0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 Alt

0x05 0x00 0x4c 0x00 0x00 0x00 0x00 0x00 Ctrl+Alt+Del

0x40 0x00 0x00 0x00 0x00 0x00 0x00 0x00 AltGr
0x40 0x00 0x08 0x00 0x00 0x00 0x00 0x00 ° (AltGr+E)
0x40 0x00 0x14 0x00 0x00 0x00 0x00 0x00 / (AltGr+Q)
0x40 0x00 0x1a 0x00 0x00 0x00 0x00 0x00 ? (AltGr+W)
0x40 0x00 0x1e 0x00 0x00 0x00 0x00 0x00 ¹ (AltGr+1)
0x40 0x00 0x1f 0x00 0x00 0x00 0x00 0x00 ² (AltGr+2)
0x40 0x00 0x20 0x00 0x00 0x00 0x00 0x00 ³ (AltGr+3)
0x40 0x00 0x21 0x00 0x00 0x00 0x00 0x00 £ (AltGr+4)
0x40 0x00 0x22 0x00 0x00 0x00 0x00 0x00 ¢ (AltGr+5)
0x40 0x00 0x23 0x00 0x00 0x00 0x00 0x00 ¬ (AltGr+6)
0x40 0x00 0x2e 0x00 0x00 0x00 0x00 0x00 § (AltGr+=)
0x40 0x00 0x30 0x00 0x00 0x00 0x00 0x00 ª (AltGr+[)
0x40 0x00 0x31 0x00 0x00 0x00 0x00 0x00 º (AltGr+])

0x80 0x00 0x00 0x00 0x00 0x00 0x00 0x00 Win (Start Menu)
0x80 0x00 0x08 0x00 0x00 0x00 0x00 0x00 Win+E (Explorer)
0x80 0x00 0x15 0x00 0x00 0x00 0x00 0x00 Win+R (Run)

(Multimedia keys)
Length 3 RAW input 3 bytes interface 1, Subclass 01, Protocol 02
Data: 00 00 00

0x00 0x00 0x00 "Key released"
0x02 0x21 0x02 Search
0x02 0x23 0x02 Home
0x02 0x96 0x01 "Launch Internet Browser"
0x02 0xb5 0x00 >>|
0x02 0xb6 0x00 |<<
0x02 0xcd 0x00 |>|| (Play/Pause)
0x02 0xe2 0x00 Mute
0x02 0xe9 0x00 V+
0x02 0xea 0x00 V-

0x02 0x83 0x01 "Launch Media Player"
0x02 0x8a 0x01 "Launch Email"

(Mouse)
Length 6 RAW input 6 bytes interface 1, Subclass 01, Protocol 02
Data: 01 00 00 00 00 00

0x01 0x00 0x00 0x00 0x00 0x00 "Button released"
0x01 0x01 0x00 0x00 0x00 0x00 "Button L" (One finger)
0x01 0x02 0x00 0x00 0x00 0x00 "Button R" (Three Fingers)
0x01 0x00 0x00 0x00 0x01 0x00 "Slide DnUp" (Two Fingers)
0x01 0x00 0x00 0x00 0xff 0x00 "Slide UpDn" (Two Fingers)
0x01 0x00 0xNN 0xMM 0x00 0x00 "Finger movement"
"Finger movement": 0x01 0x00 0xNN 0x00 0x00 0x00 "Move x" (From L to R: NN = 0x00 + speed); (From R to F: NN = 0xff - speed)
"Finger movement": 0x01 0x00 0x00 0xMM 0x00 0x00 "Move y" (From Up to Dn: MM = 0x00 + speed); (From Dn to Up: MM = 0xff - speed)

(Stand by; timeout = +/-60s; touchpad: off; wake up: key press)
Length 6 RAW input 6 bytes interface 1, Subclass 01, Protocol 02
Data: 01 00 00 00 00 00
Length 3 RAW input 3 bytes interface 1, Subclass 01, Protocol 02
Data: 02 00 00

```

xxxajk, 2019-11-06:
No, that's normal. Keyboards do not output ASCII, they output keyscan codes.
http://www.philipstorr.id.au/pcbook/book3/scancode.htm

rtek1000, 2019-11-06:
What would be the best way to convert to ASCII?

I tested the code below, and it worked, but I think using a character array should be more compact.

```C++
//#define

// Load the USB Host System core
#define LOAD_USB_HOST_SYSTEM
// Load USB Host Shield
#define LOAD_USB_HOST_SHIELD
// Use USB hub, you might want this for multiple devices.
#define LOAD_UHS_HUB

// Patch printf so we can use it.
#define LOAD_UHS_PRINTF_HELPER
#define DEBUG_PRINTF_EXTRA_HUGE 0
#define DEBUG_PRINTF_EXTRA_HUGE_USB_HID 1

#define LOAD_UHS_HID

#include <Arduino.h>
#ifdef true
#undef true
#endif
#ifdef false
#undef false
#endif

#include <UHS_host.h>

class myHID_processor : public UHS_HID_PROCESSOR {
  public:
    myHID_processor(void) {};

    bool caps_lock_buff = false;
    bool shift_accent_buff = false;
    int accent_buff = 0;

    bool haveAccent(uint8_t data) {
      if (accent_buff != 0) {
        return true;
      } else {
        return false;
      }
    }

    void printAccent(uint8_t data, bool shift) {
      if (shift) {
        switch (data) {
          case 0x23:
            printf_P(PSTR("¨")); //  ASCII 249  // ¨
            break;
          case 0x2f:
            printf_P(PSTR("`")); // char(data[2] + 49)); // ASCII 96  // `
            break;
          case 0x34:
            printf_P(PSTR("^")); // char(data[2] + 42)); // ASCII 94  // ^
            break;
          default:
            break;
        }
      } else {
        switch (data) {
          case 0x2f:
            printf_P(PSTR("´")); // char(data[2] + 192): ASCII 239 // ´
            break;
          case 0x34:
            printf_P(PSTR("~")); // char(data[2] + 74): ASCII 126 // ~
            break;
[... 491 lines cut ...]
        default:
          break;
      }
    }

    //const int keyMap[3][2] = {{0x2d, '-'}, {3, 4}, {5, 6}};
};

myHID_processor HID_processor1;
myHID_processor HID_processor2;
MAX3421E_HOST UHS_Usb;
UHS_USBHub hub_1(&UHS_Usb);
UHS_HID hid1(&UHS_Usb, &HID_processor1);
UHS_HID hid2(&UHS_Usb, &HID_processor2);

void setup() {
  USB_HOST_SERIAL.begin(115200);
  while (UHS_Usb.Init(1000) != 0);
  printf_P(PSTR("\r\nHID RAW demo Begin.\r\n"));
}

void loop() {
  delay(1);
}
```

rtek1000, 2019-11-18:
I tried using Arduino Nano / UNO / Pro Mini, but it didn't work, maybe for lack of memory. I could only use it with Mega2560, I couldn't make it work with STM32 either, the code has many adaptations made with #ifdef, without indentation, which confuses a lot, after indenting the code I could understand a little better, but still not enough (make it work with STM32). Maybe it's a good idea to use esp32 (520 KiB SRAM), but it didn't compile either.

The intention was to use Arduino Pro Mini (Atmega328) as a bridge between the MAX3421E and STM32 via serial port, but it didn't work out. It only worked with the Mega2560, but it's much more expensive and takes up a lot of physical space, even the Mega2560 PRO version. (I will try to switch the project from STM32 to Mega2560, but the ILI9341 display that is currently controlled by STM32 will be very slow)

If anyone wants to try this keyboard with Mega2560, follows the code, need to use the UHS30 library:

```C++
//#define
//#define USB_HOST_SHIELD_USE_ISR 0
#define SERIAL_PORT_MONITOR Serial

// Load the USB Host System core
#define LOAD_USB_HOST_SYSTEM
// Load USB Host Shield
#define LOAD_USB_HOST_SHIELD
// Use USB hub, you might want this for multiple devices.
#define LOAD_UHS_HUB

// Patch printf so we can use it.
#define LOAD_UHS_PRINTF_HELPER
#define DEBUG_PRINTF_EXTRA_HUGE 0
#define DEBUG_PRINTF_EXTRA_HUGE_USB_HID 0

#define LOAD_UHS_HID

#include <Arduino.h>
#ifdef true
#undef true
#endif
#ifdef false
#undef false
#endif

#include <UHS_host.h>

class myHID_processor : public UHS_HID_PROCESSOR {
  public:
    myHID_processor(void) {};

    bool caps_lock_buff = false;
    bool shift_accent_buff = false;
    int accent_buff = 0;

    bool haveAccent(uint8_t data) {
      if (accent_buff != 0) {
        return true;
      } else {
        return false;
      }
    }

    void printAccent(uint8_t data, bool shift) {
      if (shift) {
        if (data == 0x23) {
          printf_P(PSTR("¨")); // char(data[2] + 192): ASCII 239 // ´
        } else if (data == 0x2f) {
          printf_P(PSTR("`")); // char(data[2] + 74): ASCII 126 // ~
        } else if (data == 0x34) {
          printf_P(PSTR("^")); // char(data[2] + 74): ASCII 126 // ~
        }
      } else {
        if (data == 0x2f) {
          printf_P(PSTR("´")); // char(data[2] + 192): ASCII 239 // ´
        } else if (data == 0x34) {
          printf_P(PSTR("~")); // char(data[2] + 74): ASCII 126 // ~
        }
      }
    }

    void accent(uint8_t data, bool data2) {
[... 657 lines cut ...]
  //  SPI.setMISO(PB4); // using pin name PY_n
  //  SPI.setSCLK(PB3); // using pin number PYn
  //  //SPI.setSSEL(PA15); // using pin number PYn
  //  //SPI.setClockDivider(32);
  //
  //  //pinMode(PB4, INPUT);
  //
  //  pinMode(PA15, OUTPUT);
  //  digitalWrite(PA15, HIGH);
  //  SPI.begin(PA15);
  //
  //#endif

  //Serial.begin(115200);
  USB_HOST_SERIAL.begin(115200);
  printf_P(PSTR("\r\nStart.\r\n"));
  while (UHS_Usb.Init(1000) != 0);
  printf_P(PSTR("\r\nHID Begin.\r\n"));
  //Serial.print(F("\r\nHID RAW demo Begin.\r\n"));
}

void loop() {
  delay(1);
}
```

rtek1000, 2019-11-18:
It worked with arduino Pro Mini! (UHS_HID_RAW.ino from UHS30 lib) Some library file must be corrupted. I deleted them all and installed it again. Thanks.

rtek1000, 2019-11-18:
Note: Keyboard receiver (USB) worked with 3.3V

xxxajk, 2019-11-20:
It has to do with the enumeration sequence. Your device is apparently sensitive to multiple probes.