felis/USB_Host_Shield_2.0 #501: Polling rate

maziac, 2019-11-28:
I'm using the USB host shield to measure the lag of some joysticks.
But I'm not sure if the correct polling rate is used.

E.g. I have a  joystick which requests 1 ms poll time in one endpoint and 10 ms poll time at 2 other end points.
What polling rate is used?
From the lag I would suspect that 10 (or 8ms) are used instead of 1ms.

Can I see (at some variable) what exact polling rate is used?

Can I force the USB shield to some polling rate?

xxxajk, 2019-11-29:
This might be caused by something else. Which MCU are you using? How often do you actually call the usb task?

maziac, 2019-11-29:
Actually quite often.

Here is my main loop.
```
    long startTime = millis();
    while(joystickButtonPressed) {
      delay(1);
      // Check if too long
      long stopTime = millis();
      long diffTime = stopTime - startTime;     
      if(diffTime > 1000) {
        // More than a second
        Error(F("Error:"), F("No response!a"));
        return;
      }
      Usb.Task();
    }
```

I have found this here in the USB shield SW:
```
               if(pollInterval < pep->bInterval) // Set the polling interval as the largest polling interval obtained from endpoints
                        pollInterval = pep->bInterval;
```

This correlates to what I am seeing.

xxxajk, 2019-11-29:
What happens if you:
1: remove delay(1);
2: pre-calculate the expire time. (if you need help, ask)
Don't forget that Task() time can and will fluctuate.

If you want something more spot-on, UHS3 would be more of what you would want to use, however the drivers be written, and the unified joystick HID hasn't been discussed yet.
I have just a few very generic USB joysticks to do any code.
As long as you are using a generic USB joystick, and at least a MEGA 256X/128X based board, I can come up with something better for you on UHS3 in my limited spare time. Anything that uses Bluetooth at this point isn't possible at all, as well as anything not using standard HID, because I not only lack such hardware, but there's no Bluetooth stack API.

maziac, 2019-11-30:
Removing the delay doesn't change anything.
I also tested with another joystick with one endpoint only that reports 1ms. For that joystick I get reasonable lag values of 3-4ms.
So the algorithm seems to work.
The system is a ATmega328.

At the moment I am using a generic USB HID joystick but I would like to extend this to xbox as well.

Is there any variable I could check to see what polling rate is really used?

Is there any way to force a polling rate?

xxxajk, 2019-11-30:
Polling rate is normally specified by the device. 
Polling faster than the value returned will only repeat the last value normally, but doesn't have to. Some devices will simply not reply at all until the amount of time needed to do what it has to do is ready.
It really depends on the device, how it's firmware is written, and other factors that we can't affect.

maziac, 2019-12-01:
To my understanding, please correct if I'm wrong:
The suggested polling rate is requested by the device.
The host chooses a polling rate that is equal or lower.
If a device requests a polling rate of 10ms the host will normally poll the device with a polling rate that is power of 2 and equal or lower. I.e. in this case 8 ms.

I.e. there must be an option in the USB host shield library that allows to set the really used polling rate. But I can't find it. It would be great if you could point me to the location in source code.

Or is this already done by the HW?

xxxajk, 2019-12-01:
> To my understanding, please correct if I'm wrong:
> The suggested polling rate is requested by the device.
Yes.
> The host chooses a polling rate that is equal or lower.
Yes.
> If a device requests a polling rate of 10ms the host will normally poll the device with a polling rate that is power of 2 and equal or lower. I.e. in this case 8 ms.
No, it should be 10ms or more.

> I.e. there must be an option in the USB host shield library that allows to set the really used polling rate. But I can't find it. It would be great if you could point me to the location in source code.
> 
> Or is this already done by the HW?

For hid-joystick, it's here:
https://github.com/felis/USB_Host_Shield_2.0/blob/06d5ed134a37e22575c0ce18c061d9ef115151e0/hiduniversal.cpp#L333

maziac, 2019-12-01:
Yes, that's what I thought too.

But when I change the code directly in the lib to:
pollInterval = 1;

It does not change anything.

I mean: I attach a joystick that request 10ms.
In the code I use 'pollInterval=1' in any case, i.e. overriding the 10ms request.
And the resulting polling rate is still 10ms, not 1ms.

How can that happen?

xxxajk, 2019-12-01:
As I explained before, if the device can only report at 10ms, that is the fastest it will go.

Telling the joystick code to report quicker won't cause it to do so.
Polling the joystick faster won't cause it to report quicker either.

maziac, 2019-12-01:
Note: vice versa it is working.
If the joystick requests 10ms, but pollInterval=16 is forced the resulting poll interval really is 16ms.
Why can't I force to a smaller value?

xxxajk, 2019-12-01:
Right, you can poll slower, that's fine, faster will never go faster than what it claimed it can.

maziac, 2019-12-01:
Why should it not report faster if I change the polling rate to a smaller value?
If the device is polled faster it should report faster.

E.g. in Linux this is often done to increase the polling rate of a mouse HID device.

xxxajk, 2019-12-01:
Simply, because it can't do that.
A mouse is a different device too, and works differently. Some mice might delay, some won't.
Again, it is nothing you can actually control, and if some joystick actually can go faster, it will.

It's unlikely most people can mash a button that fast anyway.

Bit of fun about this:
In 1985, Takahashi Meijin (real name Takahashi Toshiyuki) became a celebrity in Japan by pressing a button on a video game controller 16 times in one second on television. If you picked up a controller right now you could probably pull off 10 with minimal practice, 12 after you get used to it. But you're about as likely to hit 16 as you are to beat Usain Bolt in a footrace.

maziac, 2019-12-01:
The problem is not about how often a button is being press but about the delay. But don't let go into this discussion.

I have a joystick where I can control the firmware. I.e. it's HW is capable of reporting at 1ms rate.
I can simply change the USB descriptor to request 10ms or 1ms.
When it is polled by the USB host it will toggle an GPIO pin. Thus I can measure directly how often the device is being polled with an oscilloscope.

Then for the original problem:
There is a  2nd joystick where I don't have control about the firmware. It is requesting 1ms at one endpoint but offers 2 other endpoints where it requests 10ms.

Under Linux I can see that this device is polled more frequently than 10ms. I.e. I can reach a button response lag of 3-5ms (measured for 100 cycles).
If it would have been polled at 10ms the lag should vary up to 10ms at least.
Thus it seems that Linux is really using the 1ms poll rate. I.e. the minimum of the poll rates requested by the endpoints.

For USB Host shield library this looks different. Here it seems that the max. of the requested poll rates is used.

xxxajk, 2019-12-01:
Ok, basically, the UHS2 driver looks at all the intervals, and chooses the one that takes the longest.
That's why.
I think this was done to save on the limited available RAM.

maziac, 2019-12-03:
I now copied the HIDUniversal and used the minimum as poll interval.
Now I can use the 1ms poll interval.

I'll close the issue.

closed 2019-12-03 by maziac