felis/USB_Host_Shield_2.0 #502: Arduino Uno Wifi Rev.2 not supported ?

Rusteo, 2019-12-02:
Hi all,

I am having trouble using this library with my Arduino Uno Wifi Rev.2 (I am using a [Sparkfun shield](https://www.sparkfun.com/products/9947)). I am trying to run the "board_qc" example on my arduino. When I first compiled the sketch, I had the following error message:

_# error"Please define board in avrpins.h"_

So I have added the following pin numbers to "avrpins.h" in order to use my Uno Wifi Rev.2 and according to the drawings:

```
#elif defined(__AVR_ATmega4809__)
// Arduino Uno Wifi Rev2 pin numbers

#define P0  Pc5
#define P1  Pc4
#define P2  Pa0
#define P3  Pf5
#define P4  Pc6
#define P5  Pb2
#define P6  Pf4
#define P7  Pa1

#define P8  Pe3
#define P9  Pb0
#define P10  Pb1
#define P11  Pe0
#define P12  Pe1
#define P13  Pe2

#define P14  Pd0
#define P15  Pd1
#define P16  Pd2
#define P17  Pd3
#define P18  Pd4
#define P19  Pd5
```
This fixed my first issue. However, I now get this error when compiling (complete error report is enclosed because very long: [board_qc Error Message.txt](https://github.com/felis/USB_Host_Shield_2.0/files/3912841/board_qc.Error.Message.txt)):

_exit status 1
'class USB' has no member named 'Init'; did you mean 'init'?_

I tried to use "init()" instead of "Init()" but it did not solve the issue:

_exit status 1
'void USB::init()' is private within this context_

I would like to add that I am compiling using the Register Emulation set on ATMEGA328.

My conclusion is that the library is not compatible with the Uno Wifi Rev.2 as is. Has someone been able to use this library with the Uno Wifi Rev.2 ? If yes how ?

Any help appreciated. Thanks.

timkouters, 2022-08-22:
Hi,

I think it's not that simple.
Read: https://forum.arduino.cc/t/megacorex-arduino-support-for-atmega4809-3209-1609-809-and-4808-3208-1608-808/629534

I aso wish that the shield was supported for the Rev2 Wifi.

microbmen, 2023-06-14:
any updates on this?