felis/USB_Host_Shield_2.0 #508: USBFTDILoopback example with FT230X

mprowe, 2019-12-27:
After lengthy debugging, I have found that I need to amend the FTDI_PID in the cdcftdi.h (line 29) file to match my USB OBD dongle. Now, while this solves my immediate problem, it severely reduces the generality of the library - the FT232's with 0x6001 PID's will now not work!

Can anyone suggest how the library can be edited to work with FTDI devices that have PID=0x6001 or PID=0x6015?

Regards, ~Martin

xxxajk, 2019-12-28:
Use UHS3, see if that works better for you.
FTDI on UHS2 can be finicky if you aren't careful.
As far as editing for new PIDs, you have done it correctly by modifying the header file.

gdsports, 2020-01-04:
@mprowe You could pass the PID as a parameter to the FTDI object. The only change is in the sketch, not cdcftdi.h or cdcftdi.cpp.

`FTDI             Ftdi(&Usb, &FtdiAsync, 0x6015);`

mprowe, 2020-01-04:
@gdsports 
Thank you. That is just what I was looking for. Very neat....
Regards,