felis/USB_Host_Shield_2.0 #509: A strange issue happens on FT232B

ziv2013, 2020-01-01:
Hi All,

I'm working on a project which need to get data from a USB device(MAVO-MONITOR USB). This device works well under windows. There is an FT232B chip in it to convert Uart to USB. And I use USB Host Shield with Arduino Uno to read the data from it. It works well with FT232R, I have tried FT232R board with loopback mode and communicate with another USB2Uart board. All the data can be send an received normally. But after I connect it with MAVO-MONITOR USB device. It can just send(I have confirmed this by sending Reset Command for this device and it will reset after received it). Arduino can't get data from this device. 

From the above experiment I'm sure this is not a HW related issue. As FT232R works well and I can send command to the device.

And I have used a USB analyer for debugging (Teledyne LeCroy USB Protocol Analyser, T3).  My code is like below:

        elsp=millis();
        while (millis()-elsp<2000) {
          uint16_t rcvd = 64;
          uint8_t buf[64];
        rcode = Ftdi.RcvData(&rcvd, buf);
        if (rcode && rcode != hrNAK)
        {
            Serial.println(F("RcvData Error"));
        }

        if (rcvd>2)
        {
              buf[rcvd]=0;
              Serial.println((char*)(buf+2));
        } 
        delay(16);
        } 

There is no output from this code. I have tried, the value of rcvd is always 0. This should be at least 2 event there is valid data on the  serial port.

From protocol analyser I can see the package comes from device send to EP1 and the HOST Shield has ACK to it. But I don't know why I can't get any data in my code. And I have checked the return value 'rcode', it's always 4 when means hrNAK. So this really puzzled me: the HOST have ACK to the device, but I can see nothing.

I have tried to compare the USB packages with Windows communication. But there are too many packages.

Does anyone know what's the different between FT232R and FT232B(this version is too old to buy, I can just find FT232BM in the market, and I'm not sure if  I can see same issue with BM version)? And is there any directions for me to debug?

Thanks a lot.

Ziv

xxxajk, 2020-01-04:
Can you get to me the USB trace from your analyzer?

ziv2013, 2020-01-04:
Thanks for your reply.
I have post pictures on my website https://www.lab-z.com/ft232bissue/
If you have installed 'Teledyne Lecroy USB Protocol Suite', I can send you the trace file directly.

xxxajk, 2020-01-04:
Considering the software is windows only, that's a no-go.
I'll review the output above and let you know if I need more from you.

xxxajk, 2020-01-04:
The FT232Bx and FT232Rx are basically identical from a software point of view.
Some things to check:
Baud rate correct? Data size/parity/stop bits correct?
Can you give me the VID and PID of the device?

ziv2013, 2020-01-04:
Below is the device information I get from UsbView:

[Port1]  :  USB Serial Converter

Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            17
Companion Hub Symbolic Link Name: USB#ROOT_HUB30#4&1a6e0085&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "M 504 G"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                    0x1B
Open Pipes:                           2

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0200
bDeviceClass:                      0x00  -> This is an Interface Class Defined Device
bDeviceSubClass:                   0x00
bDeviceProtocol:                   0x00
bMaxPacketSize0:                   0x08 = (8) Bytes
idVendor:                        0x0403 = Future Technology Devices International Limited
idProduct:                       0x6001
bcdDevice:                       0x0400
iManufacturer:                     0x01
     English (United States)  "GFL"
iProduct:                          0x02
     English (United States)  "M 504 G"
iSerialNumber:                     0x03
     English (United States)  "GO31123"
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x02  -> Direction: OUT - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x02
wTotalLength:                    0x0020  -> Validated
bNumInterfaces:                    0x01
bConfigurationValue:               0x01
iConfiguration:                    0x00
bmAttributes:                      0x80  -> Bus Powered
MaxPower:                          0x32 = 100 mA

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x04
bInterfaceNumber:                  0x00
bAlternateSetting:                 0x00
bNumEndpoints:                     0x02
bInterfaceClass:                   0xFF  -> Interface Class Unknown to USBView
bInterfaceSubClass:                0xFF
bInterfaceProtocol:                0xFF
iInterface:                        0x02
     English (United States)  "M 504 G"

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x02  -> Direction: OUT - EndpointID: 2
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x00

===================================================
I have confirmed the Baud rate / Data size/parity/stop bits correct.  And as the device can receive my command (Ex. reset command will make the device reset at once), I thing these parameters must be correct.

xxxajk, 2020-01-04:
Yeah, sounds like it is correct.
NAK basically means there's no data available/not ready.
Are you getting the two status bytes? If so, you might have to try fiddling with CTS and DTR.

ziv2013, 2020-01-05:
I can't get the two status bytes. Even it's shown as ACK in protocol analyser.
But my arduino program can get these 2 bytes with FT232R.
I think if the CTS and DTR are wrong, I should also get the 2 status bytes.

xxxajk, 2020-01-06:
I'll try your code as above on a FT232RL and get back to you. I think it may be something else going on here. I'll check it in a day or so.
Also, buf should be 65 bytes allocated, because if you get 64 bytes in, you'll get corruption if you are altering buf[rcvd]. :-)
Meanwhile, are you using at least a mega2560? if so, can you try to see if UHS3 works?

ziv2013, 2020-01-06:
Below is my code, it's modified from example code. I run it from Arduino Uno.  I will try to increase the buffer size and have a try. And what does UHS3 means?

#include <cdcftdi.h>
#include <usbhub.h>

#include <SPI.h>

class FTDIAsync : public FTDIAsyncOper
{
public:
    uint8_t OnInit(FTDI *pftdi);
};

uint8_t FTDIAsync::OnInit(FTDI *pftdi)
{
    uint8_t rcode = 0;

        
    //Set band rate to 9600
    rcode = pftdi->SetBaudRate(9600);
    if (rcode)
    {
        Serial.println(F("SetBaudRate Error"));
        return rcode;
    }
    
    //Even, 2 stop, 7 data
    rcode = pftdi->SetData(FTDI_SIO_SET_DATA_PARITY_EVEN |FTDI_SIO_SET_DATA_STOP_BITS_2| 7);
    if (rcode)
    {
        Serial.println(F("SetData Error"));
        return rcode;
    }
    
    //No flow control
    rcode = pftdi->SetFlowControl(0,0,0);
    if (rcode) 
    {
        Serial.println(F("SetFlowControl"));
        return rcode;
    }

    rcode = pftdi->SetModemControl(0x0101);
    if (rcode) 
    {
        Serial.println(F("SetModemControl"));
        return rcode;
    }
    rcode = pftdi->SetLatency(16);
    if (rcode) 
    {
        Serial.println(F("SetLatency"));
        return rcode;
    }    
        
    long int elsp=millis();
    uint16_t rcvd = 64;
    uint8_t buf[64];
        rcode = pftdi->RcvData(&rcvd, buf);
        if (rcode && rcode != hrNAK)
        {
            Serial.println(F("RcvData Error"));
        }   
        
    //Every time at power on init the device
    char initCMD[] = {'*','R','S','T',0x0d,0x0a,0x00};
    //rcode = pftdi->SndData(strlen(initCMD), (uint8_t*)initCMD);
    //if (rcode)
    {
    //    Serial.println(F("SndData *RST"));
    }    
       /*
    elsp=millis();
    while (millis()-elsp<5000) 
    {
       
        rcode = pftdi->RcvData(&rcvd, buf);
        if (rcode && rcode != hrNAK)
        {
            Serial.println(F("RcvData Error"));
        }
        if (rcvd>2)
        {
              buf[rcvd]=0;
              Serial.println((char*)(buf+2));
        } 
          Serial.println(rcvd);
        delay(100);
        }  
        */
}

USB              Usb;
FTDIAsync        FtdiAsync;
FTDI             Ftdi(&Usb, &FtdiAsync);

void setup()
{
  Serial.begin( 115200 );
  Serial.println(F("Start"));

  if (Usb.Init() == -1)
      Serial.println(F("OSC did not start."));

  delay( 200 );
}

void loop()
{
    long int elsp=millis();
 
    Usb.Task();

    if( Usb.getUsbTaskState() == USB_STATE_RUNNING )
    {
        uint8_t  rcode;

        Serial.println(F("SndData getValueCMD"));
        
        // Send ID command
        //char getValueCMD[] = {'P','H','O','T','O','M','E','T','R','I','C','?',0x0d,0x0a,0x00};
        char getValueCMD[] = {'*','I','D','N','?',0x0d,0x0a,0x00};
        //char getValueCMD[] = {0x01,0x60,'*','I','D','N','?','*','I','D','N','?','*','I','D','N','?','*','I','D','N','?',0x0d,0x0a,0x00};
        rcode = Ftdi.SndData(strlen(getValueCMD), (uint8_t*)getValueCMD);
  	    if (rcode)
        {
            Serial.println(F("getValueCMD Error"));
        }
        elsp=millis();
        while (millis()-elsp<2000) {
          uint16_t rcvd = 64;
          uint8_t buf[64];
        //for (uint8_t i=0; i<64; i++)
        //    buf[i] = 0;  
            
        //Serial.println(F("Rev"));
        //Receive data
        rcode = Ftdi.RcvData(&rcvd, buf);
        //Serial.println(rcvd);
        if (rcode && rcode != hrNAK)
        {
            Serial.println(F("RcvData Error"));
        }

        if (rcvd>2)
        {
              buf[rcvd]=0;
              Serial.println((char*)(buf+2));
        } 
        //if (rcvd!=0) {
          //Serial.println(rcvd);
        //  }
        if (rcode!=0) {
          //Serial.println(rcode);
          delay(16);
          }
        else delay(16);
        }           
      // while (1==1);
    }
}

xxxajk, 2020-01-06:
UHS3 is here --> https://github.com/felis/UHS30

xxxajk, 2020-01-06:
The example is here --> https://github.com/felis/UHS30/tree/master/libraries/UHS_host/UHS_CDC_ACM/examples/USB_HOST_SHIELD/acm_terminal

ziv2013, 2020-01-06:
> The example is here --> https://github.com/felis/UHS30/tree/master/libraries/UHS_host/UHS_CDC_ACM/examples/USB_HOST_SHIELD/acm_terminal

Thanks a lot. I will try it two days later. As the device is used by someone else. And I will feedback the result after testing.

ziv2013, 2020-01-08:
I have tried the acm_terminal.ino in UHS3.0. With FT232R to the USB Host Shield and the other side to a CP2102.   But it doesn't work in my side. There is nothing out in Arduino's serial monitor. I guess it's a bug in UHS3.0 now.

[image]

marktopley, 2023-12-19:
FYI I was having the same issue with a different use of the FT232B chip and after lots of trial and error I found this PR fixes the issue https://github.com/felis/USB_Host_Shield_2.0/pull/174. Not looked at why it fixes it, just happy its working :)

xxxajk, 2023-12-19:
> I have tried the acm_terminal.ino in UHS3.0. With FT232R to the USB Host Shield and the other side to a CP2102. But it doesn't work in my side. There is nothing out in Arduino's serial monitor. I guess it's a bug in UHS3.0 now.
> 
It's not a bug. The UNO simply doesn't have enough RAM. Warning is in the readme...

marktopley, 2023-12-19:
I'm using the library on Particle B524 module and encountered the same issues described. Send and receive on FT232R works fine but receive on FT232B was returning NAK, Send was working correctly though. PR #174 fixed the issue for me.

ziv2013, 2023-12-30:
It's a great new for me. Thanks a lot.

> FYI I was having the same issue with a different use of the FT232B chip and after lots of trial and error I found this PR fixes the issue #174. Not looked at why it fixes it, just happy its working :)

closed 2023-12-30 by ziv2013