felis/USB_Host_Shield_2.0 #511: OSC did not start on DOIT ESP32 Dev Kit V1

Mikegyver, 2020-01-17:
hi felis. i tried to use USB host shield with DOIT ESP32 Dev Kit V1. i did the connection as in the commit a7f67cdfc6a1b0fde114cad4771ae6fe8060b104

GPIO5 : SS, GPIO17 : INT, GPIO18 : SCK, GPIO19 : MISO, GPIO23 : MOSI
while for the RST i connected it to EN pin of the DOIT ESP32 Dev Kit V1.
for the power, 5V is supplied only to the VBUS (the MAX3421E is supplied by 3.3V)

FYI, i using USB_desc example code however, on my serial monitor showing OSC did not start. below is the screenshot of my serial monitor.

[image]

i'm using the following usb host shield.

[image]

at the first place i thought the usb host shield might be broken or damaged. but later i try with arduino nano. and it works fine with the nano. on my serial is shown description of the USB device which is attached to it.

[image]

below is the picture of arduino nano connected to the usb host shield.

[image]

i have been trying different approach like supplying direct 3.3V to the reset pin of the USB host shield and tried toggle high to low and back to high to one of the dev kit pin to the reset pin of the usb host shield but yet still got same OSC did not start.

hope u can help me out regarding this issue. thanks in advance, felis.

best regards.

tmk, 2020-02-14:
I have any experience of ESP32, so this is just my thought.

Full-size USB host shield is designed for 5V board while ESP32 is powered with 3.3V and its IO is not 5V tolerant. I'm not sure this setup is safe. Especially GPIO18(INT) and GPIO19(MISO) pin will be driven with 5V by level shifter, this may give the pins damage or just won't work, I guess. If the pins are the causes you may want to connect the pins directly to MAX3421e without level shifter.

Or just try to give 3.3V power to 5V pin on the Shield, this will make the pins driven 3.3V instead of 5V.

Mikegyver, 2020-02-14:
> I have any experience of ESP32, so this is just my thought.
>
> Full-size USB host shield is designed for 5V board while ESP32 is powered with 3.3V and its IO is not 5V tolerant. I'm not sure this [...]

hi @tmk thanks for the reply. anyway, i don't think there should be an issue since by default, the max 3421E is driven by the 3.3V. 5V rail is just go to the USB port. kindly refer this pcb schematic.

[image]

since 5V is no the operating voltage for the max 3421E, the pcb is already design to avoid the 5V to be supply that IC in order to avoid damage on that IC. regards.

gdsports, 2020-02-14:
The upper right corner shows pads for a chip name HCT something. This chip converts 3.3V to 5V for MISO and INT. You could cut the chip out and bridge the 2 signals to bypass the converter.

Mikegyver, 2020-02-18:
> 
> 
> The upper right corner shows pads for a chip name HCT something. This chip converts 3.3V to 5V for MISO and INT. You could cut the chip out and bridge the 2 signals to bypass the converter.

hi @gdsports i have already remove the level shifter IC which convert 3.3V to 5V for SPI. unfortunately still stuck with OSC did not start. kindly refer the photo below.

[image]

i did test the USB host shield with arduino nano for funtionality. it seems works fine with arduino nano.

[image]

basically i don't any idea anymore. seems to be it such like dead end for it.

jacobsoftwares, 2020-07-26:
I also face exactly the same problem with ESP32 WROOM with USB Host Sheild (for Arduino Uno)

xxxajk, 2020-07-26:
Are you supplying 5v to the VBUS line?

jacobsoftwares, 2020-07-27:
yes i'm... when i run board_qc i get 
Reading REVISION register... Die revision invalid. Value returned: 09

dharmik768, 2021-06-02:
Same thing happening I got
 "OSC did not start."
 Using ESP32 dev board and USB host mini.
It would be really helpful if someone has some solution for this.

LalithkumarReddy10, 2022-01-20:
Even iam facing the same issue with esp32devkit v1..But, the problem is solved by using ESP32S. The usb host shield is working perfectly fine with ESP32S. Maybe the issue is with the "usb host shield 2.0" library which is supporting ESP32S but not esp32devkit v1 board. Below is the link to purchase ESP32S...
https://www.amazon.in/dp/B075YSNVPJ/ref=cm_sw_r_apan_glt_i_D75Q275Y8VE96JJF1WK6?_encoding=UTF8&psc=1

Romeo117g, 2022-01-26:
Hello how are you, the error you have is in the connection of pins 9 and 10, since in the usb host shield they do not have any connection, check the schematic that Mikegyver is showing

lalith10112000, 2022-01-27:
> Hello how are you, the error you have is in the connection of pins 9 and 10, since in the usb host shield they do not have any connection, check the schematic that Mikegyver is showing

Hey,the pins 9 and 10 are connected to the respective pins of max3421e....so no issues with that! I have checked it and the  pcb connections are absolutely fine! The problem lies with the ESP32 DEVKIT V1. You can check my reason for the malfunctioning above.

nagatho, 2022-09-12:
Still I am having problem with this shield.
Is there any solid answer for this problem?

xxxajk, 2022-09-18:
try tying shield's rst to 3.3v directly instead of using the reset from the MCU? 
Don't worry, the max chip is reset via software...

nagatho, 2022-09-19:
Hi,
I figure out the solutions.
Should change SPI speed. Now it works fine

xxxajk, 2022-09-23:
long wires causes that :-)
Glad you found the fix.

allefrodrigo, 2023-04-12:
How can I change the SPI?

golf1481, 2023-07-05:
How to change ?

xxxajk, 2023-07-12:
SPI speed is here:
https://github.com/felis/USB_Host_Shield_2.0/blob/59cc3d287dd1afe8d89856a8d4de0ad8fe9ef3c7/usbhost.h#L182

HTH...

As far as UHS30, I did something smarter, since this problem crops up often when people use wires that are too long.
UHS30 fixes this problem by finding the most reliable spi speed for you, automatically.