felis/USB_Host_Shield_2.0 #516: Unknown Device Connected - VID: 0A12 PID: 0001  BTD initiator failed,  error code:D1 #PS4BT

AakashSundarS, 2020-02-23:
I get this error while initializing my dongle... sometimes I get " couldn't find HID device..." in the PS4BT program... please help me out.... I'm doing this for my College project I need this real urgent...

AakashSundarS, 2020-02-24:
this was the output for the USBdesc program..... Please help me out... 

Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		E0
Device Subclass:	01
Device Protocol:	01
Max.packet size:	40
Vendor  ID:		0A12
Product ID:		0001
Revision ID:		2520
Mfg.string index:	00
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		00B1
Num.intf:		02
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		03
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			03
Max.pkt size:		0010
Polling interval:	01

Endpoint descriptor:
Endpoint address:	01
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Endpoint descriptor:
Endpoint address:	81
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			02
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			03
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			04
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			05
Endpoints:		02
Intf. Class:		E0
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Addr:1(0.0.1)

Jc7tee, 2020-08-14:
@AakashSundarS Did you find the solution?

AakashSundarS, 2020-08-16:
> @AakashSundarS Did you find the solution?

yes.. I did find the solution...
I inserted USB Bluetooth receiver half way through... ( sometimes even 3/4part outside... ) even though we buy the specified USB device... Only some of those work.. It just worked out of luck... 

--> Have you soldering the power supply to the USB interface on the USB Host Controller?
--> Have you checked the specifications of the Bluetooth receiver from the "README.txt" file?

closed 2020-12-08 by AakashSundarS