felis/USB_Host_Shield_2.0 #518: USB Keyboard with internal hub not recongnized as keyboard, just hub.

chrisniael, 2020-03-03:
Arduino Uno
Usb Host Shield v2
Keyboard: HHKB Professional 2 Type-S

I want to transfer usb keyboard to ps/2, I run (USBHIDBootKbd.ino)[https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino] to read usb keyboard input, but fail.

```log
Start
BM Init
Addr:01
NC:01
HID_PROTOCOL_KEYBOARD
bNumEP:01
D1
```

Then I run (USB_desc.ino)(https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USB_desc/USB_desc.ino), the output `device class`, `device subclass`, `device protocol` is `09`, `00`, `00`, I think this means the device is a HUB.

```log
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		09
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		0409
Product ID:		005A
Revision ID:		0100
Mfg.string index:	00
Prod.string index:	00
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0019
Num.intf:		01
Conf.value:		01
Conf.string:		00
Attr.:			E0
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		01
Intf. Class:		09
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0001
Polling interval:	FF

Addr:1(0.0.1)
```

Does that means Usb Host Shield not support keyboard with internal hub?

haakonnessjoen, 2020-10-26:
Did you ever figure out this? I am having the same problems

tmk, 2020-10-27:
Uncomment the line?
https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino#L100

chrisniael, 2020-10-27:
@haakonnessjoen, Not yet.
@tmk, Uncomment this line not work. And I think the line you suggest to uncomment means making Arduino works when a hub device connects to Arduino(Not a keyboard with internal hub).

tmk, 2020-10-27:
@chrisniael Did you try it actually?
I confirmed USBHIDBootKbd.ino read keys from my PD-KB400 with the line, and can't read without the line.
You need the line to support hub.

```
GPS GPS GPS GPS GPS GPS oT GPS GPS GPS DN      >2C<
ASCII:
UP      >2C<
DN      >2C<
ASCII:
UP      >2C<
DN      >2C<
ASCII:
UP      >2C<
DN      >2C<
ASCII:
UP      >2C<
DN      >2C<
ASCII:
UP      >2C<
```

chrisniael, 2020-10-30:
@tmk sorry for the late reply.
Just ran it again, the same result.
My keyboard is PD-KB400WS.

tmk, 2020-10-30:
I think you failed to change the line and build firmware for some reason. Take your time to make sure your build process.

Without `USBHub` USBHIDBootKbd.ino fails with 'D1' as your open post shows because  it sees the internal hub as keyboard.

chrisniael, 2020-10-30:
> Without `USBHub` USBHIDBootKbd.ino fails with 'D1' as your open post shows because it sees the internal hub as keyboard.

I agree with you on this point.

And I found an interesting thing. 
Code: [USBHIDBootKbd.ino](https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHIDBootKbd/USBHIDBootKbd.ino) with `USBHub`
When I connect another keyboard to the USB hub inside the PD-KB400WS, the keyboard can be recognized by the Arduino as a keyboard, and the keyboard input can be read normally.