felis/USB_Host_Shield_2.0 #520: Pro Micro + USB host Mini shield

DFL1973, 2020-03-14:
Hey guys, I'm having a problem that I can't figure out. I tested successfully the "mini" shield with an Arduino Nano. Almost no problem at all, just sometimes the keyboard doesn't initialize, but I'm ok with that, the nano reset button solves the problem.
The real issue was when I moved the exact same setup to the Pro Micro (atmega32u4). I can communicate with the shield (example "board_qc", initial messages), but I never get the response from USB hardware. Already tried 3 different keyboards, mouse, thumb drive and x360 joystick, but the "board_qc" is waiting forever for the device. Actually none of the examples works for me.
I read a lot of topics about similar problems, but still can't solve mine. Already tried external power, check connections, reset rewiring, etc, etc, etc. And again, the exact same setup works on arduino Nano.
Is there something that needs to be configured somehow to work with Pro micro?

fjkse34, 2020-07-11:
https://geekhack.org/index.php?topic=80421.0

chuchana, 2020-10-27:
Does [this](https://github.com/felis/USB_Host_Shield_2.0/issues/392) help?