felis/USB_Host_Shield_2.0 #529: Support for Teensy 4.0

BlueGene00, 2020-04-29:
Hello,

the library works fine with Teensy 3.6, but I  want to test with Teensy 4.0.

When I want to check the code in the Arduino IDE, I get the message:
`C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:1231:2: error: #error "Please define board in avrpins.h"`

So I though I just search for "MK66FX1M0" in all files and add "defined(__IMXRT1062__)".

So I have adjusted:
avrpins.h
```
// pointers are 32 bits on ARM
#define pgm_read_pointer(p) pgm_read_dword(p)

#if defined(CORE_TEENSY) && (defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__) || defined(__IMXRT1062__))

#include "core_pins.h"
#include "avr_emulation.h"
```

```
MAKE_PIN(P0, CORE_PIN0_PORTREG, CORE_PIN0_BIT, CORE_PIN0_CONFIG);
MAKE_PIN(P1, CORE_PIN1_PORTREG, CORE_PIN1_BIT, CORE_PIN1_CONFIG);
MAKE_PIN(P2, CORE_PIN2_PORTREG, CORE_PIN2_BIT, CORE_PIN2_CONFIG);
MAKE_PIN(P3, CORE_PIN3_PORTREG, CORE_PIN3_BIT, CORE_PIN3_CONFIG);
MAKE_PIN(P4, CORE_PIN4_PORTREG, CORE_PIN4_BIT, CORE_PIN4_CONFIG);
MAKE_PIN(P5, CORE_PIN5_PORTREG, CORE_PIN5_BIT, CORE_PIN5_CONFIG);
MAKE_PIN(P6, CORE_PIN6_PORTREG, CORE_PIN6_BIT, CORE_PIN6_CONFIG);
MAKE_PIN(P7, CORE_PIN7_PORTREG, CORE_PIN7_BIT, CORE_PIN7_CONFIG);
MAKE_PIN(P8, CORE_PIN8_PORTREG, CORE_PIN8_BIT, CORE_PIN8_CONFIG);
MAKE_PIN(P9, CORE_PIN9_PORTREG, CORE_PIN9_BIT, CORE_PIN9_CONFIG);
MAKE_PIN(P10, CORE_PIN10_PORTREG, CORE_PIN10_BIT, CORE_PIN10_CONFIG);
MAKE_PIN(P11, CORE_PIN11_PORTREG, CORE_PIN11_BIT, CORE_PIN11_CONFIG);
MAKE_PIN(P12, CORE_PIN12_PORTREG, CORE_PIN12_BIT, CORE_PIN12_CONFIG);
MAKE_PIN(P13, CORE_PIN13_PORTREG, CORE_PIN13_BIT, CORE_PIN13_CONFIG);
MAKE_PIN(P14, CORE_PIN14_PORTREG, CORE_PIN14_BIT, CORE_PIN14_CONFIG);
MAKE_PIN(P15, CORE_PIN15_PORTREG, CORE_PIN15_BIT, CORE_PIN15_CONFIG);
MAKE_PIN(P16, CORE_PIN16_PORTREG, CORE_PIN16_BIT, CORE_PIN16_CONFIG);
MAKE_PIN(P17, CORE_PIN17_PORTREG, CORE_PIN17_BIT, CORE_PIN17_CONFIG);
MAKE_PIN(P18, CORE_PIN18_PORTREG, CORE_PIN18_BIT, CORE_PIN18_CONFIG);
MAKE_PIN(P19, CORE_PIN19_PORTREG, CORE_PIN19_BIT, CORE_PIN19_CONFIG);
MAKE_PIN(P20, CORE_PIN20_PORTREG, CORE_PIN20_BIT, CORE_PIN20_CONFIG);
MAKE_PIN(P21, CORE_PIN21_PORTREG, CORE_PIN21_BIT, CORE_PIN21_CONFIG);
MAKE_PIN(P22, CORE_PIN22_PORTREG, CORE_PIN22_BIT, CORE_PIN22_CONFIG);
MAKE_PIN(P23, CORE_PIN23_PORTREG, CORE_PIN23_BIT, CORE_PIN23_CONFIG);
MAKE_PIN(P24, CORE_PIN24_PORTREG, CORE_PIN24_BIT, CORE_PIN24_CONFIG);
MAKE_PIN(P25, CORE_PIN25_PORTREG, CORE_PIN25_BIT, CORE_PIN25_CONFIG);
MAKE_PIN(P26, CORE_PIN26_PORTREG, CORE_PIN26_BIT, CORE_PIN26_CONFIG);
MAKE_PIN(P27, CORE_PIN27_PORTREG, CORE_PIN27_BIT, CORE_PIN27_CONFIG);
MAKE_PIN(P28, CORE_PIN28_PORTREG, CORE_PIN28_BIT, CORE_PIN28_CONFIG);
MAKE_PIN(P29, CORE_PIN29_PORTREG, CORE_PIN29_BIT, CORE_PIN29_CONFIG);
MAKE_PIN(P30, CORE_PIN30_PORTREG, CORE_PIN30_BIT, CORE_PIN30_CONFIG);
MAKE_PIN(P31, CORE_PIN31_PORTREG, CORE_PIN31_BIT, CORE_PIN31_CONFIG);
MAKE_PIN(P32, CORE_PIN32_PORTREG, CORE_PIN32_BIT, CORE_PIN32_CONFIG);
MAKE_PIN(P33, CORE_PIN33_PORTREG, CORE_PIN33_BIT, CORE_PIN33_CONFIG);
#if defined(__MK64FX512__) || defined(__MK66FX1M0__)
MAKE_PIN(P34, CORE_PIN34_PORTREG, CORE_PIN34_BIT, CORE_PIN34_CONFIG);
MAKE_PIN(P35, CORE_PIN35_PORTREG, CORE_PIN35_BIT, CORE_PIN35_CONFIG);
MAKE_PIN(P36, CORE_PIN36_PORTREG, CORE_PIN36_BIT, CORE_PIN36_CONFIG);
MAKE_PIN(P37, CORE_PIN37_PORTREG, CORE_PIN37_BIT, CORE_PIN37_CONFIG);
MAKE_PIN(P38, CORE_PIN38_PORTREG, CORE_PIN38_BIT, CORE_PIN38_CONFIG);
MAKE_PIN(P39, CORE_PIN39_PORTREG, CORE_PIN39_BIT, CORE_PIN39_CONFIG);
MAKE_PIN(P40, CORE_PIN40_PORTREG, CORE_PIN40_BIT, CORE_PIN40_CONFIG);
MAKE_PIN(P41, CORE_PIN41_PORTREG, CORE_PIN41_BIT, CORE_PIN41_CONFIG);
MAKE_PIN(P42, CORE_PIN42_PORTREG, CORE_PIN42_BIT, CORE_PIN42_CONFIG);
MAKE_PIN(P43, CORE_PIN43_PORTREG, CORE_PIN43_BIT, CORE_PIN43_CONFIG);
MAKE_PIN(P44, CORE_PIN44_PORTREG, CORE_PIN44_BIT, CORE_PIN44_CONFIG);
MAKE_PIN(P45, CORE_PIN45_PORTREG, CORE_PIN45_BIT, CORE_PIN45_CONFIG);
MAKE_PIN(P46, CORE_PIN46_PORTREG, CORE_PIN46_BIT, CORE_PIN46_CONFIG);
MAKE_PIN(P47, CORE_PIN47_PORTREG, CORE_PIN47_BIT, CORE_PIN47_CONFIG);
MAKE_PIN(P48, CORE_PIN48_PORTREG, CORE_PIN48_BIT, CORE_PIN48_CONFIG);
MAKE_PIN(P49, CORE_PIN49_PORTREG, CORE_PIN49_BIT, CORE_PIN49_CONFIG);
MAKE_PIN(P50, CORE_PIN50_PORTREG, CORE_PIN50_BIT, CORE_PIN50_CONFIG);
MAKE_PIN(P51, CORE_PIN51_PORTREG, CORE_PIN51_BIT, CORE_PIN51_CONFIG);
MAKE_PIN(P52, CORE_PIN52_PORTREG, CORE_PIN52_BIT, CORE_PIN52_CONFIG);
MAKE_PIN(P53, CORE_PIN53_PORTREG, CORE_PIN53_BIT, CORE_PIN53_CONFIG);
MAKE_PIN(P54, CORE_PIN54_PORTREG, CORE_PIN54_BIT, CORE_PIN54_CONFIG);
MAKE_PIN(P55, CORE_PIN55_PORTREG, CORE_PIN55_BIT, CORE_PIN55_CONFIG);
MAKE_PIN(P56, CORE_PIN56_PORTREG, CORE_PIN56_BIT, CORE_PIN56_CONFIG);
MAKE_PIN(P57, CORE_PIN57_PORTREG, CORE_PIN57_BIT, CORE_PIN57_CONFIG);
MAKE_PIN(P58, CORE_PIN58_PORTREG, CORE_PIN58_BIT, CORE_PIN58_CONFIG);
MAKE_PIN(P59, CORE_PIN59_PORTREG, CORE_PIN59_BIT, CORE_PIN59_CONFIG);
MAKE_PIN(P60, CORE_PIN60_PORTREG, CORE_PIN60_BIT, CORE_PIN60_CONFIG);
MAKE_PIN(P61, CORE_PIN61_PORTREG, CORE_PIN61_BIT, CORE_PIN61_CONFIG);
MAKE_PIN(P62, CORE_PIN62_PORTREG, CORE_PIN62_BIT, CORE_PIN62_CONFIG);
MAKE_PIN(P63, CORE_PIN63_PORTREG, CORE_PIN63_BIT, CORE_PIN63_CONFIG);
#endif

#if defined(__IMXRT1062__)
MAKE_PIN(P34, CORE_PIN34_PORTREG, CORE_PIN34_BIT, CORE_PIN34_CONFIG);
MAKE_PIN(P35, CORE_PIN35_PORTREG, CORE_PIN35_BIT, CORE_PIN35_CONFIG);
MAKE_PIN(P36, CORE_PIN36_PORTREG, CORE_PIN36_BIT, CORE_PIN36_CONFIG);
MAKE_PIN(P37, CORE_PIN37_PORTREG, CORE_PIN37_BIT, CORE_PIN37_CONFIG);
MAKE_PIN(P38, CORE_PIN38_PORTREG, CORE_PIN38_BIT, CORE_PIN38_CONFIG);
MAKE_PIN(P39, CORE_PIN39_PORTREG, CORE_PIN39_BIT, CORE_PIN39_CONFIG);
#endif

```

usbhost.h:

```
#elif (defined(CORE_TEENSY) && (defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__) ||defined(__IMXRT1062__)|| defined(__MKL26Z64__))) || defined(__ARDUINO_ARC__) || defined(__ARDUINO_X86__) || defined(__MIPSEL__) || defined(STM32F4)
typedef SPi< P13, P11, P12, P10 > spi;
```

But now I get a lot of those errors here:
```
C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h: In static member function 'static void P0::SetDirRead()':

C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:834:17: error: 'PORT_PCR_SRE' was not declared in this scope

     configReg = PORT_PCR_SRE | PORT_PCR_DSE | PORT_PCR_MUX(1); \

                 ^

C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:846:1: note: in expansion of macro 'MAKE_PIN'

 MAKE_PIN(P0, CORE_PIN0_PORTREG, CORE_PIN0_BIT, CORE_PIN0_CONFIG);

 ^

C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:834:32: error: 'PORT_PCR_DSE' was not declared in this scope

     configReg = PORT_PCR_SRE | PORT_PCR_DSE | PORT_PCR_MUX(1); \

                                ^

C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:846:1: note: in expansion of macro 'MAKE_PIN'

 MAKE_PIN(P0, CORE_PIN0_PORTREG, CORE_PIN0_BIT, CORE_PIN0_CONFIG);

 ^

C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:834:61: error: 'PORT_PCR_MUX' was not declared in this scope

     configReg = PORT_PCR_SRE | PORT_PCR_DSE | PORT_PCR_MUX(1); \

                                                             ^

C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:846:1: note: in expansion of macro 'MAKE_PIN'

 MAKE_PIN(P0, CORE_PIN0_PORTREG, CORE_PIN0_BIT, CORE_PIN0_CONFIG);

 ^

C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h: In static member function 'static void P0::SetDirWrite()':

C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:838:17: error: 'PORT_PCR_SRE' was not declared in this scope

     configReg = PORT_PCR_SRE | PORT_PCR_DSE | PORT_PCR_MUX(1); \

                 ^

C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:846:1: note: in expansion of macro 'MAKE_PIN'

 MAKE_PIN(P0, CORE_PIN0_PORTREG, CORE_PIN0_BIT, CORE_PIN0_CONFIG);

 ^

C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master/avrpins.h:838:32: error: 'PORT_PCR_DSE' was not declared in this scope

     configReg = PORT_PCR_SRE | PORT_PCR_DSE | PORT_PCR_MUX(1); \
```

So I thought it was easy, but now I have no idea what might be the issue or how to correctly add the Teensy 4.0 to the library. Any ideas?

xxxajk, 2020-04-29:
Use UHS 3 instead.

BlueGene00, 2020-04-29:
Some more background: I am using the mini USB shield (MAX 3421E) together with a Bluetooth-Dongle for PS4 controller connection.

So my goal is not to access the internal USB host of the teensy, but use my existing and working code just with the Teensy 4.0 instead of 3.6.

As far as I can see, the UHS3 is still in Alpha and does not have any example code for the PS4 (Bluetooth). So it would be more work to go this way instead of making the Teensy 4 work with the UHS2 library. 

Is it much work to implement the Teensy 4 or have I just missed some steps in my approach?

xxxajk, 2020-04-29:
UHS3 supports MAX 3421E on Teensy 4. 
Bluetooth needs implementation though.
Since I do not have much time, or the various joysticks or USB, the implementations for those have been on a back burner.

PaulStoffregen, 2020-04-30:
> Use UHS 3 instead.

Why can't you understand nobody wants to use UHS30 when the readme says this:

```
Pre-release of USB Host Library version 3.0. No technical support offered at this time.

This set of libraries is EXTREMELY ALPHA!
```

xxxajk, 2020-04-30:
Oh, I understand, I just don't think it is quite ready for the general public yet.
it lacks a lot of drivers, and documentation.

PaulStoffregen, 2020-05-01:
Why then do you suggest "use UHS 3 instead" here and on places like Arduino's mail list?

xxxajk, 2020-05-02:
Never thought of asking the devs for help in porting drivers. Think I should?

Lauszus, 2020-05-02:
@BlueGene00 another option would simply be to do something like this: https://github.com/felis/USB_Host_Shield_2.0/blob/master/avrpins.h#L1221-L1239, but simply use the `digitalWriteFast`, `pinModeFast` and `digitalReadFast`, as those function should be fast enough and will make it easier to support future versions of Teensy boards.

BlueGene00, 2020-05-07:
@Lauszus 
Thank you very much, this solution is working and I can it now use with the Teensy 4.0 and Bluetooth dongle.

digitalWriteFast and digitalReadFast are working, but with pinModeFast I get this error:

`error: 'pinModeFast' was not declared in this scope
`
Here is the code, which I have added:

```
#elif defined(CORE_TEENSY) && (defined(__IMXRT1062__))

// we could get lower level by making these macros work properly.
// for now just use the semi optimised version, it costs a lookup in the pin pgm table per op
// but for now it will do.
//#define GPIO_BITBAND_ADDR(reg, bit) (((volatile uint8_t *)&(reg) + ((bit) >> 3)))
//#define GPIO_BITBAND_MASK(reg, bit) (1<<((bit) & 7))
//#define GPIO_BITBAND_PTR(reg, bit) ((volatile uint8_t *)GPIO_BITBAND_ADDR((reg), (bit)))

#include "core_pins.h"
#include "avr_emulation.h"

#define MAKE_PIN(className, pin) \
class className { \
public: \
  static void Set() { \
    digitalWriteFast(pin, HIGH);\
  } \
  static void Clear() { \
    digitalWriteFast(pin, LOW); \
  } \
  static void SetDirRead() { \
    pinMode(pin, INPUT); \
  } \
  static void SetDirWrite() { \
    pinMode(pin, OUTPUT); \
  } \
  static uint8_t IsSet() { \
    return digitalReadFast(pin); \
  } \
};

MAKE_PIN(P0, 0);
MAKE_PIN(P1, 1);
MAKE_PIN(P2, 2);
MAKE_PIN(P3, 3);
MAKE_PIN(P4, 4);
MAKE_PIN(P5, 5);
MAKE_PIN(P6, 6);
MAKE_PIN(P7, 7);
MAKE_PIN(P8, 8);
MAKE_PIN(P9, 9);
MAKE_PIN(P10, 10);
MAKE_PIN(P11, 11);
MAKE_PIN(P12, 12);
MAKE_PIN(P13, 13);
MAKE_PIN(P14, 14);
MAKE_PIN(P15, 15);
MAKE_PIN(P16, 16);
MAKE_PIN(P17, 17);
MAKE_PIN(P18, 18);
MAKE_PIN(P19, 19);
MAKE_PIN(P20, 20);
MAKE_PIN(P21, 21);
MAKE_PIN(P22, 22);
MAKE_PIN(P23, 23);
MAKE_PIN(P24, 24);
MAKE_PIN(P25, 25);
MAKE_PIN(P26, 26);
MAKE_PIN(P27, 27);
MAKE_PIN(P28, 28);
MAKE_PIN(P29, 29);
MAKE_PIN(P30, 30);
MAKE_PIN(P31, 31);
MAKE_PIN(P32, 35);
MAKE_PIN(P33, 33);
MAKE_PIN(P34, 34);
MAKE_PIN(P35, 35);
MAKE_PIN(P36, 36);
MAKE_PIN(P37, 37);
MAKE_PIN(P38, 38);
MAKE_PIN(P39, 39);

#undef MAKE_PIN
```

Lauszus, 2020-05-07:
@BlueGene00 great to hear! Don't mind about the `pinMode`, as it's only called once for each pin during initialization anyway.

Do you mind sending a pull request, so others can benefit from your work?

commit 71aeadea (2020-11-17): Added support for Teensy 4.0 and 4.1

Fixes #529, fixes #560
PR #564 (2020-11-18): Added support for Teensy 4.0 and 4.1
Fixes #529, fixes #560
closed 2020-11-18 by Lauszus