felis/USB_Host_Shield_2.0 #530: Read Arduino DUE programming port

vbocci, 2020-05-03:
Why the acm_terminal does'not wok with USB CDC serial of SAM3x8e of Arduino due ?

I want to read data coming from arduino due usb programming port (output generated using SerialUSB.print ...) . I can read this data easily using serial terminal for Windows or MACos.

Try to do the same with ESP32 and host USB using acm_terminal .

1) the acm_terminal stop waiting Acm.isReady() thats come true ...never happen
2) I try to do some work around using https://github.com/felis/USB_Host_Shield_2.0/issues/295
following settings.h DEBUG on and comment in cdcacm.c two lines
//if(bNumEP < 4)
//      return USB_DEV_CONFIG_ERROR_DEVICE_NOT_SUPPORTED;
from my understanding this force in some way Acm.isReady()

In this case something happen I can read the first time 4 characters coming from my arduino due correctly but nothing more:
DEBUGON output:
Start
Start main loop waiting Acm.isReady()
ACM Init
Addr:01
NC:01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0200
Poll Intrv:	00
Endpoint descriptor:
Length:		07
Type:		05
Address:	83
Attributes:	02
MaxPktSize:	0200
Poll Intrv:	00
Conf:01
ACM configured

my print to understand what happen
rcvd:4
i:0) HEX:24 ascii:$
i:1) HEX:30 ascii:0
i:2) HEX:D ascii:

i:3) HEX:A ascii:

infinite loop  no other character is read the rcvd is always 0 and stack

4) running the USB_desc.ino I have the following descriptor about the arduino due programming USB:
tart

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		2341
Product ID:		003E
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		004B
Num.intf:		02
Conf.value:		01
Conf.string:		00
Attr.:			C0
Max.pwr:		FA
Unknown descriptor:
Length:		08
Type:		0B
Contents:	0002020201000904

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		01
Intf. Class:		02
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		05
Type:		24
Contents:	0010010524
Unknown descriptor:
Length:		05
Type:		24
Contents:	0101010424
Unknown descriptor:
Length:		04
Type:		24
Contents:	02060524
Unknown descriptor:
Length:		05
Type:		24
Contents:	0600010705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0010
Polling interval:	10

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		0A
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			02
Max.pkt size:		0200
Polling interval:	00

Endpoint descriptor:
Endpoint address:	83
Attr.:			02
Max.pkt size:		0200
Polling interval:	00

Addr:1(0.0.1)

Any suggestion or wokaround ?

Thanks

sauvant, 2021-09-15:
Exactly same issue here - did you come around that somehow?

touchgadget, 2021-09-15:
@sauvant 
The Due says its first interface class =2 , subclass =2 , and protocol =0.
```
Interface descriptor:
Intf.number: 00
Alt.: 00
Endpoints: 01
Intf. Class: 02
Intf. Subclass: 02
Intf. Protocol: 00
```

But cdcacm.cpp says it wants (2, 2, 1) so the Due interrupt interface is rejected. In the other thread, people mention changing bNumEP but this is clearly wrong. The number of endpoints is wrong because the interrupt interface is rejected.

```
                ConfigDescParser< USB_CLASS_COM_AND_CDC_CTRL,
                        CDC_SUBCLASS_ACM,
                        CDC_PROTOCOL_ITU_T_V_250,
                        CP_MASK_COMPARE_CLASS |
                        CP_MASK_COMPARE_SUBCLASS |
                        CP_MASK_COMPARE_PROTOCOL > CdcControlParser(this);
```

The following might work. This sets the protocol = 0 and removes CP_MASK_COMPARE_PROTOCOL. This SHOULD (untested) allow any value in the protocol field.

```
                ConfigDescParser< USB_CLASS_COM_AND_CDC_CTRL,
                        CDC_SUBCLASS_ACM,
                        0,
                        CP_MASK_COMPARE_CLASS |
                        CP_MASK_COMPARE_SUBCLASS > CdcControlParser(this);
```