felis/USB_Host_Shield_2.0 #533: cypress usb to serial device "usb_state_error"

fotosettore, 2020-05-28:
Hi !
I made a usb host connected to ESP32 and it works fine using midi devices or usb keys.
Unfortunately the usb communication interface of my solar inverter is not recognized.
It should by a cypress usb to serial.  (vendor 0665 - device 5161)
The error given is :

`USB_STATE_ERROR    0xa0`

This is the usb_desc output :

```
Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0110
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	08
Vendor  ID:		0665
Product ID:		5161
Revision ID:		0002
Mfg.string index:	03
Prod.string index:	01
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0022
Num.intf:		01
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		01
Intf. Class:		03
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		21
Contents:	11010001221B000705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0008
Polling interval:	0C

Addr:1(0.0.1)
```

Any help is appreciated

xxxajk, 2020-05-29:
```
Intf. Class:		03
```
This is a HID device
```
Intf. Subclass:		00
Intf. Protocol:		00
```
On to the descriptor.

```
Unknown descriptor:
Length:		09
Type:		21 <-- HID descriptor
Contents:	11010001221B000705

0x11 0x01 USB spec 1.11
0x00 Country Code None
0x01 One Descriptor
0x22 Descriptor Type is Report
0x1B 0x00  Descriptor Length total size 0x001B
0x00 optional type apparently none. This means it is vendor specific.
0x07 0x05 and this is what ever the custom host software needs to know.

```

That said, this is NOT a serial device, it is HID. How to make it do what you want is anybody's guess.

fotosettore, 2020-05-29:
so none did ever tried to connect a similar device ?
do you think i have no hope to solve this problem ?

xxxajk, 2020-05-30:
Without knowing how to communicate with this device, it's guess work.
You can also forget about cypress releasing anything. They refuse (for whatever reason) to work with the open source community and provide any information whatsoever.

I don't work (much) on UHS2 any more, but I do work on UHS30 when I get the time to do so. There has been recent commits to HID on UHS30 that might help you out, specifically triggering code within your sketch when a report packet arrives. Perhaps you could use UHS30 to gather data over the course of some time and figure out what the values mean, kind of like a data sniffer. That is exactly how I managed to monitor the keyboard and mouse packets. along with other unknown packets from other interfaces.

fotosettore, 2020-05-31:
Many thanks for your comprehensive reply.
About three years ago I fought hard to solve the same communication problem between this damn chip and the Visual Studio code.
I did it, very fortunately and after a long time, because of my lack of knowledge of the subject in terms of libraries.
Apparently we have the same problems here.
I just have to sniff again during the first connection between chip and computer and see if I can understand something.
At this point I ask you a further question: the **USB_STATE_ERROR 0xa0** code therefore means that the library cannot encode what is sent by the chip (because it has no information about it) and not because there is a physical error of incommunicability. Right ?