felis/USB_Host_Shield_2.0 #534: using pendrive with usb_host_shield

fotosettore, 2020-06-03:
hi
is it possible to read/write a file txt on a pen drive connected to usb host ?
if yes, is there an example somewhere ?
many thanks

xxxajk, 2020-06-04:
Yes, but not easy to do. 
UHS3.0 has this ability built-in.
https://github.com/felis/UHS30
NOTE, this is VERY difficult to do on smaller AVR boards. Don't expect it to be able to do much on an UNO, even with UHS2.0 Filesystems eat RAM like candy, and USB is heavy too.

fotosettore, 2020-06-05:
hi xxxajk
I'd like to use a esp32. 
In relation to this I put a issue on [https://github.com/felis/UHS30/issues/60](url). 
I saw that someone sent a mega adk for testing.
Can I help in some way with esp32 hardware that is needed for development?

xxxajk, 2020-06-05:
Have ESP32, lack time, though. That's why so much is behind.

Meins321, 2020-10-27:
@xxxajk @fotosettore 
https://github.com/greiman/UsbFat
i am currently experimenting with this library just to use some old 4gb usb sticks as measurement storage.. maybe it helps you too?
I get the same error as here reported:
https://github.com/greiman/UsbFat/issues/10

even with a 9V powersupply and different usb sticks sadly :-1: