felis/USB_Host_Shield_2.0 #536: USBHIDBootKbd does not detect the key strokes, althought HidKeyboard.isReady() == true

seasoftvuhphuong, 2020-07-09:
The events fires up when i attach a mouse, but not my keyboard.

Start
BM Init
Addr:01
NC:01
HID_PROTOCOL_KEYBOARD AND MOUSE
bNumEP:02
D1
BM Init
Addr:01
NC:01
HID_PROTOCOL_KEYBOARD
bNumEP:02
Cnf:01
bIfaceNum:00
bNumIface:00

Interface:00
PROTOCOL SET HID_BOOT rcode:00
SET_IDLE rcode:00
RPIPE rcode:00
BM configured
Keyboard is Ready
Keyboard is Ready
Keyboard is Ready
Keyboard is Ready
.
.
.

Please help

seasoftvuhphuong, 2020-07-11:
Thanks everybody!
In short: i made it successfully when using the oldest USB_Host_Shield_2.0 library (USB_Host_Shield_Library_2_0-1.0.0)

More information:

To transform my mech keyboard into a bluetooth one, i'm using:
- HC-05 as a Bluetooth Adaptor
- Arduino Pro Mini 3.3V
- USB Host Shield 2.0 ADK
- CM Storm Quick Fire TK (Disable NKRO & active 6KRO via key shortcut - See kb's manual)

[image]

I've built this combo using online instructions. Everything went smooth except:

> It only works with Generic Keyboard model (Logitech K120, ...). It even work with my USB mouse (disconnect after short period of time right after i move my mouse - since it's refresh rate is too high - 3500Mhz)

When my keyboard attached, the Arduino Serial Monitor doesn't fire up any key pressed events (even "BM configured" is printed, which mean the BOOT_PROTOCOL was initialized successfully, and the keyboard leds in On).

I guessed that's not the library bug, only a incompability issue from my kb's stock firmware initialize process, since the kb itself has to make a choice for the BOOT_PROTOCOL or active itself 6KRO/NKRO protocol. So i digged deeper into the `hidboot.h` file, try changing sooo many lines in that file to make it compatible but nothing works.

I've had many headcahe in many hours to debugging last night. After taking a sleep, take a cup of coffee, i think i will try to use the old version of this library to see if it works or not. And voila, the old version (1.0.0) works OOB, so luckily i am.

So, in general, it's not the library issue at all. But i left this comment here to help anybody which face a same problem with me.
Thanks community!

closed 2020-07-11 by seasoftvuhphuong

zpdr, 2022-02-24:
May I ask what mouse you are using? my mouse doesn't even support hisboot