felis/USB_Host_Shield_2.0 #538: Usb.Init() failes on ESP32 + Arduino USB Host Shield

turkysha, 2020-07-24:
I installed everything and ESP32 is working with Arduino IDE. I tried blink example and esp32 was working. Then I connected esp32 to Arduino USB Host Shield, installed this library and run USB_desc example. Everything is connected correctly ( I checked everything 10 times, like SS, MISO, MOSI, VCC, GND ect.). I can not get Usb.Init() working, so I suppose I can't establish connection between esp32 and arduino host shield. This is my output on serial monitor:

[image]

Shield is getting 5V from esp32 and I got mouse connected to shield. Mouse also gets power.

iwbrhwfy-source, 2020-08-10:
same problem :(

gdsports, 2020-08-11:
If using USB host shield, it has 5V logic level shifters which is bad for ESP32 3.3V inputs. This may damage ESP32 inputs. The USB host mini board is ideal because it works at 3.3V. If using the mini board be sure to pull RESET to 3.3V. More details in the thread below.

https://github.com/felis/USB_Host_Shield_2.0/issues/324

turkysha, 2020-08-12:
@iwbrhwfy-source 
I soleved the issue.

You need first to change every SPISettings in usbhost.h from 26000000 to 4000000 like this:

SPISettings(26000000, MSBFIRST, SPI_MODE0)

to

SPISettings(4000000, MSBFIRST, SPI_MODE0).

then you need to connect your reset pin from USB host shield to 5V via 2.2K resistor (probably working with restet on 5V) and you need to connect your IOREF pin from USB host shiled to 3.3 V on esp32.

Changing settings in usbhost.h give you correct signal from esp32 to USB host because esp32 is too fast for shiled. Connecting reset pin changes your USB host shiled from reset mode to working mode and connecting IOREF pin on USB shiled gets power to out modules to get power ( I think the MISO).

closed 2020-08-13 by turkysha