felis/USB_Host_Shield_2.0 #544: Connecting to USB host mini through an active usb cable

Jc7tee, 2020-08-15:
Hi all

I have a xboxone pdp afterglow controller and I managed to get it working with mega2560 using the usb host mini. I'm using a wired controller and in my application, I need extra length. So I connected the controller to the usb host mini through an active usb cable (20m). The board_qc was able to detect the controller but the PID and the VID changed. So I modified xboxone.h accordingly.

When I upload the xboxone.ino sketch, it says:

XBOX USB Library Started
Unknown Device Connected - VID: 1A40 PID: 0101
Xbox One Init Failed, error code: D1

Xbox One Controller Disconnected

Here's what I got from usb_desc with the active usb cable connected:

22:01:27.587 -> 01
22:01:27.587 -> --
22:01:27.587 -> 
22:01:27.587 -> Device descriptor: 
22:01:27.587 -> Descriptor Length:	12
22:01:27.587 -> Descriptor type:	01
22:01:27.587 -> USB version:		0200
22:01:27.587 -> Device class:		09
22:01:27.587 -> Device Subclass:	00
22:01:27.587 -> Device Protocol:	00
22:01:27.587 -> Max.packet size:	40
22:01:27.587 -> Vendor  ID:		1A40
22:01:27.587 -> Product ID:		0101
22:01:27.587 -> Revision ID:		0111
22:01:27.587 -> Mfg.string index:	00
22:01:27.587 -> Prod.string index:	01
22:01:27.587 -> Serial number index:	00
22:01:27.587 -> Number of conf.:	01
22:01:27.587 -> 
22:01:27.587 -> Configuration descriptor:
22:01:27.587 -> Total length:		0019
22:01:27.587 -> Num.intf:		01
22:01:27.621 -> Conf.value:		01
22:01:27.621 -> Conf.string:		00
22:01:27.621 -> Attr.:			E0
22:01:27.621 -> Max.pwr:		32
22:01:27.621 -> 
22:01:27.621 -> Interface descriptor:
22:01:27.621 -> Intf.number:		00
22:01:27.621 -> Alt.:			00
22:01:27.621 -> Endpoints:		01
22:01:27.621 -> Intf. Class:		09
22:01:27.621 -> Intf. Subclass:		00
22:01:27.621 -> Intf. Protocol:		00
22:01:27.621 -> Intf.string:		00
22:01:27.621 -> 
22:01:27.621 -> Endpoint descriptor:
22:01:27.621 -> Endpoint address:	81
22:01:27.621 -> Attr.:			03
22:01:27.621 -> Max.pkt size:		0001
22:01:27.621 -> Polling interval:	FF
22:01:27.621 -> 
22:01:27.621 -> 
22:01:27.621 -> Addr:1(0.0.1)

Here's what I got without the active usb connected:

01:57:49.106 -> Device descriptor: 
01:57:49.106 -> Descriptor Length:	12
01:57:49.106 -> Descriptor type:	01
01:57:49.106 -> USB version:		0200
01:57:49.106 -> Device class:		FF
01:57:49.106 -> Device Subclass:	47
01:57:49.106 -> Device Protocol:	D0
01:57:49.106 -> Max.packet size:	40
01:57:49.106 -> Vendor  ID:		0E6F
01:57:49.140 -> Product ID:		02A8
01:57:49.140 -> Revision ID:		010F
01:57:49.140 -> Mfg.string index:	01
01:57:49.140 -> Prod.string index:	02
01:57:49.140 -> Serial number index:	03
01:57:49.140 -> Number of conf.:	01
01:57:49.140 -> 
01:57:49.140 -> Configuration descriptor:
01:57:49.140 -> Total length:		0040
01:57:49.140 -> Num.intf:		02
01:57:49.140 -> Conf.value:		01
01:57:49.140 -> Conf.string:		00
01:57:49.140 -> Attr.:			A0
01:57:49.140 -> Max.pwr:		FA
01:57:49.140 -> 
01:57:49.140 -> Interface descriptor:
01:57:49.140 -> Intf.number:		00
01:57:49.140 -> Alt.:			00
01:57:49.140 -> Endpoints:		02
01:57:49.140 -> Intf. Class:		FF
01:57:49.140 -> Intf. Subclass:		47
01:57:49.140 -> Intf. Protocol:		D0
01:57:49.140 -> Intf.string:		00
01:57:49.140 -> 
01:57:49.140 -> Endpoint descriptor:
01:57:49.140 -> Endpoint address:	81
01:57:49.173 -> Attr.:			03
01:57:49.173 -> Max.pkt size:		0040
01:57:49.173 -> Polling interval:	04
01:57:49.173 -> 
01:57:49.173 -> Endpoint descriptor:
01:57:49.173 -> Endpoint address:	01
01:57:49.173 -> Attr.:			03
01:57:49.173 -> Max.pkt size:		0040
01:57:49.173 -> Polling interval:	04
01:57:49.173 -> 
01:57:49.173 -> Interface descriptor:
01:57:49.173 -> Intf.number:		01
01:57:49.173 -> Alt.:			00
01:57:49.173 -> Endpoints:		00
01:57:49.173 -> Intf. Class:		FF
01:57:49.173 -> Intf. Subclass:		47
01:57:49.173 -> Intf. Protocol:		D0
01:57:49.173 -> Intf.string:		00
01:57:49.173 -> 
01:57:49.173 -> Interface descriptor:
01:57:49.173 -> Intf.number:		01
01:57:49.173 -> Alt.:			01
01:57:49.173 -> Endpoints:		02
01:57:49.207 -> Intf. Class:		FF
01:57:49.207 -> Intf. Subclass:		47
01:57:49.207 -> Intf. Protocol:		D0
01:57:49.207 -> Intf.string:		00
01:57:49.207 -> 
01:57:49.207 -> Endpoint descriptor:
01:57:49.207 -> Endpoint address:	02
01:57:49.207 -> Attr.:			01
01:57:49.207 -> Max.pkt size:		00E0
01:57:49.207 -> Polling interval:	01
01:57:49.207 -> 
01:57:49.207 -> Endpoint descriptor:
01:57:49.207 -> Endpoint address:	83
01:57:49.207 -> Attr.:			01
01:57:49.207 -> Max.pkt size:		0080
01:57:49.207 -> Polling interval:	01
01:57:49.207 -> 
01:57:49.207 -> 
01:57:49.207 -> Addr:1(0.0.1)

Any help will be highly appreciated.

gdsports, 2020-08-16:
The active usb cable may be a USB hub so you may have to add hub support in your sketch. Look at the other examples for help.

Jc7tee, 2020-08-16:
@gdsports I uncommented `USBHub hub(&Usb)` in board_qc example:

```
/* objects */
USB Usb;
USBHub hub(&Usb);
```

It passed the test but hangs at "Waiting for devices..." in the serial monitor

Jc7tee, 2020-08-30:
I tried connecting the usb cable to my PC. In the device manager, two Generic USB Hubs were registered. I find it strange as the cable has only one input port and an output port. I tried declaring two USBHub objects as follows:

```
USBHub hub1(&Usb);
USBHub hub2(&Usb);
```

but still doesn't work

Jc7tee, 2020-09-22:
I did some more testing today.
I ran the hub_demo and I managed to get some output. I managed to get this output consistently with an xbox one controller connected:

```
41
--
String Descriptors:
Product:		USB 2.0 Hub

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		09
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		1A40
Product ID:		0101
Revision ID:		0111
Mfg.string index:	00
Prod.string index:	01
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0019
Num.intf:		01
Conf.value:		01
Conf.string:		00
Attr.:			E0
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		01
Intf. Class:		09
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0001
Polling interval:	FF

Addr:41(1.0.1)
```
Other times I got this:

```
Start

41
--
String Descriptors:
Product:		USB 2.0 Hub

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		09
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		1A40
Product ID:		0101
Revision ID:		0111
Mfg.string index:	00
Prod.string index:	01
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0019
Num.intf:		01
Conf.value:		01
Conf.string:		00
Attr.:			E0
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		01
Intf. Class:		09
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0001
Polling interval:	FF

0C
--
String Descriptors:
Product:		Error retrieving LangID table length
D

Request error. Error code:	0D
Addr:41(1.0.1)
Addr:C(0.1.4)
```
All the above cases were achieved with only one instance of USBHub declared. When I declared two instances, the second address somehow becomes 4A.

I think I'm getting to something. Can anyone please help?

Jc7tee, 2020-10-04:
I ended up using an aoc cable

closed 2020-10-04 by Jc7tee