felis/USB_Host_Shield_2.0 #550: Alright guys, it works now. I just have to modify the [`OemToAscii`](https://github.com/felis/USB_Host_Shield_2.0/blob/master/hidboot.cpp#L165) to satisfy my need. Here is my final code :

NoahsArk021913, 2020-09-18:
Alright guys, it works now. I just have to modify the [`OemToAscii`](https://github.com/felis/USB_Host_Shield_2.0/blob/master/hidboot.cpp#L165) to satisfy my need. Here is my final code :

```c
#include <usbhid.h>
#include <usbhub.h>
#include <hiduniversal.h>
#include <hidboot.h>
#include <SPI.h>

class MyParser : public HIDReportParser {
  public:
    MyParser();
    void Parse(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf);
  protected:
    uint8_t KeyToAscii(bool upper, uint8_t mod, uint8_t key);
    virtual void OnKeyScanned(bool upper, uint8_t mod, uint8_t key);
    virtual void OnScanFinished();
};

MyParser::MyParser() {}

void MyParser::Parse(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf) {
  // If error or empty, return
  if (buf[2] == 1 || buf[2] == 0) return;

  for (uint8_t i = 7; i >= 2; i--) {
    // If empty, skip
    if (buf[i] == 0) continue;

    // If enter signal emitted, scan finished
    if (buf[i] == UHS_HID_BOOT_KEY_ENTER) {
      OnScanFinished();
    }

    // If not, continue normally
    else {
      // If bit position not in 2, it's uppercase words
      OnKeyScanned(i > 2, buf, buf[i]);
    }

    return;
  }
}

uint8_t MyParser::KeyToAscii(bool upper, uint8_t mod, uint8_t key) {
  // Letters
  if (VALUE_WITHIN(key, 0x04, 0x1d)) {
    if (upper) return (key - 4 + 'A');
    else return (key - 4 + 'a');
  }

  // Numbers
  else if (VALUE_WITHIN(key, 0x1e, 0x27)) {
    return ((key == UHS_HID_BOOT_KEY_ZERO) ? '0' : key - 0x1e + '1');
  }

  return 0;
}

void MyParser::OnKeyScanned(bool upper, uint8_t mod, uint8_t key) {
  uint8_t ascii = KeyToAscii(upper, mod, key);
  Serial.print((char)ascii);
}

void MyParser::OnScanFinished() {
  Serial.println(" - Finished");
}

USB          Usb;
USBHub       Hub(&Usb);
HIDUniversal Hid(&Usb);
MyParser     Parser;

void setup() {
  Serial.begin( 115200 );
  Serial.println("Start");

  if (Usb.Init() == -1) {
    Serial.println("OSC did not start.");
  }

  delay( 200 );

  Hid.SetReportParser(0, &Parser);
}

void loop() {
  Usb.Task();
}
```

Now that it's done, the code looks really simple. I guess it took me so long because I never use C or C++ before, which make me a bit confused about pointer, bits and bytes.

---

Before I close this issue, I have several questions to ask. While I'm making this parser, I've tried this code :

```c
void MyParser::Parse(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf) {
  for (uint8_t i = 0; i < 8; i++) {
    Serial.print(buf[i]);
    Serial.print(" ");
  }
}
```

When I use this code to scan barcode with letter `A` on it, it gives following result :

```
0 0 225 4 0 0 0 0 
```

My questions are :

1. Why the bit value only started and checked from index 2 ? 
    What is the purpose of bit in index 0 and 1 ? 
    Are they always empty ?

2. According to this [blog](https://oscarliang.com/arduino-difference-byte-uint8-t-unsigned-cha/), `uint8_t` data type is basically the same as `byte` in Arduino. Both of them is used to store an 8-bit unsigned number, from 0 to 255. So, when I loop through a variable with data type `uint8_t` or `byte`, I should get the bit value on index 0 to 7. Now, I know that bit is only 1 and 0. So, why that loop print value 225 and 4 ? Or maybe I'm misunderstanding the purpose of that loop ?

Well, that's all. Thank you.

_Originally posted by @RadhiFadlillah in https://github.com/felis/USB_Host_Shield_2.0/issues/323#issuecomment-338474528_

rousir, 2020-09-18:
hi，Recently encountered a problem that you met, you may be able to deal with it like this
https://github.com/felis/USB_Host_Shield_2.0/issues/551

Dave-Osborne, 2020-11-04:
@NoahsArk021913 Your code sample really helped me get my barcode scanner working!

I resolved the compile errors by removing the 'mod' parameter on the KeyToAscii and OnKeyScanned functions, as the variable wasn't being used anyway.

furiousdragon164, 2021-01-11:
Hi, i´m new to programing arduino so I don´t know much about but i would like to know where do you keep you code storage so I could compare it with some pre-defined codes so I can activate the arduino exits