felis/USB_Host_Shield_2.0 #555: Support for the xbox one elite controller v2

Pouletgrill, 2020-10-06:
Hello, i am not sure to write this at the right place but is there a plan to support the xbox one elite controller v2 via Bluetooth in the near futur ? im thinking of doing it myself but i have no experience in reverse engeneering.

thanks!

Lauszus, 2020-11-16:
@Pouletgrill please try the latest code on the master branch. I just added support for the Xbox One S controller via Bluetooth, so I believe this controller should work as well.

Pouletgrill, 2020-11-25:
Great, thank you, i will gladly test it a soon as i can.
in the meantime if anyone try it before me you can juste let me know if it worked.

Good work!, love the project

Lauszus, 2020-11-25:
@Pouletgrill great, thanks!

Viper7171, 2023-03-14:
Trying to get mine to connect with a USB cable and can't get it to connect. When I plug the controller in my Adafruit LCD shield goes out. Thinking I may have a bad USB Shield or something.

Viper7171, 2023-03-29:
Solved part of my problem. My LCD shield was not RGB so when it set the color to Green it went off. Took that out.

The screen worked but the controller didn't work so I tried my new Series X controller and that worked. 

So I'm still unable to get my Elite Series 2 to connect.